espressif/esp-idf #7141: Reading ILI9341 registers (ILI9488, etc.) using the ESP-IDF SPI API - the response is always zero. [There is no answer on the forum!!!]

SinglWolf, 2021-06-12:
How to read ILI9341(ILI9488, etc) registers using ESP-IDF SPI API?
The forum has a topic [ILI9341 read registers in ESP-IDF](https://esp32.com/viewtopic.php?f=13&t=14277) - no answer!
There is also a mention of this problem on stackoverflow.com [Reading from an SPI bus using the ESP-IDF (ESP32) and an ILI9341 display adapter. Not getting meaningful data](https://stackoverflow.com/questions/67686453/reading-from-an-spi-bus-using-the-esp-idf-esp32-and-an-ili9341-display-adapter) - there is no solution.
For my [Radiola-S2](https://serverdoma.ru/viewtopic.php?f=70&t=1178) project, reading the display controller registers is necessary to organize automatic detection of the connected display.
My project also uses a touchscreen on the XPT2046 controller. All attempts to read XPT2046 registers are in vain. Zeros are returned.
I am using ESP32-S2 and ESP-32.
Why does register reading work correctly on the espressif/arduino-esp32 framework, but not on esp-idf?

nopnop2002, 2021-12-10:
>My project also uses a touchscreen on the XPT2046 controller. All attempts to read XPT2046 registers are in vain. Zeros are returned.

Try this.
https://github.com/nopnop2002/esp-idf-ili9340

SinglWolf, 2021-12-12:
> Try this.

What exactly should I try there?
I already know that when using the display and the touchscreen on the same SPI bus, the SDO display pin (MISO) cannot be connected to the SPI bus. The pin must be free.

closed 2021-12-12 by SinglWolf