espressif/esp-idf #7142: Getting "E (527) esp_image: Image length 1096624 doesn't fit in partition length 1048576" when my app is only 1MB in size.

DerrickLau, 2021-06-13:
## Environment

- Development Kit:     ESP32-PICO-Kit
- Kit version PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    4.3.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    8.2.0
- Operating System:     Windows10
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes Visual Studio Code 2019 + Platform IO (but building using idf.py build)
- Power Supply:         USB

## Problem Description
Ran idf.py -p COM4 -b 115200 app-flash:

E (527) esp_image: Image length 1096624 doesn't fit in partition length 1048576
E (527) boot: Factory app partition is not bootable
E (529) boot: No bootable app partitions in the partition table

06/12/2021  05:42 PM         1,096,624 StupidFun_ToyBase.bin
06/12/2021  05:42 PM        10,517,588 StupidFun_ToyBase.elf
06/12/2021  05:42 PM         6,509,957 StupidFun_ToyBase.map

I am on ESP-IDF v4.3.  Yet this issue says the problem is fixed:
https://github.com/micropython/micropython-esp32/pull/236

### Expected Behavior
ESP32 pico board is flashed with my firmware.

### Actual Behavior
E (527) esp_image: Image length 1096624 doesn't fit in partition length 1048576
E (527) boot: Factory app partition is not bootable
E (529) boot: No bootable app partitions in the partition table

### Steps to reproduce

1. idf.py build
2. idf.py -p COM4 -b 115200 app-flash

chegewara, 2021-06-13:
You have to edit partition table, because app is bigger than partition.

DerrickLau, 2021-06-13:
That's what I thought, but according to the related issue in micropython that shouldn't be necessary because the flasher should do the preliminary size checks and make room on the partition for the bigger firmware.

chegewara, 2021-06-13:
I dont know micropython and from your description you are using esp-idf not micropython.

DerrickLau, 2021-06-13:
Good point...where is the partition table?  Is it in the ESP-IDF/components directory somewhere?

chegewara, 2021-06-13:
https://www.google.com/search?q=esp32+custom+partition+table&oq=esp32+custom+partition+table&aqs=chrome..69i57.9745j0j1&sourceid=chrome&ie=UTF-8

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#creating-custom-tables

DerrickLau, 2021-06-13:
Thanks.

I have 4MB of SPI flash available on my ESP32 Pico dev board, so I set that in the idf.py menuconfig.

I am trying to build my partition table but it won't.  Is this the correct format for a partition table?  I want to put my firmware in the 1.5MB factory partition:

# Name,   Type, SubType, Offset,   Size
nvs,      data, nvs,     0x9000,  0x3000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
factory,  app,  factory, 0x10000,  1.5M
ota_0,    app,  ota_0,   ,         1M
ota_1,    app,  ota_1,   ,         1M

chegewara, 2021-06-13:
https://raw.githubusercontent.com/espressif/arduino-esp32/master/tools/partitions/default.csv

https://github.com/espressif/arduino-esp32/tree/master/tools/partitions

KonstantinKondrashov, 2021-06-28:
Hi @DerrickLau!
Could you please close this issue if it has been resolved? Thank you.
Seems your partition table should look like this:
```
nvs, data, nvs, , 0x3000
otadata, data, ota,, 0x2000
phy_init, data, phy,, 0x1000
factory, app, factory, , 0x140000
ota_0, app, ota_0, , 0x140000
ota_1, app, ota_1, , 0x140000
```
Also, I can suggest omitting the Offsets, in this case, CONFIG_PARTITION_TABLE_OFFSET can help to move the whole table to the new offset without changing the csv file.

DerrickLau, 2021-06-29:
I tried the above and get this in my idf.py -p COM4 -b 115200 monitor:

I (112) boot: Chip Revision: 1
I (131) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (79) boot: ESP-IDF v4.0.1-dirty 2nd stage bootloader
I (79) boot: compile time 19:37:16
I (80) boot: Enabling RNG early entropy source...
I (85) boot: SPI Speed      : 40MHz
I (89) boot: SPI Mode       : DIO
I (93) boot: SPI Flash Size : 2MB
I (97) boot: Partition Table:
I (101) boot: ## Label            Usage          Type ST Offset   Length
I (108) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (116) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (124) boot:  2 factory          factory app      00 00 00010000 00100000
I (131) boot: End of partition table
I (135) boot_comm: chip revision: 1, min. application chip revision: 0
I (143) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2aa80 (174720) map
E (214) esp_image: invalid segment length 0x4008c000
0x4008c000: get_rate_target_power at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:1829

E (214) boot: Factory app partition is not bootable
E (214) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57

Why is that?  Now, when I look at the sample above, is 0x140000 the size of the partition in bytes, represented as hexadecimal?  Just trying to understand so I can make my own file.  Thanks!

KonstantinKondrashov, 2021-06-29:
@DerrickLau 
Yes, when the size is represented as hexadecimal it means size in bytes. You can create your own partition table 

(Top) → Partition Table
      Partition Table (**Custom partition table CSV**) 
     **(partitions.csv) Custom partition CSV file**

You can try to run the `image_info` cmd to make sure that the bin file is well. It does the same as in `esp_image:segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2aa80 (174720) map ...`
esptool.py -c esp32 image_info build/blink.bin 
esptool.py v3.2-dev
Image version: 1
Entry point: 40080fc0
7 segments

Segment 1: len 0x07e5c load 0x3f400020 file_offs 0x00000018 [DROM]
Segment 2: len 0x022d0 load 0x3ffb0000 file_offs 0x00007e7c [BYTE_ACCESSIBLE,DRAM]
Segment 3: len 0x05ebc load 0x40080000 file_offs 0x0000a154 [IRAM]
Segment 4: len 0x15de4 load 0x400d0020 file_offs 0x00010018 [IROM]
Segment 5: len 0x047a8 load 0x40085ebc file_offs 0x00025e04 [IRAM]
Segment 6: len 0x00010 load 0x50000000 file_offs 0x0002a5b4 [RTC_DATA]
Segment 7: len 0x059fc load 0x00000000 file_offs 0x0002a5cc [PADDING]
Checksum: bd (valid)
Validation Hash: bb833d74b4714f86b861b55a2b21ba3975f76c9f2d47290056934525db16f8e4 (valid)

DerrickLau, 2021-07-04:
OK...I am able to flash my partition table onto my esp32 dev board.  Thanks for your help!

closed 2021-07-04 by DerrickLau