espressif/esp-idf #7145: Missing preprocessor condition in bta_hf_client_co.c

spleen1981, 2021-06-13:
As per object in v4.3-rc, possibly:
`#if (HFP_DYNAMIC_MEMORY == TRUE)`
is missing before evaluating
`if(bta_hf_client_co_cb_ptr == NULL) {
            return 0;
        }`
in  "[/components/bt/host/bluedroid/btc/profile/std/hf_client/bta_hf_client_co.c:332:12](https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/btc/profile/std/hf_client/bta_hf_client_co.c#L332)", as if HFP_DYNAMIC_MEMORY  is FALSE source will not compile for bta_hf_client_co_cb_ptr has never been defined.

Alvin1Zhang, 2021-06-30:
Thanks for reporting, and sorry for late reply. We had a fix under internal reviewing. The issue will be closed automatically once the fix is available on GitHub. Thanks.

Alvin1Zhang, 2021-07-14:
Thanks for reporting, the fix has been available at https://github.com/espressif/esp-idf/commit/976594b8530998277852d578b608be7573f99445, feel free to reopen. Thanks.

closed 2021-07-14 by Alvin1Zhang