espressif/esp-idf #7151: Project commit: HEAD-HASH-NOTFOUND

Rajssss, 2021-06-14:
## Environment

- Development Kit:      [ESP32-DevKitC (Clone)]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): v4.4.0
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [Yes] Eclipse CDT 2021.03
- Power Supply:         [USB]

## Problem Description
I am trying to include esp rainmaker components in my project, I have cloned the rainmaker repo in a different directory, and copied the directories inside components. But when I tried to build I am getting the below error. I have done this before but haven't got this error. Please help.

### Expected Behavior
Build success with no errors.
### Actual Behavior
CMake Error at build/esp-idf/esp_rainmaker/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):

    /home/rajssss/Documents/Microcontroller/IICDC/smart-watch-work/max30102_test_arduino/build/esp-idf/esp_rainmaker/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  /home/rajssss/Documents/Microcontroller/esp/tools/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  /home/rajssss/Documents/Microcontroller/esp/tools/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  components/esp_rainmaker/CMakeLists.txt:64 (git_describe)

ESP RainMaker Project commit: HEAD-HASH-NOTFOUND
### Steps to reproduce

1. Clone ESP RainMaker master to esp-rainmaker
2. Copy all the folders from esp-rainmaker/components to project_root/components
3. Create a required CMakeLists.txt file 
4. idf.py build

### Code to reproduce this issue
[GIST](https://gist.github.com/Rajssss/e4c58fb99b2241a498e1126544faf349)

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may co

ntain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

nopnop2002, 2021-11-27:
```
$ git clone --recursive https://github.com/espressif/esp-rainmaker.git
$ cd esp-rainmaker
$ cp -r components/ project_root
$ cp -r example/common/ project_root
```

build Cmakefile.txt into project_root

```
$ cat CMakeLists.txt
# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set(RMAKER_PATH ${CMAKE_CURRENT_LIST_DIR})

# Add RainMaker components and other common application components
set(EXTRA_COMPONENT_DIRS ${RMAKER_PATH}/components ${RMAKER_PATH}/common ${RMAKER_PATH}/components/esp-insights/components)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(gpio)
```

ESP-Marius, 2024-08-15:
Since this issue is quite old and do not point to any specific issues with IDF I think I will close it.

If you are still having any issues @Rajssss, feel free to update and reopen this one and ill help take a look

closed 2024-08-15 by ESP-Marius