espressif/esp-idf #7152: URL is erased during unsuccesfull request (error 404)

Lisa999, 2021-06-14:
**Environment**
Module or chip used: [ESP32-WROVERKIT2]
IDF version (run git describe --tags to find it): ESP-IDF v4.2.1
Build System: [Eclipse]
Operating System: [ubuntu]
Using an IDE?: [Yes (Eclipse]
Power Supply: [external 3.3V]
**Problem Description**
URL address is changed during a unsuccesful request (404). During a succesfull request it stays the same.
This is a memory leak because i can no longer free the memory...

**Expected Behavior**
URL address remains the same, content remains the same

**Actual Behavior**
URL address changed, is now empty

**Code to reproduce this issue**
    char buffer[10000] = { 0 }, *url;
    asprintf(&url, "https://www.espressif.com/hi");

    esp_http_client_config_t config = {
        .url = url,
        .timeout_ms 	= 10000,
        .event_handler = _http_event_handler,
	.transport_type = HTTP_TRANSPORT_OVER_SSL,
        .user_data 		= buffer        			// Pass address of local buffer to get response
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_err_t err = esp_http_client_perform(client);

Lisa999, 2021-06-14:
But doing this works:
```
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_http_client_set_method(client, HTTP_METHOD_GET);
    esp_err_t err = esp_http_client_perform(client);
```

negativekelvin, 2021-06-14:
Is there space for a 10k buffer in the task stack?

ESP-YJM, 2021-06-22:
@Lisa999 esp_http_client will not modify the url you malloc. Maybe the url was covered. And for buffer, is it a temporary variable. Maybe auto-recycle when the function return.

mahavirj, 2021-07-08:
@Lisa999 Closing as we suspect issue with test code here, if you are still facing problem please feel free to re-open adding more details with complete example project.

closed 2021-07-08 by mahavirj