espressif/esp-idf #7153: ASSER_WARN with BT SPP

m-bruck, 2021-06-15:
- Development Kit: none
- Module or chip used:  ESP32
- IDF version: v4.3-rc
- Build System: idf.py
- Compiler version: 2020r3
- Operating System: Windows, ESP Command Prompt
- Using an IDE?: No

I get this assertion (It always shows up twice, and has no new line at the end):
ASSERT_WARN(103 23), in lc_task.c at line 8775

This happens while trying to use SPP from an Android 5.0.2 device. The Android code looks like this:

BluetoothDevice dev = bluetoothAdapter.getRemoteDevice("7C:9E:BD:94:51:D6");
UUID uuidSpp = UUID.fromString("00001101-0000-1000-8000-00805F9B34FB");
BluetoothSocket sock = dev.createInsecureRfcommSocketToServiceRecord(uuidSpp);

On the ESP32:
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
esp_bt_controller_init(&bt_cfg);
esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT);
esp_bluedroid_init();
esp_bluedroid_enable();

esp_bt_gap_register_callback(esp_bt_gap_cb);
esp_spp_register_callback(esp_spp_cb);
esp_spp_init(ESP_SPP_MODE_CB);

esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_NONE;
esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));

// callback handlers mostly from one of the examples
static void
esp_spp_cb(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
{
    switch (event)
    {
    case ESP_SPP_INIT_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_INIT_EVT");
        esp_bt_dev_set_device_name("thspp");
        esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
        esp_spp_start_srv(ESP_SPP_SEC_AUTHENTICATE, ESP_SPP_ROLE_SLAVE, 0, SPP_SERVER_NAME);
        break;
    case ESP_SPP_DISCOVERY_COMP_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_DISCOVERY_COMP_EVT");
        break;
    case ESP_SPP_OPEN_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_OPEN_EVT");
        break;
    case ESP_SPP_CLOSE_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_CLOSE_EVT");
        break;
    case ESP_SPP_START_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_START_EVT");
        break;
    case ESP_SPP_CL_INIT_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_CL_INIT_EVT");
        break;
    case ESP_SPP_DATA_IND_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_DATA_IND_EVT len=%u handle=%u",
                 param->data_ind.len, param->data_ind.handle);
        esp_log_buffer_hex("",param->data_ind.data,param->data_ind.len);
        break;
    case ESP_SPP_CONG_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_CONG_EVT");
        break;
    case ESP_SPP_WRITE_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_WRITE_EVT");
        break;
    case ESP_SPP_SRV_OPEN_EVT:
        ESP_LOGW(SPP_TAG, "ESP_SPP_SRV_OPEN_EVT");
        gettimeofday(&time_old, NULL);
        break;
    default:
        break;
    } /* switch */
}

static void
esp_bt_gap_cb(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
    switch(event)
    {
    case ESP_BT_GAP_RMT_SRVCS_EVT:
        ESP_LOGW(SPP_TAG, "ESP_BT_GAP_RMT_SRVCS_EVT");
        break;
    case ESP_BT_GAP_RMT_SRVC_REC_EVT:
        ESP_LOGW(SPP_TAG, "ESP_BT_GAP_RMT_SRVC_REC_EVT");
        break;
    case ESP_BT_GAP_AUTH_CMPL_EVT:{
        if (param->auth_cmpl.stat == ESP_BT_STATUS_SUCCESS) {
            ESP_LOGW(SPP_TAG, "authentication success: %s", param->auth_cmpl.device_name);
            esp_log_buffer_hex(SPP_TAG, param->auth_cmpl.bda, ESP_BD_ADDR_LEN);
        } else {
            ESP_LOGE(SPP_TAG, "authentication failed, status:%d", param->auth_cmpl.stat);
        }
        break;
    }
    case ESP_BT_GAP_PIN_REQ_EVT:{
        ESP_LOGW(SPP_TAG, "ESP_BT_GAP_PIN_REQ_EVT min_16_digit:%d", param->pin_req.min_16_digit);
        if (param->pin_req.min_16_digit) {
            ESP_LOGW(SPP_TAG, "Input pin code: 0000 0000 0000 0000");
            esp_bt_pin_code_t pin_code = {0};
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 16, pin_code);
        } else {
            ESP_LOGW(SPP_TAG, "Input pin code: 1234");
            esp_bt_pin_code_t pin_code;
            pin_code[0] = '1';
            pin_code[1] = '2';
            pin_code[2] = '3';
            pin_code[3] = '4';
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 4, pin_code);
        }
        break;
    }

#if (CONFIG_BT_SSP_ENABLED == true)
    case ESP_BT_GAP_CFM_REQ_EVT:
        ESP_LOGW(SPP_TAG, "ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: %u", param->cfm_req.num_val);
        esp_bt_gap_ssp_confirm_reply(param->cfm_req.bda, true);
        break;
    case ESP_BT_GAP_KEY_NOTIF_EVT:
        ESP_LOGW(SPP_TAG, "ESP_BT_GAP_KEY_NOTIF_EVT passkey:%u", param->key_notif.passkey);
        break;
    case ESP_BT_GAP_KEY_REQ_EVT:
        ESP_LOGW(SPP_TAG, "ESP_BT_GAP_KEY_REQ_EVT Please enter passkey!");
        break;
#endif

    default:
        break;
    }
}

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6653607/sdkconfig.txt)

xiongweichao, 2021-06-30:
Does this warning affect the normal executing of your project?
From my perspective, warning log is just a warning.

m-bruck, 2021-06-30:
I can not connect from that device to SPP. I don't know if the assert condition is the cause for this, but it is the only device that triggers it and it is also the only one that can not connect. Without the code or a warning text it is not clear what it signifies.

The device does successfully go through ESP_BT_GAP_AUTH_CMPL_EVT, so some communication works.

xiongweichao, 2021-07-05:
Hi, @m-bruck 

Can I ask if this phenomenon also occurs when testing with other versions of Android devices?

Thanks

m-bruck, 2021-07-05:
This is the only device/android version that triggers the warning and the only one that fails to open the SPP channel. (We haven't tested this yet on our larger pool of devices though)

What is that warning warning about?

RobBrownNZ, 2021-07-08:
@xiongweichao I have this same problem connecting to ESP32 with a Bluetooth 5.0 USB dongle (Edimax BT-8500). Also there's a similar issue on Stack Overflow - https://stackoverflow.com/questions/67389906/how-do-i-connect-esp32-bluetooth-with-windows-10-using-a-5-0-bluetooth-adapter.
I'm using idf 4.2 and Visual Studio Code.

rokmarko, 2021-11-09:
Hi 

Any news, I still cannot connect to ESP32 with Linux Ubuntu 20.04.

I am also attaching debug log of the command `btmon --write ~/hcitrace.snoop | tee ~/hcitrace.txt`:

> < HCI Command: Create Connection (0x01|0x0005) plen 13    #36 [hci0] 177.326443
        Address: 30:83:98:C7:ED:6A (OUI 30-83-98)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow slave (0x01)
> HCI Event: Command Status (0x0f) plen 4                 #37 [hci0] 177.427904
      Create Connection (0x01|0x0005) ncmd 2
        Status: Success (0x00)
> HCI Event: Connect Complete (0x03) plen 11              #38 [hci0] 178.468052
        Status: Success (0x00)
        Handle: 3
        Address: 30:83:98:C7:ED:6A (OUI 30-83-98)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Suppor.. (0x01|0x001b) plen 2  #39 [hci0] 178.468176
        Handle: 3
> HCI Event: Max Slots Change (0x1b) plen 3               #40 [hci0] 178.471021
        Handle: 3
        Max slots: 5
> HCI Event: Command Status (0x0f) plen 4                 #41 [hci0] 178.472019
      Read Remote Supported Features (0x01|0x001b) ncmd 2
        Status: Success (0x00)
> HCI Event: Read Remote Supported Feat.. (0x0b) plen 11  #42 [hci0] 178.479030
        Status: Success (0x00)
        Handle: 3
        Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV3 packets
          u-law log synchronous data
          A-law log synchronous data
          CVSD synchronous data
          Power control
          Transparent synchronous data
          Flow control lag (most significant bit)
          Broadcast Encryption
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
          Extended SCO link (EV3 packets)
          EV4 packets
          EV5 packets
          AFH capable slave
          AFH classification slave
          LE Supported (Controller)
[... 82 lines cut ...]
        Encryption: Enabled with E0 (0x01)
< HCI Command: Read Encryption Ke.. (0x05|0x0008) plen 2  #59 [hci0] 178.682134
        Handle: 3
> HCI Event: Command Complete (0x0e) plen 7               #60 [hci0] 178.689072
      Read Encryption Key Size (0x05|0x0008) ncmd 2
        Status: Success (0x00)
        Handle: 3
        Key size: 16
< ACL Data TX: Handle 3 flags 0x00 dlen 10                #61 [hci0] 178.689134
      L2CAP: Information Request (0x0a) ident 1 len 2
        Type: Extended features supported (0x0002)
< ACL Data TX: Handle 3 flags 0x00 dlen 16                #62 [hci0] 178.689151
      L2CAP: Information Response (0x0b) ident 2 len 8
        Type: Extended features supported (0x0002)
        Result: Success (0x0000)
        Features: 0x000002b8
          Enhanced Retransmission Mode
          Streaming Mode
          FCS Option
          Fixed Channels
          Unicast Connectionless Data Reception
< ACL Data TX: Handle 3 flags 0x00 dlen 12                #63 [hci0] 182.692765
      L2CAP: Connection Request (0x02) ident 2 len 4
        PSM: 3 (0x0003)
        Source CID: 64

BetterJincheng, 2021-11-09:
Hi @rokmarko,

The issue is fixed and not merged right now. See https://github.com/espressif/esp-idf/issues/7763#issuecomment-963945965 for the fixed lib and try again.

Thanks

rokmarko, 2021-11-09:
Hi

Great. I can confirm that connection is now succesfull both with Linux and Android.

Thank you and best regards

closed 2021-11-30 by Alvin1Zhang