espressif/esp-idf #7154: No valid images after changing partition table offset

ZweiEuro, 2021-06-15:
## Environment

- Development Kit:    ESP32-Wrover-Kit
- Kit version : ?
- Module or chip used:  ESP32-WROVERE
- IDF version : .40200.210118 (4.2.0) 
- Build System:  idf.py / pio
- Operating System:     Linux
- Using an IDE?: Yes, VSCode with pio plugin (and package platformio installed for use in cli)
- Power Supply:        external 5V

## Problem Description
I first thought my problem is with the PIO IDE so i posted some information there: 
https://community.platformio.org/t/how-to-compile-against-esp32-wrover-e/21920/12

In short: 
When trying to get PSRAM running i kept getting an error that the partition table was overlapping at 0x8000, after a lot of trying around I found out that the size of the bootloader was too big and didn't fit between 0x1000 and 0x8000. 
I am not sure why it grew in size but okay. 
After adding `CONFIG_PARTITION_TABLE_OFFSET=0x9000` to my sdk defaults and removing all the offsets from the partition table it uploaded successfully.
Sadly then it kept starting from the ota partition which had some old firmware on it and crashing in 
`packages/framework-espidf/components/app_update/esp_ota_ops.c", line 542
`
I just thought something has to be wrong there since i changed the partition table and the ota shouldn't even have any code on it right now.

After running `pio run --target erase` and erasing the flash and uploading it again later on, i am now stuck on an infinite loop telling me neither of my app partitions are bootable, and I have no clue why.

### Expected Behavior
That the app i just uploaded to the chip is bootable

### Actual Behavior

boot loop:
`I (14) boot: ESP-IDF 3.40200.210118 2nd stage bootloader
I (14) boot: compile time 15:42:14
I (14) boot: chip revision: 3
I (17) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (24) boot.esp32: SPI Speed      : 40MHz
I (28) boot.esp32: SPI Mode       : DIO
I (33) boot.esp32: SPI Flash Size : 4MB
I (37) boot: Enabling RNG early entropy source...
I (43) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (61) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (69) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  3 factory          factory app      00 00 00010000 001db000
I (83) boot:  4 ota_0            OTA app          00 10 001f0000 001db000
I (91) boot:  5 storage          Unknown data     01 82 003cb000 0001e000
I (98) boot: End of partition table
I (103) boot: Defaulting to factory image
E (107) esp_image: image at 0x10000 has invalid magic byte
W (113) esp_image: image at 0x10000 has invalid SPI mode 255
W (120) esp_image: image at 0x10000 has invalid SPI size 15
E (126) boot: Factory app partition is not bootable
E (132) esp_image: image at 0x1f0000 has invalid magic byte
W (138) esp_image: image at 0x1f0000 has invalid SPI mode 255
W (144) esp_image: image at 0x1f0000 has invalid SPI size 15
E (151) boot: OTA app partition slot 0 is not bootable
E (156) boot: No bootable app partitions in the partition table`

### Steps to reproduce

described above.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6656145/sdkconfig.txt)
[sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/6656147/sdkconfig.defaults.txt)

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

chegewara, 2021-06-15:
It is because platformio is using hardcoded `0x10000` flash offset for app binary. You can change offset in partition table and bootloader will try to find app at new offsets, but binary has been flashed at 0x10000.

Bootloader size usually is increased when its debug level is increased.

ZweiEuro, 2021-06-15:
Okay so it happens because of the hard coding, so  i can just rearrange my partition table and this should be fixed:

```

# Name,   Type, SubType,  Offset,   Size,  Flags
nvs,      data, nvs,      ,  0x4000
otadata,  data, ota,      ,  0x2000
factory,  app,  factory,  0x10000,  1900k
ota_0,    app,  ota_0,    ,  1900k
storage,  data, spiffs,   ,  120k
phy_init, data, phy,      ,  0x1000

```

Thank you! now my issue is gone but it still crashes, now on some invalid memory region inside nvs init, but I don't think that's related to this.

I am guessing i did something illegal in my new partition table

closed 2021-06-15 by ZweiEuro