espressif/esp-idf #7157: Bug in esp_http_client_fetch_headers

dannybackx, 2021-06-16:
## Environment

- Development Kit:      any esp32
- Kit version (for WroverKit/PicoKit/DevKitC): any
- Module or chip used:  any
- IDF version (run ``git describe --tags`` to find it): all v3.x and v4.x versions (I think)
- Build System:         any
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): any
- Operating System:    N/A (Linux)
- (Windows only) environment type: N/A
- Using an IDE?: N/A (No)
- Power Supply:    N/A (USB)

## Problem Description
esp_http_client_fetch_headers has a bug. It calls http_parser_execute with incomplete headers, and when it does that, the latter will call the on_header_value functions with erroneous data.

To make this concrete :
- I'm requesting stuff from the letsencrypt.org
- on some requests, the set of headers exceeds 512 bytes
- it regularly cuts the line with the nonce in half (so my code picks up a short nonce, and further communication is not accepted)

Workaround is possible by setting a larger value than by default :
httpc.buffer_size = 1024; // HACK to ensure the nonce header doesn't get cut in two
but that's arbitrary : the same could still happen.

I propose a fix like below.

Here's a diff for esp-idf-v4.2.1/components/esp_http_client/esp_http_client.c :
```
hp: {16} diff esp_http_client.c.orig esp_http_client.c
1005a1006,1012
> int _header_end(const char *ptr, const int len) {
>   for (int i=0; i<len-3; i++)
>     if (ptr[i] == 0x0D && ptr[i+1] == 0x0A && ptr[i+2] == 0x0D && ptr[i+3] == 0x0A)
>       return i+3;
>   return -1;
> }
> 
1015a1023,1024
> #if 0
>     // Original code calls http_parser_execute with a buffer that may end on an incompletely read line
1022a1032,1051
> #else
>     // Original code calls http_parser_execute with a buffer that may end on an incompletely read line
>     // In this version, detect incomplete header, allocate more memory, only call http_parser when that's ok.
>     int off = 0;
>     buffer->len = esp_transport_read(client->transport, buffer->data, client->buffer_size_rx, client->timeout_ms);
>     int eoh;
> 
>     while ((eoh = _header_end(buffer->data, buffer->len)) < 0) {
>       off = buffer->len;
>       buffer->data = realloc(buffer->data, buffer->len + 200);
> 
>       int r = esp_transport_read(client->transport, buffer->data+off, 200, client->timeout_ms);
>       if (r < 0) {
>       return ESP_FAIL;
>       }
>       buffer->len += r;
>     }
> 
>     http_parser_execute(client->parser, client->parser_settings, buffer->data, buffer->len);
> #endif
hp: {17}

```

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce
### Code to reproduce this issue
Example code in https://sourceforge.net/p/esp32-acme-client/code/HEAD/tree/trunk/ .

ESP-YJM, 2021-07-09:
@dannybackx Sorry for later reply. I understand what you describe. I admit that parse header will fail if the header exceeds 512 bytes. But it's difficult to add header check in `esp_http_client` component.  Because header check belongs to http layer logic. So i suggestion that you can configure the header size by `esp_http_client_config_t client = {
.buffer_size = 2048，
}` to match the http server response header size which maybe not change.

dannybackx, 2021-07-09:
I'm not happy with that reply, for several reasons :

- Your assumption that a given server always responds with a similar set 
of headers is false, especially for the ACME servers that I work with.

- The esp_http code that I talk about has a default allocation size (in 
include/esp_http_client.h) which causes the problem. It also has the 
buggy function that I point to. So why would the solution be elsewhere.

- The bug, and the functionality available, do not allow application 
code to detect the problem and work around it. (Once you've hit it, 
already have a problem you can't fix without digging into 
esp_http_client internals from the application layer.)

I'm not claiming my solution is the right one. But you have a bug and 
it's in that layer, so the fix should be as well.

     Danny

dannybackx, 2021-07-09:
Trying to analyze further. A better implementation could :

1. initially allocate a smaller buffer than 512, but with a maximum 
settable by the app

2. fail if the maximum is hit and not all headers have been processed

3. keep allocating until malloc fails if no maximum is set

IMO my code comes close to this, just needs to add stuff for the maximum.

     Danny

dannybackx, 2021-07-10:
Something like this :

```
*** include/esp_http_client.h.orig      2021-07-10 10:44:56.153994676 +0200
--- include/esp_http_client.h   2021-07-10 10:48:13.956594505 +0200
***************
*** 24,30 ****
  extern "C" {
  #endif

! #define DEFAULT_HTTP_BUF_SIZE (512)

  typedef struct esp_http_client *esp_http_client_handle_t;
  typedef struct esp_http_client_event *esp_http_client_event_handle_t;
--- 24,32 ----
  extern "C" {
  #endif

! #define DEFAULT_HTTP_BUF_SIZE (256)
! #define HTTP_BUF_SIZE_INCREMENT (200)
! #define HTTP_BUF_SIZE_MAX (-1)

  typedef struct esp_http_client *esp_http_client_handle_t;
  typedef struct esp_http_client_event *esp_http_client_event_handle_t;
*** esp_http_client.c.orig      2021-06-15 14:26:40.112239359 +0200
--- esp_http_client.c   2021-07-10 11:01:09.235131104 +0200
***************
*** 120,125 ****
--- 120,127 ----
      int                         header_index;
      bool                        is_async;
      esp_transport_keep_alive_t  keep_alive_cfg;
+     int                               buffer_size_max;
+     int                               buffer_size_increment;
  };

  typedef struct esp_http_client esp_http_client_t;
***************
*** 366,371 ****
--- 368,376 ----
          client->buffer_size_tx = DEFAULT_HTTP_BUF_SIZE;
      }

+     client->buffer_size_max = HTTP_BUF_SIZE_MAX;
+     client->buffer_size_increment = HTTP_BUF_SIZE_INCREMENT;
+ 
      if (client->max_redirection_count == 0) {
          client->max_redirection_count = DEFAULT_MAX_REDIRECT;
      }
***************
*** 1003,1008 ****
--- 1008,1020 ----
      return ESP_OK;
  }

+ static int esp_http_client_header_end(const char *ptr, const int len) {
+   for (int i=0; i<len-3; i++)
+     if (ptr[i] == 0x0D && ptr[i+1] == 0x0A && ptr[i+2] == 0x0D && ptr[i+3] == 0x0A)
+       return i+3;
+   return -1;
+ }
+ 
  int esp_http_client_fetch_headers(esp_http_client_handle_t client)
  {
      if (client->state < HTTP_STATE_REQ_COMPLETE_HEADER) {
***************
*** 1013,1018 ****
--- 1025,1032 ----
      esp_http_buffer_t *buffer = client->response->buffer;
      client->response->status_code = -1;

+ #if 0
+     // Original code calls http_parser_execute with a buffer that may end on an incompletely read line
      while (client->state < HTTP_STATE_RES_COMPLETE_HEADER) {
          buffer->len = esp_transport_read(client->transport, buffer->data, client->buffer_size_rx, client->timeout_ms);
          if (buffer->len <= 0) {
***************
*** 1020,1025 ****
--- 1034,1064 ----
          }
          http_parser_execute(client->parser, client->parser_settings, buffer->data, buffer->len);
      }
+ #else
+     // Original code calls http_parser_execute with a buffer that may end on an incompletely read line
+     // In this version, detect incomplete header, allocate more memory, only call http_parser when that's ok.
+     int off = 0;
+     buffer->len = esp_transport_read(client->transport, buffer->data, client->buffer_size_rx, client->timeout_ms);
+     int eoh;
+ 
+     while ((eoh = esp_http_client_header_end(buffer->data, buffer->len)) < 0) {
+       off = buffer->len;
+       // If we need to allocate more, check if we're allowed to. If not, fail.
+       if ((client->buffer_size_max > 0) && (buffer->len + client->buffer_size_increment > client->buffer_size_max)) {
+         return ESP_FAIL;
+       }
+ 
+       buffer->data = realloc(buffer->data, buffer->len + client->buffer_size_increment);
+ 
+       int r = esp_transport_read(client->transport, buffer->data+off, client->buffer_size_increment, client->timeout_ms);
+       if (r < 0) {
+       return ESP_FAIL;
+       }
+       buffer->len += r;
+     }
+ 
+     http_parser_execute(client->parser, client->parser_settings, buffer->data, buffer->len);
+ #endif
      ESP_LOGD(TAG, "content_length = %d", client->response->content_length);
      if (client->response->content_length <= 0) {
          client->response->is_chunked = true;
***************
*** 1028,1033 ****
--- 1067,1086 ----
      return client->response->content_length;
  }

+ // Set max buffer size, return previous value
+ int esp_http_client_set_buffer_size_max(esp_http_client_handle_t client, int len) {
+   int prev = client->buffer_size_max;
+   client->buffer_size_max = len;
+   return prev;
+ }
+ 
+ // Set buffer size increment, return previous value
+ int esp_http_client_set_buffer_size_increment(esp_http_client_handle_t client, int len) {
+   int prev = client->buffer_size_increment;
+   client->buffer_size_increment = len;
+   return prev;
+ }
+ 
  static esp_err_t esp_http_client_connect(esp_http_client_handle_t client)
  {
      esp_err_t err;

```

ESP-YJM, 2021-07-12:
@dannybackx Sorry, I may not have carefully looked at the code structure and manual verification before. I think the problem you mentioned is not a problem. IDF should already support it. What version are you using? Could you use the latest master branch for verification?

dannybackx, 2021-07-12:
The problem is in esp_http_client_fetch_headers because it calls http_parser_execute with incomplete data, so it takes the wrong actions.
I am using esp-idf 4.2.1 and have just verified also the latest github version. The code is unchanged, so the bug is still there :

```
    while (client->state < HTTP_STATE_RES_COMPLETE_HEADER) {
        buffer->len = esp_transport_read(client->transport, buffer->data, client->buffer_size_rx, client->timeout_ms);
        if (buffer->len <= 0) {
            return ESP_FAIL;
        }
        http_parser_execute(client->parser, client->parser_settings, buffer->data, buffer->len);
    }
```
Without adding additional allocation at this level, the bug is present.

ESP-YJM, 2021-07-13:
@dannybackx This will read header data util the client state change to HTTP_STATE_RES_COMPLETE_HEADER. Although the header read once is uncompletely,it will read again util read all header data.

AxelLin, 2021-07-13:
@ESP-YJM 

It was mentioned in the original bug report:
esp_http_client_fetch_headers has a bug. It calls http_parser_execute with incomplete headers, and when it does that, the latter will call the on_header_value functions with erroneous data.

The http_parser_execute() is called **within** the while (client->state < HTTP_STATE_RES_COMPLETE_HEADER){} loop.
Are you sure there is no problem if http_parser_execute() is called with incomplete headers?
e.g. It recognized a header with a truncated value.

ESP-YJM, 2021-07-13:
@AxelLin Yes, i decrease the `buffer_size` to 20 in `esp_http_client_config` to receive http header. It will be divided into many segments and call `http_parser_execute` multi times. But it will be combined internal with `http_utils_append_string` .

dannybackx, 2021-07-13:
Hmm I may have reacted too harshly earlier, there appear to be changes in places I didn't look.

Can you explain if or how the new code catches the cases where a line is partially read with two examples :
- header name incomplete (e.g. "Replay-N" instead of "Replay-Nonce")
- value incomplete (e.g. only first 10 chars read)

A real line which I saw being cut up in several places looked like : 
Replay-Nonce : 00017rl2IdSp4U1wx3W20H4ATIrTriV7jbH2jz0GBf66icM

Thanks,
   Danny

AxelLin, 2021-07-14:
@dannybackx 
I think the fix was from below:
https://github.com/espressif/esp-idf/issues/6157#issuecomment-781506761
https://github.com/espressif/esp-idf/commit/308c31e2f137e622714d700df88fb8a2b0a31cec
Please check if you can still hit the issue in master tree.

@ESP-YJM 
The fix is not in v4.3, please take a look. Thanks.

ESP-YJM, 2021-07-15:
@AxelLin Thanks for your kindly remind. I will check it in every release/v4.x branch.

dannybackx, 2021-07-15:
Thanks, submitting it for the latest 3.x branch would be great also.
I know I was (am) still using 3.x, mostly because you can't use 4.x with make.

ESP-YJM, 2021-07-15:
Hi @dannybackx, Sure, we will backport this to release/v3.3 branch. Could you please cherry-pick the commit [308c31e](https://github.com/espressif/esp-idf/commit/308c31e2f137e622714d700df88fb8a2b0a31cec) to your branch and check it.

dannybackx, 2021-07-15:
Testing this with my application is not easy but it does appear to work.

Alvin1Zhang, 2021-08-04:
Thanks for reporting, fix on release/4.3 is https://github.com/espressif/esp-idf/commit/8f75f93c14095e43b37a4e0075176340df4a8c02, we are back porting the fix until release/3.3, will keep updated.

Alvin1Zhang, 2021-08-11:
Thanks for reporting.

- Fix on release/4.3 is https://github.com/espressif/esp-idf/commit/8f75f93c14095e43b37a4e0075176340df4a8c02;
- Fix on release/4.2 is https://github.com/espressif/esp-idf/commit/88b3e0d8d8af4cea5edf7c365f96a8fe9b5436b0;
- Fix on release/4.0 is https://github.com/espressif/esp-idf/commit/40cf3c5e9977ba3eba784223d00917d73a5e7510;
- Fix on release/3.3 is https://github.com/espressif/esp-idf/commit/8f0bcc570a4a2b1447b04a9e75992387e990fb76.

Will update once fix on release/4.1 synchronizes to GitHub.

Alvin1Zhang, 2021-08-31:
Fix on release/4.1 is available https://github.com/espressif/esp-idf/commit/99c1ef8a320135ffa0c431cc56b4d0512920b8c4#diff-412ae49adbb93f62f94a0a758f8c2de821caf83f51ab0a94198832a4831c4d23, feel free to reopen.

closed 2021-08-31 by Alvin1Zhang