espressif/esp-idf #7158: The `esp_eth_start` function starts negotiation before checking link status

b1ackviking, 2021-06-16:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-S2 + KSZ8851SNL Ethernet module
- IDF version: v4.4-dev-1594-g1d7068e4b
- Build System: idf.py
- Compiler version: xtensa-esp32s2-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 3.3V and 5V

## Problem Description

If I start the Ethernet driver when the cable is unplugged, I receive the following warning from the KSZ8851SNL driver: `ksz8851snl-phy: Ethernet PHY auto negotiation timeout`. The negotiation obviously fails, but it should not be started in the first place: the cable is not plugged (the link is DOWN).

Here is [the commit that introduced the problem](https://github.com/espressif/esp-idf/commit/0c25793b0663f9ae8ad70713d5898a9ba61a743f).

### Expected Behavior

No warnings. Negotiation starts when I plug the Ethernet cable, and link status changes to UP.

### Actual Behavior

The `esp_eth_start` does not reset the PHY but starts negotiation before checking if the link is UP or DOWN.

### Steps to reproduce

1. Set up the latest ESP-IDF version
1. Set up the test rig: it is possible to use the ESP32-S2-saola-1 and the KSZ8851SNL-eval boards, which need to connect power, SPI, and INTR signals
1. Start the Ethernet driver without plugging in the Ethernet cable

### Code to reproduce this issue

The following snippet is a basic example of the KSZ8851SNL driver initialization.

```cpp
#include <array>
#include <cstdint>
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_eth.h>
#include <esp_event.h>
#include <esp_log.h>

void InitEthernet()
{
  constexpr auto SPI_HOST = SPI2_HOST;
  constexpr auto MOSI = GPIO_NUM_10;
  constexpr auto MISO = GPIO_NUM_33;
  constexpr auto SCK = GPIO_NUM_5;
  constexpr auto CS = GPIO_NUM_15;

  spi_bus_config_t buscfg{};
  buscfg.mosi_io_num   = MOSI;
  buscfg.miso_io_num   = MISO;
  buscfg.sclk_io_num   = SCK;
  buscfg.quadwp_io_num = -1;
  buscfg.quadhd_io_num = -1;

  ESP_ERROR_CHECK(spi_bus_initialize(SPI_HOST, &buscfg, SPI_HOST));
  spi_device_interface_config_t devcfg{};
  devcfg.clock_speed_hz = SPI_MASTER_FREQ_40M;
  devcfg.spics_io_num   = CS;
  devcfg.queue_size     = 1;

  spi_device_handle_t pSpiHandle{};
  ESP_ERROR_CHECK(spi_bus_add_device(SPI_HOST, &devcfg, &pSpiHandle));
  const eth_ksz8851snl_config_t kszConfig = ETH_KSZ8851SNL_DEFAULT_CONFIG(pSpiHandle);

  const eth_mac_config_t macConfig = ETH_MAC_DEFAULT_CONFIG();
  esp_eth_mac_t* pMac = esp_eth_mac_new_ksz8851snl(&kszConfig, &macConfig);

  eth_phy_config_t phyConfig = ETH_PHY_DEFAULT_CONFIG();
  phyConfig.phy_addr = 1;
  phyConfig.reset_gpio_num = -1;
  esp_eth_phy_t* pPhy = esp_eth_phy_new_ksz8851snl(&phyConfig);

  const esp_eth_config_t config = ETH_DEFAULT_CONFIG(pMac, pPhy);
  esp_eth_handle_t pEthHandle{};
  ESP_ERROR_CHECK(esp_eth_driver_install(&config, &pEthHandle));

  // TODO: 02:00:00 is a Locally Administered OUI range so
  // should not be used except when testing on a LAN under your control.
  std::array<std::uint8_t, MAC_ADDR_LEN> macAddr{ 0x02, 0x00, 0x00, 0x12, 0x34, 0x56 };
  ESP_ERROR_CHECK(esp_eth_ioctl(pEthHandle, ETH_CMD_S_MAC_ADDR, macAddr.data()));

  const esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
  esp_netif_t* pEthNetif = esp_netif_new(&cfg);
  ESP_ERROR_CHECK(esp_eth_set_default_handlers(pEthNetif));

  ESP_ERROR_CHECK(esp_netif_attach(pEthNetif, esp_eth_new_netif_glue(pEthHandle)));
  ESP_ERROR_CHECK(esp_eth_start(pEthHandle));
}
```

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

The minimal sdkconfig.

```
CONFIG_ETH_ENABLED=y
CONFIG_ETH_USE_SPI_ETHERNET=y
CONFIG_ETH_SPI_ETHERNET_KSZ8851SNL=y
```

suda-morris, 2021-06-16:
Does change the `ESP_LOGW` to `ESP_LOGD` OK to you?

b1ackviking, 2021-06-16:
I'm not sure I understand the logic behind replacing the PHY reset with just a new negotiation attempt. But even if it is required to be that way, the negotiation does not make any sense when the link is down.

So instead of hiding the log output, I would rather suggest checking the link status before starting the negotiation.

For the KSZ8851SNL in particular, it can be achieved with the following code:

```
static esp_err_t ksz8851_get_link_status(phy_ksz8851snl_t *ksz8851, eth_link_t *link)
{
    ESP_GOTO_ON_FALSE(eth, ESP_ERR_INVALID_ARG, err, TAG, "mediator can not be null");
    ESP_GOTO_ON_FALSE(link, ESP_ERR_INVALID_ARG, err, TAG, "link can not be null");

    esp_err_t ret = ESP_OK;
    esp_eth_mediator_t *eth = ksz8851->eth;

    uint32_t status;
    ESP_GOTO_ON_ERROR(eth->phy_reg_read(eth, ksz8851->addr, KSZ8851_P1SR, &status), err, TAG, "P1SR read failed");
    *link = (status & P1SR_LINK_GOOD) ? ETH_LINK_UP : ETH_LINK_DOWN;
    return ESP_OK;
err:
    return ret;
}
```

suda-morris, 2021-06-17:
@b1ackviking  Thanks for your report, we perform a new negotiation in `esp_eth_start` instead of resetting the PHY:
1. we have already reset the PHY in `esp_eth_driver_install`
2. we want to speed up obtaining the IP address for the first time (a reset can somehow slow that down on some PHY chip)

But I agree to check the link status first before we start a new negotiation.

b1ackviking, 2021-06-17:
@suda-morris OK, I see. Thank you!

Alvin1Zhang, 2021-08-30:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/22dddb46d81809a9f9c70d8354334ae893141e2a, feel free to reopen.

closed 2021-08-30 by Alvin1Zhang