espressif/esp-idf #7159: WiFi problems after switching to ESP32-WROOME chip

ZweiEuro, 2021-06-16:
## Environment

- Development Kit:    ESP32-Wrover-Kit
- Kit version : ?
- Module or chip used:  ESP32-WROVERE
- IDF version : .40200.210118 (4.2.0) 
- Build System:  idf.py / pio
- Operating System:     Linux
- Using an IDE?: Yes, VSCode with pio plugin (and package platformio installed for use in cli)
- Power Supply:        external 5V

## Problem Description

I've had code running on the ESP32-WROOM-32 but my project got too resource intensive so i switched to the ESP32-WROVER-E.
My previous project used the BT and WiFi functions extensively, especially the sockets and the esp-http-client.
As far as i can tell all needed to do (and i did) was change the sdkconfig (attached) to activate the additonal RAM to work with the it.
I have also tried to run and the himem example, which works fine.

Bluetooth works fine, but i can't connect to the chip anymore using a tcp python script in the same network, or my phone with a simple tcp socket in the same network.

I can't find much in the documentation about this.
Do i need to allocate everything manually now in the SPIRAM? Shouldn't this just be done automatically ?
I've also had problems starting tasks, maybe because the stack couldn't be allocated? but i am not sure.
I've also had the issue 
From the esp-http-client:
E (16252) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7f00
E (16252) esp-tls: create_ssl_handle failed
E (16252) esp-tls: Failed to open new connection
E (16262) TRANS_SSL: Failed to open a new connection
E (16262) HTTP_CLIENT: Connection failed, sock < 0

 TCP and udp sockets also fail sometimes and behave unstable.
 
 
### Expected Behavior

Working networking

What heavily confuses me is that I can't find any documentation on what to do if i am using this, the himem example just tests the memory regions, and all i can find is that it should just work fine out of the box (replacing malloc and free in the background ) without me needing to do much more.

When I activated the RAM the bootloader also increased in size so i changed the partition table slightly making nvs (which i am not using a lot ) a little bit smaller. Is there a minimum size to the nvs that wifi/networking needs ? Couldn't find that in the documentation either.

I don't know if this is a bug, expected behaviour, missing docs or 'me too stupid' and i just can't find the relevant documentation

in the sdk defaults: what i added was the section SPIRAM, mostly copied from the himem example.
[sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/6664591/sdkconfig.defaults.txt)
[partition_table.csv](https://github.com/espressif/esp-idf/files/6664593/partition_table.csv)

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2021-06-16:
https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/mbedtls/Kconfig#L3

ZweiEuro, 2021-06-17:
Okay, so I have to change that setting, I did.

did it in the menuconfig and it added:
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y

This does not solve the problem though, I still can't connect to the chip.

@negativekelvin thanks for pointing out that!

ZweiEuro, 2021-06-17:
@negativekelvin  thank you for the tip! and eventually it wasn't the chips fault but my router was making problems, thanks a lot

closed 2021-06-17 by ZweiEuro