espressif/esp-idf #7160: Contents of segment at SHA256 digest offset 0xb0 are not all zero when building with protobuf for c++

MarcGeh, 2021-06-16:
## Environment

- Development Kit: [ESP32-Wrover-Kit]
- Kit version: [v3]
- Module or chip used:  [ESP32-WROVER]
- IDF version: v4.2.1
- Build System: [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: [Linux]
- Using an IDE?: [No]
- Power Supply: [USB]

## Problem Description

Hey guys, I've been trying to get protobuf c++ to run on the esp32. I created a protobuf component with all the source files from the protobuf git (version 3.15.2). 

When running 'idf.py build' I get this error:

```
A fatal error occurred: Contents of segment at SHA256 digest offset 0xb0 are not all zero. Refusing to overwrite.
```
I have no idea what is causing this and why it only happens when I compile with the protobuf source files. When I comment out the lines 2056 - 2058 of esp-idf/components/esptool_py/esptool/esptool.py the binary is generated and I can run it on my dev-kit. However, there are some warnings during startup.

Since I really need the c++ implementation of protobuf to run on the esp32 and I have no idea how to fix this issue I thought I might get help here.

I created an example project to reproduce the issue.

### Expected Behavior
Generate a binary image which can be flashed onto the target.

### Actual Behavior
Generating binary image is aborted because of this error:

```
A fatal error occurred: Contents of segment at SHA256 digest offset 0xb0 are not all zero. Refusing to overwrite.
```

### Steps to reproduce

Go and checkout the git repository I created for this: https://github.com/MarcGeh/esp32-cpp-protobuf
and build the project. The log output if the binary is generated with the empty check for the SHA256 digest commented out can be seen in the readme.

o-marshmallow, 2021-06-17:
Hello @MarcGeh !
Thank you for your report, this is a bug that occurs when the alignment of the `text` or `rodata` section is changed because of the presence of an explicit alignment attribute within the code. In your repo, it happens because of the following piece of code:
```
PROTOBUF_FUNC_ALIGN(32)
void* Arena::AllocateAlignedNoHook(size_t n) { 
    return impl_.AllocateAligned(n, nullptr);
}
```
in [arena.cc](https://github.com/MarcGeh/esp32-cpp-protobuf/blob/409d7afeae8ee0f5673ec1de7e5062f2e880f187/components/protobuf/src/google/protobuf/arena.cc#L460)

Which is defined as:
```
// Function alignment attribute introduced in gcc 4.3
#define PROTOBUF_FUNC_ALIGN(bytes) __attribute__((aligned(bytes)))
```
in [port_def.inc file](https://github.com/MarcGeh/esp32-cpp-protobuf/blob/409d7afeae8ee0f5673ec1de7e5062f2e880f187/components/protobuf/src/google/protobuf/port_def.inc#L241)

 Fortunately, this bug has been fixed on `main`, `release/4.3` and `release/v4.2` branches. So you have multiple solutions:
* Switch to `release/4.2` or newer branch
* Wait for a `v4.2.2` release which will include this fix
* Patch locally your v4.2.1 release with the following commands:
```
# Go to ESP-IDF repo and make sure v4.2.1 is checked out
cd /path/to/esp-idf
git checkout v4.2.1

# Fetch the project to be sure to have the fix
git fetch

# Apply the newer commits that fix that bug
git cherry-pick 974b8dd4c40cc38c451089eeb9bf90c6dc22a22d af7b21851e9d5a7d1edc128c04f2e307bd2f9d98

# Go to esptool project to apply its fixes
cd components/esptool_py/esptool
git fetch
git cherry-pick 5619ecd723fd5c09d11d54f4768305ba45a341b0 627222a55d97b9886bc882672f07a5bf58d3f9b4

# You may experience a conflict here, no problem, the file in question is only used for testing, it can be ignored:
git reset test/test_imagegen.py
git checkout test/test_imagegen.py
git cherry-pick --continue

# Go back to your project to compile it 
```

MarcGeh, 2021-06-17:
Hi @o-marshmallow
Thank you very much for the quick response and the clear and easy to understand solution! I can confirm that the binary is now generated after patching my esp-idf v4.2.1 locally like you have described. I will go ahead and close this issue.

closed 2021-06-17 by MarcGeh