espressif/esp-idf #7161: Mysterious Panic with Bluedroid Bluetooth Classic

BenjaminWKidd, 2021-06-16:
Hi all,

I have a bug/issue/i'm not sure what that occurs intermittently when running the ESP32-WROOM-32D in bluetooth classic with SPP.

xiongweichao, 2021-06-22:
Hi @BenjaminWKidd 

Could you provide following info:
  - whole log (better provide with attachment)
  - IDF commit ID
  - elf (in the build folder)
  
Thanks

BenjaminWKidd, 2021-06-22:
Full log and ELF file (with extension changed to .txt) are attached.

[bt_spp_acceptor_demo.elf.txt](https://github.com/espressif/esp-idf/files/6694137/bt_spp_acceptor_demo.elf.txt)

[putty_esp_20210517_COM16.log](https://github.com/espressif/esp-idf/files/6694116/putty_esp_20210517_COM16.log)

I'm not exactly sure how to obtain my commit ID for IDF - could you point me to where I can get that? My version is "v4.2-dirty"

Thank you!

xiongweichao, 2021-06-30:
Hi, @BenjaminWKidd 

Can you try the bt-lib (with extension changed to .txt) attached below and check if it solves your issue?
Replace the file at $IDF_PATH/components/bt/controller/lib/libbtdm_app.a with this file.
 Let me know if it solves the issue.

Thanks

[libbtdm_app.a.txt](https://github.com/espressif/esp-idf/files/6739014/libbtdm_app.a.txt)

BenjaminWKidd, 2021-07-01:
Thank you. I've overwritten the file and rebuilt my code and will be trying it out overnight.

BenjaminWKidd, 2021-07-06:
Hi xiongweichao,

We ran the unit with the updated libbtdm_app.a file over the last 4 days and we had one panic. There is some interesting debug output that begins

[07/04/2021 05:13:13 PM] E8192
[07/04/2021 05:13:13 PM] FEX 0 8
[07/04/2021 05:13:15 PM] FEX 0 8

and this "FEX 0 8" appears to keep occuring once per second up to the panic event.

Furthermore, there are a lot of error messages and disconnects interspersed throughout the debug log such as:

[0;31mE (182771006) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3[0m

 BT_APPL: bta_dm_pm_btm_status hci_status=34[0m
[0;33mW (182791106) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)[0m
[0;32mI (182791116) ESP_SPP_CLOSE_EVT: Connection Closed[0m

The actual panic occurs about 18 minutes after the whole "FEX 0 8" thing starts happening. 

[07/04/2021 05:31:16 PM] FEX 0 8
[07/04/2021 05:31:16 PM] ASSERT_PARAM(1 1), in ld_fm.c at line 649
[07/04/2021 05:31:17 PM] Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x400832cc  PS      : 0x00060834  A0      : 0x80085435  A1      : 0x3ffbe300  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000  
A6      : 0x00000004  A7      : 0x3ffbdbfc  A8      : 0x800832cc  A9      : 0x3ffbe2e0  
A10     : 0x00000000  A11     : 0x0000002a  A12     : 0x00000014  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40083204  LEND    : 0x4008320b  LCOUNT  : 0x00000000  
Core  0 was running in ISR context:
EPC1    : 0x401a50b7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400832cc

[07/04/2021 05:31:17 PM] Backtrace:0x400832c9:0x3ffbe300 0x40085432:0x3ffbe320 0x4008572d:0x3ffbe340 0x40088f2e:0x3ffbe370 0x40089847:0x3ffbe390 0x400829b5:0x3ffbe3b0 0x401a18fb:0x3ffbbd10 0x400d492e:0x3ffbbd30 0x40090d21:0x3ffbbd50 0x40090395:0x3ffbbd70
Core  1 register dump:
PC      : 0x401a18fe  PS      : 0x00060334  A0      : 0x800d4931  A1      : 0x3ffbc470  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d4882  A9      : 0x3ffbc440  
A10     : 0x00000000  A11     : 0x00060823  A12     : 0x00060820  A13     : 0x00060823  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401a18fb:0x3ffbc470 0x400d492e:0x3ffbc490 0x40090d21:0x3ffbc4b0 0x40090395:0x3ffbc4d0
[07/04/2021 05:31:17 PM] 

ELF file SHA256: f638bf7c96c2e988

Rebooting...
ets Jun  8 2016 00:22:57

I've attached the full debug log in case that helps. Thank you for your help.

[Debug Log after testing libbtdm_app file change 7-2-2021.txt](https://github.com/espressif/esp-idf/files/6770293/Debug.Log.after.testing.libbtdm_app.file.change.7-2-2021.txt)

BenjaminWKidd, 2021-07-22:
Hi @xiongweichao or anyone else,

We're at a bit of a impasse here without really any solid fixes or work-arounds that we can come up with. Any further suggestions?

BenjaminWKidd, 2021-08-09:
Since I've last posted, I've upgraded to IDF version 4.3, but this issue still persists. If anyone has any suggestions they would be greatly appreciated.

xiongweichao, 2021-08-12:
Hi, @BenjaminWKidd 

Sorry for the late reply.  Can you re-try the bt-lib (with extension changed to .txt) attached below and check if it solves your issue? Replace the file at $IDF_PATH/components/bt/controller/lib/libbtdm_app.a with this file.

Thanks
[libbtdm_app.a.txt](https://github.com/espressif/esp-idf/files/6972663/libbtdm_app.a.txt)

BenjaminWKidd, 2021-08-12:
Ok, finally got my environment re-setup for 4.2.2, and I went ahead and put @xiongweichao 's version of libbtdm_app.a in the \esp-idf-v4.2.2\components\bt\controller\lib\esp32\ folder, where it replaced the original.

The problem i'm getting immediately is: "E (891) AWPL_MODEM: bt_awpl_setup initialize controller failed: ESP_ERR_INVALID_ARG"

Just to make sure, I cleaned the project, and rebuilt with the original version of libbtdm_app.a, and this problem did not occur.

Here is the complete log:

Executing action: monitor
Running idf_monitor in directory c:\dropbox\embedded\3wf104b_3w_bt_modem\3w_bt_m
odem
Executing "C:\Users\3WBKidd\.espressif\python_env\idf4.2_py3.8_env\Scripts\pytho
n.exe C:/Users/3WBKidd/Desktop/esp-idf-2/tools/idf_monitor.py -p \\.\COM4 -b 115
200 --toolchain-prefix xtensa-esp32-elf- --print_filter  c:\dropbox\embedded\3wf
104b_3w_bt_modem\3w_bt_modem\build\bt_spp_acceptor_demo.elf -m 'C:\Users\3WBKidd
\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe' 'C:\Users\3WBKidd\De
sktop\esp-idf-2\tools\idf.py'"...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7164
ho 0 tail 12 room 4
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (28) boot: ESP-IDF v4.2.2-dirty 2nd stage bootloader
I (29) boot: compile time 11:00:51
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 factory          factory app      00 00 00010000 00280000
I (99) boot:  4 ota_0            OTA app          00 10 00290000 00280000
I (106) boot:  5 ota_1            OTA app          00 11 00510000 00280000
I (114) boot: End of partition table
I (118) boot: Defaulting to factory image
I (123) boot_comm: chip revision: 1, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x32554 (20
6164) map
I (218) esp_image: segment 1: paddr=0x0004257c vaddr=0x3ffbdb60 size=0x04274 ( 1
7012) load
I (225) esp_image: segment 2: paddr=0x000467f8 vaddr=0x40080000 size=0x09820 ( 3
8944) load
0x40080000: _WindowOverflow4 at C:/Users/3WBKidd/Desktop/esp-idf-2/components/fr
eertos/xtensa/xtensa_vectors.S:1730

I (242) esp_image: segment 3: paddr=0x00050020 vaddr=0x400d0020 size=0xd6888 (87
8728) map
0x400d0020: _stext at ??:?

I (577) esp_image: segment 4: paddr=0x001268b0 vaddr=0x40089820 size=0x15534 ( 8
7348) load
0x40089820: r_vhci_isr at ??:?

I (632) boot: Loaded app from partition at offset 0x10000
I (632) boot: Disabling RNG early entropy source...
I (633) cpu_start: Pro cpu up.
I (636) cpu_start: Application information:
I (641) cpu_start: Project name:     bt_spp_acceptor_demo
I (647) cpu_start: App version:      3a7aa86-dirty
I (653) cpu_start: Compile time:     Aug 12 2021 11:00:39
I (659) cpu_start: ELF file SHA256:  6aa9fefb7aadbf67...
I (665) cpu_start: ESP-IDF:          v4.2.2-dirty
I (670) cpu_start: Starting app cpu, entry point is 0x400819e4
0x400819e4: call_start_cpu1 at C:/Users/3WBKidd/Desktop/esp-idf-2/components/esp
32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (681) heap_init: Initializing. RAM available for dynamic allocation:
I (687) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (693) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (700) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (706) heap_init: At 3FFCEBF0 len 00011410 (69 KiB): DRAM
I (712) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (718) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (725) heap_init: At 4009ED54 len 000012AC (4 KiB): IRAM
I (731) cpu_start: Pro cpu start user code
I (749) spi_flash: detected chip: generic
I (750) spi_flash: flash io: dio
W (750) spi_flash: Detected size(16384k) larger than the size in the binary imag
e header(8192k). Using the size in the binary image header.
I (760) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (771) PrintResetReason: Reset due to power-on event
I (781) app_main(): Firmware Version: 1.14
I (821) ../main/ota.c: Factory App Found
I (821) ../main/ota.c: Executing from factory partition
I (821) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:0
I (831) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:0
I (841) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:0
I (851) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldo
wn: 0| Intr:0
I (861) GPIO_setup(): System restart as a result of 'POWERON' event, trigger res
et of NXP chip
I (871) get_from_nvs(): Opening Non-Volatile Storage (NVS) handle 'Device_Name'.
..
I (871) get_from_nvs(): Reading value from NVS ... 
I (881) get_from_nvs(): psNVSoutString = AWPL146478
I (891) get_from_nvs(): sDeviceName = AWPL146478
E (891) AWPL_MODEM: bt_awpl_setup initialize controller failed: ESP_ERR_INVALID_
ARG

xiongweichao, 2021-08-18:
Hi @BenjaminWKidd ,

Please provide the idf version and commit id you are currently using.

```
I (665) cpu_start: ESP-IDF: v4.2.2-dirty
```

Thanks

BenjaminWKidd, 2021-08-18:
Hi @xiongweichao ,

I'm somewhat ignorant of how to obtain this, but my best guess is 7ab8f793ca5b026f37ae812bcc103e3aa698d164 (I obtained this by running "git log" in the ESP-IDF 4.2 command prompt and then looking at the most recent commit id - not sure if this is the right way to do this, but its all I could find after about 45 minutes of googling.)

The IDF version is as you stated  - 4.2.2.

xiongweichao, 2021-08-19:
Hi @BenjaminWKidd ,

Can you re-try the bt-lib (with extension changed to .txt) attached below and check if it solves your issue?

Thanks
[libbtdm_app.a.txt](https://github.com/espressif/esp-idf/files/7012232/libbtdm_app.a.txt)

BenjaminWKidd, 2021-08-20:
Hi @xiongweichao ,

Thanks. I've updated the library. It does appear to load and allow bluetooth connections now. We'll run this for a bit and let you know what happens.

Here's the log from bootup:

Executing action: monitor
Running idf_monitor in directory c:\dropbox\embedded\3wf104b_3w_bt_modem\3w_bt_m
odem
Executing "C:\Users\3WBKidd\.espressif\python_env\idf4.2_py3.8_env\Scripts\pytho
n.exe C:/Users/3WBKidd/Desktop/esp-idf-2/tools/idf_monitor.py -p \\.\COM4 -b 115
200 --toolchain-prefix xtensa-esp32-elf- --print_filter  c:\dropbox\embedded\3wf
104b_3w_bt_modem\3w_bt_modem\build\bt_spp_acceptor_demo.elf -m 'C:\Users\3WBKidd
\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe' 'C:\Users\3WBKidd\De
sktop\esp-idf-2\tools\idf.py'"...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7164
ho 0 tail 12 room 4
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (28) boot: ESP-IDF v4.2.2-dirty 2nd stage bootloader
I (28) boot: compile time 11:00:51
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 factory          factory app      00 00 00010000 00280000
I (99) boot:  4 ota_0            OTA app          00 10 00290000 00280000
I (106) boot:  5 ota_1            OTA app          00 11 00510000 00280000
I (114) boot: End of partition table
I (118) boot: Defaulting to factory image
I (123) boot_comm: chip revision: 1, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3251c (20
6108) map
I (217) esp_image: segment 1: paddr=0x00042544 vaddr=0x3ffbdb60 size=0x04284 ( 1
7028) load
I (225) esp_image: segment 2: paddr=0x000467d0 vaddr=0x40080000 size=0x09848 ( 3
8984) load
0x40080000: _WindowOverflow4 at C:/Users/3WBKidd/Desktop/esp-idf-2/components/fr
eertos/xtensa/xtensa_vectors.S:1730

I (242) esp_image: segment 3: paddr=0x00050020 vaddr=0x400d0020 size=0xd7264 (88
1252) map
0x400d0020: _stext at ??:?

I (578) esp_image: segment 4: paddr=0x0012728c vaddr=0x40089848 size=0x15614 ( 8
7572) load
0x40089848: vhci_recv at ??:?

I (633) boot: Loaded app from partition at offset 0x10000
I (633) boot: Disabling RNG early entropy source...
I (634) cpu_start: Pro cpu up.
I (637) cpu_start: Application information:
I (642) cpu_start: Project name:     bt_spp_acceptor_demo
I (648) cpu_start: App version:      3a7aa86-dirty
I (654) cpu_start: Compile time:     Aug 20 2021 11:22:56
I (660) cpu_start: ELF file SHA256:  51a5aebb0b7076e1...
I (666) cpu_start: ESP-IDF:          v4.2.2-dirty
I (671) cpu_start: Starting app cpu, entry point is 0x400819e4
0x400819e4: call_start_cpu1 at C:/Users/3WBKidd/Desktop/esp-idf-2/components/esp
32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (682) heap_init: Initializing. RAM available for dynamic allocation:
I (688) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (694) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (700) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (707) heap_init: At 3FFCECA8 len 00011358 (68 KiB): DRAM
I (713) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (719) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (726) heap_init: At 4009EE5C len 000011A4 (4 KiB): IRAM
I (732) cpu_start: Pro cpu start user code
I (750) spi_flash: detected chip: generic
I (751) spi_flash: flash io: dio
W (751) spi_flash: Detected size(16384k) larger than the size in the binary imag
e header(8192k). Using the size in the binary image header.
I (761) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (772) PrintResetReason: Reset due to power-on event
I (782) app_main(): Firmware Version: 1.15
I (822) ../main/ota.c: Factory App Found
I (822) ../main/ota.c: Executing from factory partition
I (822) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:0
I (832) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:0
I (842) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:0
I (852) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldo
wn: 0| Intr:0
I (862) GPIO_setup(): System restart as a result of 'POWERON' event, trigger res
et of NXP chip
I (872) get_from_nvs(): Opening Non-Volatile Storage (NVS) handle 'Device_Name'.
..
I (872) get_from_nvs(): Reading value from NVS ... 
I (882) get_from_nvs(): psNVSoutString = AWPL146478
I (892) get_from_nvs(): sDeviceName = AWPL146478
I (892) BTDM_INIT: BT controller compile version [b195d78]
I (902) system_api: Base MAC address is not set
I (902) system_api: read default base MAC address from EFUSE
I (912) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1532) esp_spp_cb(): Setting BT Device String to AWPL146478
I (1542) AWPL_MODEM: event: 10
I (1542) AWPL_MODEM: Bluetooth Configuration Complete

BenjaminWKidd, 2021-08-23:
Hi @xiongweichao,

After running the code with the modified libbtdm_app.a file you provided, things looked ok for the first 2 hours, but then Bluetooth communication errors started showing up on the client side (beaglebone), and we started noticing the following in the esp debug log (full log is also attached). I'll update you if we have had any "panics" over the weekend as soon as I find out from my customer.

[teraterm_laptop_socket_espv115_test_20210820.log](https://github.com/espressif/esp-idf/files/7032493/teraterm_laptop_socket_espv115_test_20210820.log)

...
[2021-08-20 16:34:06.124] I (8332942) BT DATA: 4 bytes received
[2021-08-20 16:34:06.156] I (8332972) UART DATA: 25 bytes received
[2021-08-20 16:34:15.910] FEX 0 8
[2021-08-20 16:34:17.203] FEX 0 8
[2021-08-20 16:34:17.626] I (8344442) BT DATA: 4 bytes received
[2021-08-20 16:34:17.668] I (8344492) UART DATA: 25 bytes received
[2021-08-20 16:34:18.501] FEX 0 8
[2021-08-20 16:34:19.787] FEX 0 8
[2021-08-20 16:34:21.078] FEX 0 8
[2021-08-20 16:34:22.371] FEX 0 8
[2021-08-20 16:34:23.666] FEX 0 8
[2021-08-20 16:34:24.867] E (8351692) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3
[2021-08-20 16:34:24.958] FEX 0 8
[2021-08-20 16:34:26.251] FEX 0 8
[2021-08-20 16:34:27.542] FEX 0 8
[2021-08-20 16:34:28.835] FEX 0 8
[2021-08-20 16:34:30.131] FEX 0 8
[2021-08-20 16:34:31.420] FEX 0 8
[2021-08-20 16:34:32.712] FEX 0 8
[2021-08-20 16:34:34.005] FEX 0 8
[2021-08-20 16:34:35.304] FEX 0 8
[2021-08-20 16:34:36.590] FEX 0 8
[2021-08-20 16:34:37.880] FEX 0 8
[2021-08-20 16:34:39.173] FEX 0 8
[2021-08-20 16:34:40.472] FEX 0 8
[2021-08-20 16:34:41.765] FEX 0 8
[2021-08-20 16:34:43.054] FEX 0 8
[2021-08-20 16:34:44.343] FEX 0 8
[2021-08-20 16:34:45.643] FEX 0 8
[2021-08-20 16:34:46.930] FEX 0 8
[2021-08-20 16:34:49.515] FEX 0 8
[2021-08-20 16:34:49.764] W (8376592) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
[2021-08-20 16:34:49.780] I (8376592) ESP_SPP_CLOSE_EVT: Connection Closed
[2021-08-20 16:34:50.526] W (8377352) BT_APPL: new conn_srvc id:26, app_id:255
[2021-08-20 16:34:50.550] I (8377372) BT DATA: 4 bytes received
[2021-08-20 16:34:50.588] I (8377412) UART DATA: 25 bytes received
[2021-08-20 16:34:52.106] FEX 0 8
[2021-08-20 16:34:52.119] FEX 0 8
[2021-08-20 16:34:53.450] FEX 0 8
[2021-08-20 16:34:57.325] FEX 0 8
[2021-08-20 16:34:57.344] FEX 0 8
...

BenjaminWKidd, 2021-08-24:
Hi @xiongweichao,

So it looks like all the Bluetooth disconnects and such caused our test script to fail, but we did not see the saw the "panic" and associated restart in the esp logs while the test script was running. The esp continued to print out the "FEX 0 8" message. What should we try next?

Thanks!

xiongweichao, 2021-08-25:
Hi @BenjaminWKidd  ,

Can you try the bt-lib (with extension changed to .txt) attached below and check if it solves your issue?

Thanks

[libbtdm_app.a.txt](https://github.com/espressif/esp-idf/files/7047358/libbtdm_app.a.txt)

BenjaminWKidd, 2021-08-25:
Thanks, We've recompiled with the new copy of libbtdm_app.a you provided and will start testing within the next few hours.

BenjaminWKidd, 2021-08-30:
Hi @xiongweichao 

We ran the logging over the weekend, and we haven't seen any "panic" events, but we are still seeing quite a bit of the error messages as described in the previous post. 

One interesting thing to note, when we do data requests more frequently (every second), we don't notice any problems. When we stretch out that time interval to every 10 seconds, or every 20 seconds, we see error messages and disconnects, usually after about 1.5 hours (in the case of every 20 seconds between data requests) or 2 hours (in the case of every 10 seconds.). This is very strange, as we would expect that more data, not less, would cause problems, but not the opposite. 

Let us know what we should try next.

xiongweichao, 2021-08-31:
Hi @BenjaminWKidd 

How long is the data packet sent each time? I will try to reproduce this issue.

Thanks

BenjaminWKidd, 2021-08-31:
Hi @xiongweichao, The data packet that is sent to the ESP32 via bluetooth is 4 bytes long which the ESP passes to an external processor by uart. The response from the external processor is 25 bytes long.

xiongweichao, 2021-09-01:
Hi @BenjaminWKidd 

Please confirm whether there is a memory leak. You can use `xPortGetFreeHeapSize()` to view the free heap size.

Thanks

xiongweichao, 2021-09-02:
Hi @BenjaminWKidd 

Sorry, I did not reproduce this issue. could you please provide a reproduced demo?

Thanks

BenjaminWKidd, 2021-09-02:
Hi @xiongweichao ,

I've implemented the xPortGetFreeHeapSize() into our code and we'll starting testing it soon. Thank you for the suggestion.

Reproducibility has been one of the central problems of this issue. We've only been able to "reliably" reproduce it at our customer's facility with their equipment, and even then it can be 2 hours or more before and between occurences. I'll see about getting together some code that would be testable by you.

BenjaminWKidd, 2021-09-03:
Hi @xiongweichao ,

For the first two hours of running, we're seeing the heap not change at all - everything in the debug log comes out like this:
...
[2021-09-03 12:19:46.542] I (2985824) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165352
[2021-09-03 12:19:46.550] I (2985824) BT DATA: 4 bytes received
[2021-09-03 12:19:46.580] I (2985854) UART DATA: 25 bytes received
...

However, after about that magical 2-hour mark, we start seeing the free heap size changing:

[2021-09-03 13:02:36.555] I (5555824) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165352
[2021-09-03 13:02:36.561] I (5555824) BT DATA: 4 bytes received
[2021-09-03 13:02:36.603] I (5555874) UART DATA: 25 bytes received
[2021-09-03 13:02:57.049] I (5576324) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165352
[2021-09-03 13:02:57.056] I (5576324) BT DATA: 4 bytes received
[2021-09-03 13:02:57.102] I (5576364) UART DATA: 25 bytes received
[2021-09-03 13:03:17.558] I (5596824) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165352
[2021-09-03 13:03:17.572] I (5596824) BT DATA: 4 bytes received
[2021-09-03 13:03:17.597] I (5596864) UART DATA: 25 bytes received
[2021-09-03 13:03:41.564] I (5620834) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165352
[2021-09-03 13:03:41.577] I (5620834) BT DATA: 4 bytes received
[2021-09-03 13:03:41.606] I (5620874) UART DATA: 25 bytes received
[2021-09-03 13:04:03.058] I (5642324) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165348
[2021-09-03 13:04:03.074] I (5642324) BT DATA: 4 bytes received
[2021-09-03 13:04:03.094] I (5642364) UART DATA: 25 bytes received
[2021-09-03 13:04:10.294] E (5649564) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3
[2021-09-03 13:04:35.197] W (5674464) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
[2021-09-03 13:04:35.231] I (5674474) ESP_SPP_CLOSE_EVT: Connection Closed
[2021-09-03 13:04:37.263] W (5676534) BT_APPL: new conn_srvc id:26, app_id:255
[2021-09-03 13:04:37.285] I (5676554) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165344
[2021-09-03 13:04:37.292] I (5676554) BT DATA: 4 bytes received
[2021-09-03 13:04:37.323] I (5676594) UART DATA: 25 bytes received
[2021-09-03 13:04:58.549] I (5697824) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165356
[2021-09-03 13:04:58.556] I (5697824) BT DATA: 4 bytes received
[2021-09-03 13:04:58.593] I (5697864) UART DATA: 25 bytes received
[2021-09-03 13:05:05.794] E (5705064) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3
[2021-09-03 13:05:30.696] W (5729964) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
[2021-09-03 13:05:30.709] I (5729974) ESP_SPP_CLOSE_EVT: Connection Closed
[2021-09-03 13:05:31.448] W (5730724) BT_APPL: new conn_srvc id:26, app_id:255
[2021-09-03 13:05:31.471] I (5730744) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165356
[2021-09-03 13:05:31.477] I (5730744) BT DATA: 4 bytes received
[2021-09-03 13:05:31.515] I (5730784) UART DATA: 25 bytes received
[2021-09-03 13:05:57.833] I (5752334) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165356
[2021-09-03 13:05:57.833] I (5752334) BT DATA: 4 bytes received
[2021-09-03 13:05:57.833] I (5752374) UART DATA: 25 bytes received
[2021-09-03 13:06:13.552] I (5772824) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165360
[2021-09-03 13:06:13.560] I (5772824) BT DATA: 4 bytes received
[2021-09-03 13:06:13.594] I (5772864) UART DATA: 25 bytes received
[2021-09-03 13:06:20.795] E (5780064) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3
[2021-09-03 13:06:45.697] W (5804964) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
[2021-09-03 13:06:45.714] I (5804974) ESP_SPP_CLOSE_EVT: Connection Closed
[2021-09-03 13:06:47.735] W (5807004) BT_APPL: new conn_srvc id:26, app_id:255
[2021-09-03 13:06:47.754] I (5807024) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165352
[2021-09-03 13:06:47.761] I (5807024) BT DATA: 4 bytes received
[2021-09-03 13:06:47.795] I (5807064) UART DATA: 25 bytes received
[2021-09-03 13:07:09.553] I (5828824) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165344
[2021-09-03 13:07:09.561] I (5828824) BT DATA: 4 bytes received
[2021-09-03 13:07:09.588] I (5828854) UART DATA: 25 bytes received
[2021-09-03 13:07:16.784] E (5836054) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3
[2021-09-03 13:07:41.696] W (5860964) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
[2021-09-03 13:07:41.708] I (5860974) ESP_SPP_CLOSE_EVT: Connection Closed
[2021-09-03 13:07:43.735] W (5863004) BT_APPL: new conn_srvc id:26, app_id:255
[2021-09-03 13:07:44.434] I (5863704) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165336
[2021-09-03 13:07:44.444] I (5863704) BT DATA: 4 bytes received
[2021-09-03 13:07:44.476] I (5863744) UART DATA: 25 bytes received
[2021-09-03 13:08:05.549] I (5884824) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165332
[2021-09-03 13:08:05.556] I (5884824) BT DATA: 4 bytes received
[2021-09-03 13:08:05.597] I (5884864) UART DATA: 25 bytes received
[2021-09-03 13:08:12.792] E (5892064) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3
[... 131 lines cut ...]
[2021-09-03 13:20:49.727] I (6648974) ESP_SPP_CLOSE_EVT: Connection Closed
[... cut ...]
[2021-09-03 13:22:43.721] W (6762964) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
[2021-09-03 13:22:43.729] I (6762974) ESP_SPP_CLOSE_EVT: Connection Closed
[2021-09-03 13:22:45.159] W (6764424) BT_APPL: new conn_srvc id:26, app_id:255
[2021-09-03 13:22:45.178] I (6764444) esp_spp_cb():ESP_SPP_DATA_IND_EVT: Free Heap Size = 165360
[2021-09-03 13:22:45.184] I (6764444) BT DATA: 4 bytes received
[2021-09-03 13:22:45.216] I (6764484) UART DATA: 25 bytes received

BenjaminWKidd, 2021-09-09:
Hi @xiongweichao ,

We're continuing to see if we can come up with a reproduceable setup that you can try, but in the mean time, as we've dug deeper, it appears that the original panic issue centers around (or is related to) the hciconfig link policy in the linux (beaglebone) device connected by bluetooth to the ESP32. When the link policy includes the SNIFF policy, we see panics as described in the original post above (with the original libbtdm_app.a file). 

(On the beaglebone system:)
# hciconfig -a hci0
hci0:   Type: Primary  Bus: UART
        BD Address: A4:D5:78:6D:73:4B  ACL MTU: 1021:6  SCO MTU: 180:4
        UP RUNNING
        RX bytes:1264824 acl:16013 sco:0 events:65077 errors:0
        TX bytes:483651 acl:16552 sco:0 commands:16330 errors:0
        Features: 0xff 0xfe 0x2d 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        **Link policy: RSWITCH HOLD SNIFF**
        Link mode: SLAVE ACCEPT
        Name: ''
        Class: 0x001f00
        Service Classes: Unspecified
        Device Class: Invalid Device Class!
        HCI Version: 4.1 (0x7)  Revision: 0x0
        LMP Version: 4.1 (0x7)  Subversion: 0xac7c
        Manufacturer: Texas Instruments Inc. (13)

When the link policy of the linux device is modified to remove "SNIFF", the problem appears to go away. We have not verified this with the updated/patched versions of libbtdm_app.a, but I suspect it behaves similarly, since we seem to get "bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=3" (and similar errors) instead of panics.

Another interesting thing that we've noticed: With the link policy set to anything other than SNIFF (such as HOLD and RSWITCH), we've noticed hcistatus errors 31 and 36 show up in the esp32 debug stream, e.g.: 

BT_APPL: bta_dm_pm_btm_status hci_status=36 
BT_APPL: bta_dm_pm_btm_status hci_status=31

These don't seem to amount to anything other than a nuisance when looking at the debug output, but we can't find any documentation on what they mean. When SNIFF is in the policy, the hcistatus errors go away but we start seeing the panics again.

xiongweichao, 2021-09-10:
Hi @BenjaminWKidd ,

Sorry, I don’t have a Beaglebone right now. Can the problem be reproduced using two ESP32 test?
By the way, can this problem be reproduced if you only run idf spp demo?

Thanks

xiongweichao, 2021-09-14:
Hi @BenjaminWKidd ,

I used Ubuntu 18.04 and idf spp demo for 6 hours of testing, but this issue was not reproduced.  Can you reproduce this issue with other linux environments?

Thanks

BenjaminWKidd, 2021-09-14:
Hi @xiongweichao - We have not yet been able to do so. We're still trying some different environments.

BenjaminWKidd, 2021-09-15:
hi @xiongweichao,

This may not help you reproduce the issue exactly, since we still haven't found a reliable way to do so other than with a few specific hardware platforms on the client side, but I can give you this code to look over. Essentially it is the bt_spp_acceptor example with a few modifications to allow it to communicate with some coderunning on a beaglebone (I've taken a screen shot of the file differences of the example_spp_acceptor_demo.c file and included that with the rest of the files). The changes I've made are:
1. Changed the sec_mask to ESP_SPP_SEC_NONE from ESP_SPP_SEC_AUTHENTICATE
2. Added some code to parse the incoming bytes during the esp_spp_cb() ESP_SPP_DATA_IND_EVT and respond back with another stream of bytes.

The client connects to the bt_spp_acceptor (running on a Devkit4C) and every 10 seconds sends the following HEX: 0xF0 0x48 0x86 0x45 and the bt_spp_acceptor responds back with a preprogrammed string of bytes. As you can see in our log, we had 3 panic events over the course of about 12 hours.

You can look through my code and see if I'm doing anything unknowingly stupid, but I think everything is fairly normal.

[bt_spp_acceptor modified.zip](https://github.com/espressif/esp-idf/files/7170120/bt_spp_acceptor.modified.zip)

xiongweichao, 2021-10-09:
Hi @BenjaminWKidd ,

Have you found a way that is easy to reproduce?

Thanks

Alvin1Zhang, 2021-12-07:
@BenjaminWKidd Thanks for reporting, would you please help share if any updates? Thanks.

BenjaminWKidd, 2021-12-07:
Unfortunately I cannot come up with a way to reproduce the issue without my customer's equipment. I've asked them to see if they can come up with a way to do it, but they've been very busy, so I guess this issue is going to have to be set aside for a while.

xiongweichao, 2024-07-09:
Hi @BenjaminWKidd ,  May I ask if the problem still exists? Thanks!

BenjaminWKidd, 2024-07-17:
@xiongweichao - I'm actually not sure if the issue exists or not. The company we were doing this has either resolved it or otherwise found a workaround, though they have not kept me updated about it. I had forgotten about it, but I'll go ahead and close it. If it comes up again, I believe they are better equipped to raise the issue with you guys and I'll encourage them to do so. Thanks for following up.

closed 2024-07-17 by BenjaminWKidd