espressif/esp-idf #7162: Nimble Ble GATT server enable Notifications Encrypted

JosueSanchezEZM, 2021-06-16:
Hi everybody,

I am implementing a solution in which I use one characteristic with Reading and Writing operations. I am only using the Write operation because I send a custom frame to the server with the real request ( Read some information or write some information) to be analyzed on the server-side. What I have accomplished so far is to use the Notification operation to enable this characteristic to receive a  frame response from the GATT server. 

I was implementing encryption but I noticed that only the encrypted channel and the bonding process occur when using Write and Read operations but not Notification. Could you please give some suggestions to possible solutions about how to encrypt the Notification operation of my characteristics or whether you have a comment about a new solution to use in a better way the functionalities that BLE offers. 

I am using esp-idf 4.2 release version example/nimble/bleprph as a reference.

ocESP, 2021-06-21:
Hello @JosueSanchezEZM,

As per [Vol. 3] Part G, Section 4 “GATT feature requirements” of Bluetooth Core Specification, GATT read and write of characteristic have provision for authentication / authorisation and encryption permissions. But, specification does not mandate link encryption for notification (and also indication for that matter). As it is not mandated by the specifications, the NimBLE stack does not support it. 

Moreover, adding such a change would break other third party devices which are obeying to the specifications (that is, they would not support encryption for notification).

One suggestion we could think of is, instead of link encryption (which specification does not mention about), you can try data encryption and data decryption at application layer.

Alvin1Zhang, 2021-11-30:
@JosueSanchezEZM  Thanks for reporting, would you please help share if any updates? Thanks.

rahult-github, 2022-05-24:
Closing this ticket. Please feel free to reopen, if unclear.

closed 2022-05-24 by rahult-github