espressif/esp-idf #7164: [usb host] question about actual_num_bytes

chegewara, 2021-06-17:
Here is description, but it is not clear:
`int actual_num_bytes;                   /**< Actual number of bytes transmitted/receives in the IRP */`

In case of CTRL pipe response `data_buffer` is filled with by ctrl pipe request followed by actual payload. The thing is that `actual_num_bytes` does not include ctrl request and it is just payload length.
My question is is it intended?

Let say i want to request HID report map. Here is output:
```
81 06 00 22 00 00 46 00 05 01 09 06 a1 01 05 08 19 01 29 03 15 00 25 01 75 01 95 03 91 02 95 05 91 01 05 07 19 e0 29 e7 95 08 81 02 75 08 95 01 81 01 19 00 29 91 26 ff 00 95 06 81 00 c0 
```
Total length is 62 bytes, but `actual_num_bytes` value is 54 (+ 8 request bytes).

tore-espressif, 2021-07-26:
Hello @chegewara ,

Thank you for the issue.
USB host library is currently under development. In the final state, users won't interact with the HCD layer but with USB_Host API that hasn't been made public yet.

The behavior you are observing is not intended and it will be fixed and better documented in upcoming USB_Host release.

Dazza0, 2021-09-22:
@chegewara The initial USB Host Library API has updated the behavior of `actual_num_bytes`. It should now reflect the total number of bytes in the transfer, thus:
- For Control, it should include the bytes for the setup packet
- For Isochronous, it should account for total number of bytes in all packets

The description [updated API description](https://github.com/espressif/esp-idf/blob/963836f49125427c8f026c7c3732af86be360efb/components/usb/include/usb/usb_types_stack.h#L121) can be found here.

I'm closing this issue. Please feel free to reopen if you feel the description of `num_bytes` should be updated further

closed 2021-09-22 by Dazza0

chegewara, 2021-09-22:
Thank @Dazza0 
i saw changes in API and already updated my tiny library
https://github.com/chegewara/esp32-usb-host