espressif/esp-idf #7165: Assigning NULL to a size_t* results in a LoadProhibited.

timkoers, 2021-06-17:
## Environment

- Development Kit:      ESP32-DOIT-Devkit
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): ESP-IDF 3.40200.210118
- Build System:         CMake with idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PlatformIO (PowerShell).
- Using an IDE?: Yes, Visual Studio Code with PlatformIO
- Power Supply:         USB

## Problem Description

When creating an empty size_t pointer, using ```size_t* activeCacheArraySize = NULL;```, the code crashes due to an LoadProhibited exception.

### Expected Behavior

Assign NULL, or 0x0 to the pointer's address

### Actual Behavior

Crash, the ESP is trying to load(?) something from somewhere

### Steps to reproduce

1. assign NULL to a size_t* pointer.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
size_t *test = NULL;
}

```

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d54b2  PS      : 0x00060230  A0      : 0x800db584  A1      : 0x3ffbc3a0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffbc430  A5      : 0x00000003  
A6      : 0x00000066  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000001
A10     : 0x0000000a  A11     : 0x3ffbc8e4  A12     : 0x00001800  A13     : 0x3ffb6458  
A14     : 0x3ffe436c  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc
```

Sorry, I can't attach the source.

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder) [sdkconfig.zip](https://github.com/espressif/esp-idf/files/6671279/sdkconfig.zip)

igrr, 2021-06-21:
Hi @timkoers,
It is unlikely that simple assignment such as you show in "Code to reproduce the issue" section will trigger this exception. There has to be some load (memory read) operation in the program to trigger this. For example, after assigning NULL to a pointer some code later in the program tries to dereference the pointer.

I'd recommend doing one of the following to understand the cause of the issue:

* Enable ["GDB Stub" feature](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#gdb-stub), that will invoke GDB in command line mode when an exception occurs. From GDB, you can do `bt` (`backtrace`) command to see the backtrace, and `disassemble /m` to see the disassembly annotated with the C source lines. This usually allows finding the reason for the issue (i.e. which pointer was NULL, and what operation has lead to dereferencing it).
* Reducing the compiler optimization level in menuconfig can help, as the source code line numbers reported in the backtrace will more accurately match the generated code.
* If you can't figure it out using GDB Stub, please do the following:
  1. When the exception happens, copy the complete output of the panic handler, starting from the "Guru Meditation" line and including the backtrace, attach it here.
  2. Run `xtensa-esp32-elf-gdb build/your-app-name.elf` and there run `disassemble /m <address>`, where `<address>` is the value of `PC` register as reported by the panic handler. Copy the disassembly output produced by GDB, attach it here as well. Make sure it actually includes the PC where the crash has happened, as the output may be split across multiple pages.
  3. The logic for debugging [LoadProhibited](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited) exception from the above information looks like this:
     - Look at the source code line where the exception is reported. Does it look like some pointer is dereferenced there? If yes, check if that pointer is NULL.
     - If from the source code it is not obvious that some pointer will be dereferenced: start looking at assembly code. Find the instruction in the disassembly at the address of `PC` register. This should be some kind of a load instruction (`l32`). The 2nd operand of the load instruction is the register containing the address, the 3rd operand is the offset (constant) added to the register. In your case, `EXCVADDR=0x00000000 `, which means that the program has attempted to load value 0. If the instruction located at `PC` looks like `l32 an, am, 0`, then it means that `am` register (m=0..15) contained 0.
     - Now you can look at the previous lines of disassembly and trace where the value `am` was obtained from. Usually it doesn't take long from here to determine which C variable or an expression was dereferenced. As mentioned above, reducing the compiler optimization level can help, since the relation between source code and assembly will be easier to follow.

Alvin1Zhang, 2021-07-28:
@timkoers Would you please help share if any updates for the issue? Thanks.

Alvin1Zhang, 2021-08-18:
@timkoers Thanks for reporting, would you please help share if any updates for the issue? Thanks.

timkoers, 2021-08-28:
@Alvin1Zhang I haven't experienced the issue in a while now, I think it might have been something strange in my code

closed 2021-08-28 by timkoers