espressif/esp-idf #7167: Can't build C++ project with IDF v4.3 release, some guards missed?

thecashit, 2021-06-18:
## Environment
- ESP32-WROOM32D
- IDF version: v4.3
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: macOS
- Using an IDE?: No

## Problem Description
After switching to v4.3 can't build project.
Seems like some guard directives missed but i can't find where.
Could you please help to resolve this issue?

## Debug Logs

```
idf.py build

...
In file included from /Users/SomeProject/Firmware/Toolchain/esp-idf/components/esp_system/include/esp_system.h:21,
                 from ../libs/OTA/OTA.h:17,
                 from ../libs/OTA/OTA.cpp:7:
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/hal/include/hal/spi_types.h: At global scope:
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:111:34: error: conflicting declaration of C function 'constexpr spi_event_t operator~(spi_event_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator~ (TYPE a) { return (TYPE)~(INT_TYPE)a; } \
                                  ^~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:123:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:124:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/hal/include/hal/spi_types.h:46:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(spi_event_t)
 ^~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:111:34: note: previous declaration 'constexpr timer_intr_t operator~(timer_intr_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator~ (TYPE a) { return (TYPE)~(INT_TYPE)a; } \
                                  ^~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:123:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:124:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/hal/include/hal/timer_types.h:80:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(timer_intr_t)
 ^~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:112:34: error: conflicting declaration of C function 'constexpr spi_event_t operator|(spi_event_t, spi_event_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator| (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a | (INT_TYPE)b); } \
                                  ^~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:123:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:124:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/hal/include/hal/spi_types.h:46:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(spi_event_t)
 ^~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:112:34: note: previous declaration 'constexpr timer_intr_t operator|(timer_intr_t, timer_intr_t)'
 FORCE_INLINE_ATTR constexpr TYPE operator| (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a | (INT_TYPE)b); } \
                                  ^~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:123:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:124:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/hal/include/hal/timer_types.h:80:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(timer_intr_t)
[... 198 lines cut ...]
                             ^~~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:124:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/hal/include/hal/spi_types.h:46:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(spi_event_t)
 ^~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:121:25: note: previous declaration 'timer_intr_t& operator<<=(timer_intr_t&, int)'
 FORCE_INLINE_ATTR TYPE& operator<<=(TYPE& a, int b) { a <<= b; return a; }
                         ^~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:123:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/xtensa/include/esp_attr.h:124:19: note: in expansion of macro 'FLAG_ATTR_U32'
 #define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
/Users/SomeProject/Firmware/Toolchain/esp-idf/components/hal/include/hal/timer_types.h:80:1: note: in expansion of macro 'FLAG_ATTR'
 FLAG_ATTR(timer_intr_t)
 ^~~~~~~~~
    ^
```

## Other items if possible
sdkconfig file in attachement
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6678662/sdkconfig.txt)

projectgus, 2021-06-21:
Hi @thecashit,

Thanks for the clear bug report. I can't reproduce this by including any of "esp_system.h", "hal/spi_types.h", "hal/timer_types.h" or "esp_attr.h" in a C++ source file with ESP-IDF v4.3.

I am guessing that OTA.h , OTA.cpp, or another file included from them may define something different, or something has gone wrong with the environment installation.

Can you please try running `idf.py fullclean` and then rebuild and check for any other warnings anywhere in the build log output before it fails?

If the problem still occurs and there are no other warnings shown, is it possible to please share the source files that you're building (or cut down versions that still exhibit the problem)?

thecashit, 2021-06-21:
@projectgus 
Thank you for your answer!

I just tried to create a cleaned project to provide you with information and came across the fact that when you disable most of the logic, everything works.

I will try to dive deeper into the issue within 24 hours and understand on which specific include the compilation stops working.

thecashit, 2021-06-24:
@projectgus 
Latest news: problem was in our code, so, changing one of components solve this issue.
Thank you

closed 2021-06-24 by thecashit