espressif/esp-idf #7169: ESP_HIDH_INPUT_EVENT event_data integrity

xing-w, 2021-06-19:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    ESP-IDF v4.3-rc-3-g3c3d8dfbf-dirty
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [Yes (VS code)]
- Power Supply:         [USB]

## Problem Description
The HIDH_event_data was wrong when print out to the console.
//Detailed problem description goes here.
I made a HID_host and try to connect to a bluetooth HID keyboard. When any key was pressed, I will get an input event and print out the raw data received from the bluetooth. 

### Expected Behavior
if I press key "a" I should receive 9 bytes data: 0x00 0x00 0x04 0x00 0x00 0x00 0x00 0x00 0x00.
When I release the key "a" I should receive 9 bytes data : 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00.
### Actual Behavior
But I got data for press "a": 0x00 0x00 0x04 0x00 0x00 0x1c 0xb2 0xfb 0x3f.
And data for release "a":     0x00 0x00 0x00 0x00 0x00 0x1c 0xb2 0xfb 0x3f.
The last four bytes data were wrong. 
### Hypothesis of the bug:
```c++
    struct {
        esp_hidh_dev_t *dev;                        /*!< HID Remote bluetooth device */
        esp_hid_usage_t usage;                      /*!< HID report usage */
        uint16_t report_id;                         /*!< HID report index */
        uint16_t length;                            /*!< HID data length */
        uint8_t  *data;                             /*!< The pointer to the HID data */
        uint8_t map_index;                          /*!< HID report map index */
    } input;   
```
The  struct "esp_hidh_event_data_t : : input" only pass the data pointer not the data contents. So when the callback try to read the data from the pointer, the data contents may changed or not valid anymore.  
### Some tests
#### test 1: print out the data contents before post to the loop_handle and print out in the callback function.
##### first, print out the data contents before post to the loop_handle.
```C++
void bta_hh_co_data(uint8_t handle, uint8_t *p_rpt, uint16_t len, tBTA_HH_PROTO_MODE  mode, uint8_t sub_class, uint8_t country_code, esp_bd_addr_t bda, uint8_t app_id)
{ 
...
if (event_loop_handle) {
        esp_hidh_event_data_t p = {0};
        if (report->report_type == ESP_HID_REPORT_TYPE_FEATURE) {
            p.feature.dev = dev;
            p.feature.report_id = report->report_id;
            p.feature.usage = report->usage;
            p.feature.data = p_rpt + 1;
            p.feature.length = len - 1;
            esp_event_post_to(event_loop_handle, ESP_HIDH_EVENTS, ESP_HIDH_FEATURE_EVENT, &p, sizeof(esp_hidh_event_data_t), portMAX_DELAY);
        } else {
            p.input.dev = dev;
            p.input.report_id = report->report_id;
            p.input.usage = report->usage;
            p.input.data = p_rpt + 1;
            p.input.length = len - 1;
            ESP_LOG_BUFFER_HEX(TAG, p.input.data, p.input.length);  // print out before post to event_loop_handle
            esp_event_post_to(event_loop_handle, ESP_HIDH_EVENTS, ESP_HIDH_INPUT_EVENT, &p, sizeof(esp_hidh_event_data_t), portMAX_DELAY);
        }
    }
...
}
```
##### Second, print out the data contents in the callback function.
In callback function:
```C++
case ESP_HIDH_INPUT_EVENT: {
        ESP_LOG_BUFFER_HEX(TAG, param->input.data, param->input.length);   // print out data contentx from data pointer
       }
```
They should print out the same 9 bytes data. 
##### Result:
[image]

The print out of the 9 bytes are different. However, the first 5 bytes are always same.
#### test 2: Save the data contents in the input event data and compare the print out again.
##### First add an data_array[10] member to the input struct.
```c++
    struct {
        esp_hidh_dev_t *dev;                        /*!< HID Remote bluetooth device */
        esp_hid_usage_t usage;                      /*!< HID report usage */
        uint16_t report_id;                         /*!< HID report index */
        uint16_t length;                            /*!< HID data length */
        uint8_t  *data;                             /*!< The pointer to the HID data */
        uint8_t map_index;                          /*!< HID report map index */
        uint8_t data_array[10];             // I use size 10 because I know my input data is only 9 bytes.
    } input; 
```
##### Second, copy the data contents to the data_array before post to the event_handle.
```C++
if (event_loop_handle) {
        esp_hidh_event_data_t p = {0};
        if (report->report_type == ESP_HID_REPORT_TYPE_FEATURE) {
            p.feature.dev = dev;
            p.feature.report_id = report->report_id;
            p.feature.usage = report->usage;
            p.feature.data = p_rpt + 1;
            p.feature.length = len - 1;
            esp_event_post_to(event_loop_handle, ESP_HIDH_EVENTS, ESP_HIDH_FEATURE_EVENT, &p, sizeof(esp_hidh_event_data_t), portMAX_DELAY);
        } else {
            p.input.dev = dev;
            p.input.report_id = report->report_id;
            p.input.usage = report->usage;
            p.input.data = p_rpt + 1;
            p.input.length = len - 1;
            if (p.input.length<=10){        // copy data contents to the data_array
                memcpy(p.input.data_array,p.input.data,p.input.length);
            }
            ESP_LOG_BUFFER_HEX(TAG, p.input.data, p.input.length); // print out before post to event_loop_handle
            esp_event_post_to(event_loop_handle, ESP_HIDH_EVENTS, ESP_HIDH_INPUT_EVENT, &p, sizeof(esp_hidh_event_data_t), portMAX_DELAY);
        }
    }
}
```
##### In the callback function, print out data contents both from data_array and data pointer.
```C++
case ESP_HIDH_INPUT_EVENT: {
                
        ESP_LOG_BUFFER_HEX(TAG, param->input.data_array, param->input.length);  // print out data contents from data array
        ESP_LOG_BUFFER_HEX(TAG, param->input.data, param->input.length);           // print out data contentx from data pointer
}
```
##### Result:
[image]
The data contents print out from the data_array (first HID_HOST) is same as the data contents (BT_HIDH) before posting to the event_loop_handle. 
The data contents print out from the data pointer (second HID_HOST) are different from the data contents (BT_HIDH) before posting to the event_loop_handle. 
Again, the first 5 bytes are same and the last 4 bytes are different. But the last 4 bytes are different from last test.

xing-w, 2021-06-19:
I also tried the esp_hid_host example and got a similar result.
[image]
The last 4 bytes were wrong.

Then I add a print out before post to the event_loop_handle.
[image]
The data contents before post to the event_loop_handle is correct and the data contents get from the pointer were wrong.

boblane1, 2021-06-22:
Hi xing-w,

Thanks for your reporting, it seems like a bug about shallow copy. Please try the following patch to see if the problem still exists.
[0001-fix-hid-deep-copy-bugs.zip](https://github.com/espressif/esp-idf/files/6691881/0001-fix-hid-deep-copy-bugs.zip)

xing-w, 2021-06-23:
Hi boblane1,
I tried the patch and it still have problems.
I can fix it by doing the follows:
```c++
--- a/components/esp_hid/src/bt_hidh.c
+++ b/components/esp_hid/src/bt_hidh.c
@@ -409,7 +409,7 @@ void bta_hh_co_data(uint8_t handle, uint8_t *p_rpt, uint16_t len, tBTA_HH_PROTO_
             p->feature.usage = report->usage;
             memcpy(p->feature.data, p_rpt + 1, len - 1);
             p->feature.length = len - 1;
-            esp_event_post_to(event_loop_handle, ESP_HIDH_EVENTS, ESP_HIDH_FEATURE_EVENT, p, sizeof(esp_hidh_event_data_t), portMAX_DELAY);
+            esp_event_post_to(event_loop_handle, ESP_HIDH_EVENTS, ESP_HIDH_FEATURE_EVENT, p, offsetof(esp_hidh_event_data_t, feature.data) + len - 1, portMAX_DELAY);
         } else {
             p = (esp_hidh_event_data_t *)malloc(offsetof(esp_hidh_event_data_t, input.data) + len - 1);
             if (!p) {
@@ -421,7 +421,7 @@ void bta_hh_co_data(uint8_t handle, uint8_t *p_rpt, uint16_t len, tBTA_HH_PROTO_
             p->input.usage = report->usage;
             memcpy(p->input.data, p_rpt + 1, len - 1);
             p->input.length = len - 1;
-            esp_event_post_to(event_loop_handle, ESP_HIDH_EVENTS, ESP_HIDH_INPUT_EVENT, p, sizeof(esp_hidh_event_data_t), portMAX_DELAY);
+            esp_event_post_to(event_loop_handle, ESP_HIDH_EVENTS, ESP_HIDH_INPUT_EVENT, p, offsetof(esp_hidh_event_data_t, input.data) + len - 1, portMAX_DELAY);
         }
```
I only changed for the BT not BLE.

boblane1, 2021-06-24:
Hi xing-w,

Yes, you are right! I have fixed this bug in the updated patch. Anywhere, thank you very much!

closed 2021-09-10 by espressif-bot