espressif/esp-idf #7171: Add -j option to idf.py

kevin-june, 2021-06-19:
Is it possible/reasonable to allow the user to specify the number of cores to be used at the command line? For example:

```
idf.py build -j N
```

Occasionally I will troubleshoot CMake configuration by reading build output, specifically to verify that components and their dependencies are built in the correct order. This is easy to do when running a single-threaded GNU Make, but is unreasonable for a multi-threaded build. Builds using `idf.py` seem to be multi-threaded with a [hard-coded number of cores](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/tools/idf_py_actions/constants.py#L32). I'm wondering if there is a reasonable way for this option to be configurable.

An proposed workflow is:

```
idf.py -G 'Unix Makefiles' build -j 1
```

At the moment, I create this effect by either:

* temporarily editing `constants.py` to set `-j 1`
* manually running a single-core build: `make -C build`

The first idea works nicely, but I hesitate to get in the habit of editing the IDF.
The second approach seems to work, but I'm concerned that it's lacking something that `idf.py` sets up (e.g. environment variables).

I'm not sure of a more effective way to troubleshoot build order other than running a build. If there are other approaches I would appreciate the perspective.

Thanks!

dobairoland, 2021-06-20:
Hi @kevin-june. Thank you for the suggestion. We will consider it. 

Are you debugging build order of individual files or components?

kevin-june, 2021-06-20:
Hi @dobairoland ,

In this case it's a bit more complex - both between components and within a single component, but not quite between files. It's actually between a component, a CMake `ExternalProject` that the component depends on, and the dependencies of the `ExternalProject`.

The backstory is that I'm experimenting with Espressif Rust support in this repo [espressif/rust-esp32-example](https://github.com/espressif/rust-esp32-example/tree/components2). If it's easier to glance at the code (it's short, I promise), see [this CMakeLists.txt](https://github.com/espressif/rust-esp32-example/blob/components2/components/rustlib/CMakeLists.txt).

Thanks for your consideration!

igrr, 2021-06-21:
@kevin-june I usually use `ninja -C build -j 1` to force a single-threaded build.

Both that and `make -C build` (if using the "Unix Makefiles" generator) should work just as well as `idf.py build` does, assuming the project has already been configured once using `idf.py build` or `idf.py reconfigure`. Using CMake directly (and Ninja/Make) is something we do support: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-cmake-directly.

kevin-june, 2021-06-21:
@igrr Ah! I didn't notice that invoking CMake and ninja/make directly is officially supported. Since that is the case, it probably isn't worth adding idf.py support for special options. There would be no end to the feature requests, and it's just as easy to invoke the build system directly.
Thanks!

closed 2021-06-21 by kevin-june