espressif/esp-idf #7173: Call of export.sh in bash will end with an error, when $LANG is set (to non-english)

happenpappen, 2021-06-20:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3
    v4.4-dev-1594-g1d7068e4b
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

When sourcing the export.sh file with ". ./export.sh" the script will fail in bash, when $LANG is set to non-english. I tried this on Ubuntu 21.04 with $LANG set to "de_DE.UTF-8". The error stack is:

$ .  ./export.sh 
Detecting the Python interpreter
Checking "python" ...
Python 3.9.5
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /home/cs/.espressif/python_env/idf4.4_py3.9_env/bin/python
Checking if Python packages are up to date...
Python requirements from /tmp/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /tmp/esp-idf/components/esptool_py/esptool
  /tmp/esp-idf/components/espcoredump
  /tmp/esp-idf/components/partition_table
  /tmp/esp-idf/components/app_update
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Traceback (most recent call last):
  File "/tmp/esp-idf/tools/idf.py", line 822, in <module>
    main()
  File "/tmp/esp-idf/tools/idf.py", line 740, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var='_IDF.PY_COMPLETE')
  File "/home/cs/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/home/cs/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1057, in main
    self._main_shell_completion(extra, prog_name, complete_var)
  File "/home/cs/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/core.py", line 1132, in _main_shell_completion
    rv = shell_complete(self, ctx_args, prog_name, complete_var, instruction)
  File "/home/cs/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 45, in shell_complete
    echo(comp.source())
  File "/home/cs/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 324, in source
    self._check_version()
  File "/home/cs/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/click/shell_completion.py", line 319, in _check_version
    raise RuntimeError(
RuntimeError: Couldn't detect Bash version, shell completion is not supported.

### Expected Behavior

The script should work without above error, regardless the setting of $LANG, maybe it should store/restore the LANG setting while being executed.

### Actual Behavior

See above traceback, there should be no error.

### Steps to reproduce

1. Start bash with $LANG set to de_DE.UTF-8 (i guess the error will happen with any language other than english)
2. Change to esp-idf directory
3. Source export.sh by entering ". ./export.sh"

### Code to reproduce this issue

None needed.

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

sio13, 2021-06-22:
@happenpappen Thank you for reporting the issue. RuntimeError is caused because of the way the `click` library is parsing the bash version. We contacted `click` developers and proposed a solution. Now we are waiting for the response. We will also publish a workaround soon.

martinespgano, 2021-06-22:
@happenpappen Thank you for reporting the issue. RuntimeError is caused because of the way the click library is parsing the bash version. We contacted click developers and proposed a solution. Now we are waiting for the response. We will also publish a workaround soon.

ondrejdudasek, 2021-06-23:
For those in need of quick workaround, it's possible to get it working by modifiing two lines in shell_completion.py, on line 305:
```` python
output = subprocess.run(["bash", "-c", "echo $BASH_VERSION"], stdout=subprocess.PIPE)
match = re.search(r"(\d)\.(\d)\.\d", output.stdout.decode())
````
It requires echo command, but I haven't seen anything with bash without echo yet, so I think it should be universal.

martinespgano, 2021-06-23:
There's already a patch pull request to click https://github.com/pallets/click/pull/1942.

closed 2021-06-28 by espressif-bot

franz-ms-muc, 2021-09-26:
it seems to be not fixed in the current ESP-IDF 4.3 ? right ?

franz-ms-muc, 2021-09-26:
but i can confirm, using the Patch from 23.06 works ...

dobairoland, 2021-09-27:
@franz-ms-muc v4.3.1 released 18 days ago contains the fix.

nopnop2002, 2022-06-23:
This happens for python 3.9.
```
$ python --version
Python 3.9.2
```

Occurs with ESP-IDF v5.0

I changed shell_completion.py:

```
        """
        output = subprocess.run(
            ["bash", "-c", "echo ${BASH_VERSION}"], stdout=subprocess.PIPE
        )
        match = re.search(r"^(\d+)\.(\d+)\.\d+", output.stdout.decode())
        """
        output = subprocess.run(["bash", "-c", "echo $BASH_VERSION"], stdout=subprocess.PIPE)
        match = re.search(r"(\d)\.(\d)\.\d", output.stdout.decode())
```