espressif/esp-idf #7175: NVS Partition Generator Utility lost ability to parse multiline string values in v4.3

DaStoned, 2021-06-21:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32E
- IDF version: v4.3-rc-2-g88c2b69c68
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Given NVS partition data described in CSV file below, `nvs_partition_gen.py` fails to parse the multiline string value (a PEM format x509 key which contains newlines). This works in ESP IDF v4.2.1 and started to fail with v4.3.

```
key,type,encoding,value
identity,namespace,,
pubKey,data,string,"-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEeCiKY3/V2mAkDmpg5ebdN5DgRGkm
fWOkgc5rAvEGuYlWXGFABI4KIex3OCljygDt5R1f0uatA7lnrziunqACuQ==
-----END PUBLIC KEY-----
"
```

### Expected Behavior

In ESP IDF 4.2.1 this generates a valid NVS partition binary with the multi-line PEM key as a string:

```
$ nvs_partition_gen.py generate out/part_nvs_ego.UBIK-2D3D00.csv part_nvs_ego.UBIK-2D3D00.bin 0x4000
```

### Actual Behavior

In ESP IDF 4.3 this started to fail:

```
$ nvs_partition_gen.py generate out/part_nvs_ego.UBIK-2D3D00.csv part_nvs_ego.UBIK-2D3D00.bin 0x4000
Creating NVS binary with version: V2 - Multipage Blob Support Enabled

Error:
Length of key `MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEeCiKY3/V2mAkDmpg5ebdN5DgRGkm` should be <= 15 characters.

Warning: NVS binary not created...
```

### Steps to reproduce

1. Copy-paste the content of sample CSV into a file
2. Run command for generating the NVS data partition

Side note: in addition to this issue, comments in the file (lines starting with `#`) are no longer supported, which is slightly annoying but not critical.

### Other

My python version is 3.9.1 (note this didn't change when I upgraded ESP IDF) and below is output from `pip freeze` (which was upgraded quite a bit with the upgrade) if relevant:

```
bitstring==3.1.7
Brotli==1.0.9
cachetools==4.2.2
certifi==2021.5.30
cffi==1.14.5
chardet==4.0.0
click==8.0.1
construct==2.10.54
cryptography==3.4.7
ecdsa==0.17.0
Flask==0.12.5
Flask-Compress==1.10.1
Flask-SocketIO==2.9.6
future==0.18.2
gdbgui==0.13.2.0
gevent==1.5.0
google-api-core==1.30.0
google-auth==1.31.0
google-cloud-iot==2.0.1
google-cloud-secret-manager==2.1.0
googleapis-common-protos==1.53.0
greenlet==1.1.0
grpc-google-iam-v1==0.12.3
grpcio==1.38.0
idna==2.10
itsdangerous==2.0.1
Jinja2==3.0.1
kconfiglib==13.7.1
libcst==0.3.19
MarkupSafe==2.0.1
mypy-extensions==0.4.3
packaging==20.9
proto-plus==1.18.1
protobuf==3.17.3
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.20
pyelftools==0.27
pygdbmi==0.9.0.2
Pygments==2.9.0
pyparsing==2.3.1
pyserial==3.5
python-engineio==3.14.2
python-socketio==4.6.1
pytz==2021.1
PyYAML==5.4.1
reedsolo==1.5.4
requests==2.25.1
rsa==4.7.2
six==1.16.0
typing-extensions==3.10.0.0
typing-inspect==0.7.1
urllib3==1.26.5
Werkzeug==0.16.1
```

shivani-tipnis, 2021-07-15:
@DaStoned Thank you. We are looking into adding the fix.

dbahrdt, 2021-09-01:
As far as I can tell it's also not possible to have commas within escaped strings.

I think the problem is in the following line:
https://github.com/espressif/esp-idf/blob/5f38b766a83d18f78167d1d0dd8c8427ea1a36cb/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py#L936

This simply splits the line at the commas. The problem mentioned in this issue is also related to that code since the file is read line by line and thus the newlines within escaped strings are lost.

shivani-tipnis, 2021-09-15:
Hi @DaStoned 

For IDF >=v4.3, suggest using the `file` type for using multiline strings in the CSV.
You can use it this way -

```
key,type,encoding,value 
identity,namespace,, 
pubKey,file,string,pubKey.txt
```

And then you can have the string in the pubKey.txt as follows -

```
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEeCiKY3/V2mAkDmpg5ebdN5DgRGkm
fWOkgc5rAvEGuYlWXGFABI4KIex3OCljygDt5R1f0uatA7lnrziunqACuQ==
-----END PUBLIC KEY-----
```

shivani-tipnis, 2021-09-15:
Hi @dbahrdt 
If the string contains a comma, you can use the `file` type in the same way as shown in above snippet about using multiline strings.
Thanks

DaStoned, 2021-09-21:
Hi, thank you for the reply. That is a viable workaround, though not nearly as convenient as the earlier behaviour I'm sorry to say.

DaStoned, 2021-09-23:
@shivani-tipnis - I see you've marked this issue as Done. Unfortunately I cannot accept this resolution. The proposed workaround is not viable for the problem that @dbahrdt reported, for strings containing commas. I will be generating strings during the manufacturing process and will not know in advance which characters they contain. A CSV file must support a string field with commas (and spaces, and many other characters) if enclosed within quotes, e.g:
```
wifiFbPass,data,string,"Hello,World"
```
Gentlemen. This script is proposed by Espressif as the tool for preparing device images in manufacturing. Please show it some respect and fix this, otherwise you'll be left with hobbyist customers.

shivani-tipnis, 2021-09-24:
Hi @DaStoned 
We will consider your feedback and work on improving this further.
Thanks

will-emmerson, 2022-01-17:
I've also just been affected by this. I think it's a good idea to force people to use the `file` type for more complicated strings as it's actually more explicit and less likely to go wrong. However it should definitely refuse to run (and suggest the fix) when a string contains a comma  as currently it is silently failing.

DaStoned, 2022-01-17:
> I've also just been affected by this. I think it's a good idea to force people to use the `file` type for more complicated strings as it's actually more explicit and less likely to go wrong. However [...]

I agree that the silent failure is the worst problem. 

But as a side note - the Python CSV module (which was used in previous versions of this same script) is perfectly capable of handling "complicated" strings. Replacing this with a naive _split string on commas_ is the source of the current problem.

Please revert back to the Python CSV module (which did its job admirably) or use another, more deterministic text format (JSON, YAML,...)

dobairoland, 2022-06-21:
Hi all. We are sorry it took us so long but this has now been addressed in the master branch (v5.0). The fix will be backported to previous releases shortly.

DaStoned, 2022-06-21:
> Hi all. We are sorry it took us so long but this has now been addressed in the master branch (v5.0). The fix will be backported to previous releases shortly.

Wonderful news. I'll be happy to verify once it reaches a 4.x release.