espressif/esp-idf #7177: Compile time error using esp32c3.cmake

pn21, 2021-06-21:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:   ESP32-C3-DEVKITM-1
- Kit version (for DevKitM): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-C3-MINI
- IDF version (run ``git describe --tags`` to find it): v4.3
- Build System:  CMAKE       [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp32-c3.cmake
    // 1.22.0-80-g6c4433a 
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt 
- Using an IDE?: Yes Eclipse 
- Power Supply:       USB 

## Problem Description

I am getting the following error code when I compile our code using the esp32c3 cmake file:
.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\riscv32-esp-elf\include\c++\8.4.0\cstdlib:232:11: error: '::strtold' has not been declared
   using ::strtold;

My code has a combination of .c & .c++ files. The above error message shows up in all the files. I am using IDF v4.3(stable version). I am already using the foll:
#ifdef __cplusplus
extern "C"
{
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <cstdlib>
}
#endif

Any thoughts on how to resolve this error?

### Expected Behavior
Compile without any issues. Compiler should be able to differentiate the difference between .c & c++ files.

### Actual Behavior
Every time I compile get the following error message:
using the esp32c3 cmake file:
.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\riscv32-esp-elf\include\c++\8.4.0\cstdlib:232:11: error: '::strtold' has not been declared
   using ::strtold;

## Other items if possible

- Attached is the sdkconfig.h file.
[sdkconfig_info.txt](https://github.com/espressif/esp-idf/files/6688646/sdkconfig_info.txt)

projectgus, 2021-06-22:
Hi @pn21,

I replied to your forum post about this: https://esp32.com/viewtopic.php?f=13&t=21567&p=78611#p78611

projectgus, 2021-07-13:
Hi @pn21,

Did you make any progress finding a reproducible case for this?

Alvin1Zhang, 2021-07-20:
Thanks for reporting, sorry we didn't manage to reproduce the issue locally, will close due to short of updates, feel free to reopen with more details. Thanks.

closed 2021-07-20 by Alvin1Zhang