espressif/esp-idf #7179: How can I include l2c_api.h in my projects?

aguaviva, 2021-06-21:
And to be more precise all what is in 

> ~/esp-idf-v4.2.2/components/bt/host/bluedroid/stack/include/stack

Thank you

Alvin1Zhang, 2021-06-22:
@aguaviva Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

aguaviva, 2021-06-24:
I have filled that template before for otehr bugs, but for this particular one that info is of no use. 

Any espressif engineeer shoud be able to answer my question without much effort, if you could have someone reply that would be awesome :)

Wth-Esp, 2021-06-28:
Hi, @aguaviva 

Why do you want to call the API of `l2c`? What app do you want to implement?
The `ESP-APIs` are defined in the path [`components/bt/host/bluedroid/api`](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/tree/master/components/bt/host/bluedroid/api).

And the path you mentioned, named after `stack`, is the main function body of the Bluedroid Dual Mode Stack. It include the logic module and the utils functions of core stack like L2CAP... and the profile like AVDPT, HFP...

Hope this answer help you.

Thanks

aguaviva, 2021-06-28:
I implemented a PS3 controller to BLE https://github.com/aguaviva/PS3ToBle and for that I need to have access to the l2cap

I know where the l2cap stuff is, what I dont know is how to include the header files in my project. What I did was to copy over the files into my project, a thing that I would like to avoid doing.

Hence I'd need to know the exact change I'd need to do in a generic makefile (pick any IDF project with a simple makefile, not my arduino project) to be able to `#include "l2c_api.h"`  in my project

Wth-Esp, 2021-06-29:
Hi, @aguaviva 

You want to implement the PS3 Controller (the gamepad?) which plays a role in HID device and to connect ESP32 which plays a role in HID Host via Bluetooth Radio. (Classic BT or BLE?)

Am I right?

So, from my perspective, what you need is a set of HID Host APIs to make use of Bluetooth radio? And the HID components we now support is locate in [HID Component](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/tree/master/components/esp_hid).

We now support BLE and BT HID host.

May this help you. Thanks.

Wth-Esp, 2021-06-29:
You can use the ESP-HID first, we have an internal optimization for this component and will be released on v4.4.

Wth-Esp, 2021-06-30:
@aguaviva 

aha, I have already finished Zelda too ;)

You should add the include path and all the source file into the CmakeList.txt of your project.
Please refer to [include path](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/bt/CMakeLists.txt#L72) and [source files](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/bt/CMakeLists.txt#L231).

Thanks

closed 2021-09-02 by Alvin1Zhang