espressif/esp-idf #7180: ESP32 S3 Beta3 Crash at BLE MESH Provisioning with ESP IDF 4.3 Master

nitinbhosaleemb, 2021-06-22:
----------------------------- Delete below -----------------------------

- [coredump]
- [ ESP-ROM:esp32s3beta3-20201106
Build:Nov  6 2020
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0100,len:0x1644
load:0x403b6000,len:0x800
load:0x403ba000,len:0x2aac
entry 0x403b61a0
I (22) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (22) boot: compile time 19:34:15
I (22) boot: chip revision: 0
I (25) boot.esp32s3: SPI Speed      : 80MHz
I (29) boot.esp32s3: SPI Mode       : DIO
I (34) boot.esp32s3: SPI Flash Size : 2MB
I (39) boot: Enabling RNG early entropy source...
W (44) bootloader_random: RNG for ESP32-S3 not currently supported
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2a7bch (174012) map
I (128) esp_image: segment 1: paddr=0003a7e4 vaddr=3fc913c0 size=03b04h ( 15108) load
I (131) esp_image: segment 2: paddr=0003e2f0 vaddr=40374000 size=01d28h (  7464) load
I (136) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=93394h (603028) map
I (250) esp_image: segment 4: paddr=000d33bc vaddr=40375d28 size=0b690h ( 46736) load
I (261) esp_image: segment 5: paddr=000dea54 vaddr=600fe000 size=00010h (    16) load
I (266) boot: Loaded app from partition at offset 0x10000
I (266) boot: Disabling RNG early entropy source...
W (269) bootloader_random: RNG for ESP32-S3 not currently supported
I (288) cpu_start: Pro cpu up.
I (288) cpu_start: Single core mode
I (296) cpu_start: Pro cpu start user code
I (296) cpu_start: cpu freq: 160000000
I (296) cpu_start: Application information:
I (299) cpu_start: Project name:     sensor_server
I (304) cpu_start: App version:      1
I (308) cpu_start: Compile time:     Jun 18 2021 20:10:18
I (315) cpu_start: ELF file SHA256:  fc21571b5ecd3e85...
I (321) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (327) heap_init: Initializing. RAM available for dynamic allocation:
I (334) heap_init: At 3FC9B890 len 00044770 (273 KiB): D/IRAM
I (340) heap_init: At 3FCE0000 len 0000EF34 (59 KiB): STACK/DRAM
I (347) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (353) spi_flash: detected chip: gd
I (357) spi_flash: flash io: dio
W (361) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (375) cpu_start: Starting scheduler on PRO CPU.
I (380) EXAMPLE: Initializing...
W (380) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (390) BT_INIT: BT controller compile version [501d88d]
I (390) coexist: coexist rom version a91193a
I (400) phy_init: phy_version 201,d731b8e,Apr 14 2021,18:22:14
I (490) system_api: Base MAC address is not set
I (490) system_api: read default base MAC address from EFUSE
I (490) BT_INIT: Bluetooth MAC: 7c:df:a1:42:b2:0a

I (580) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (590) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (590) EXAMPLE: BLE Mesh sensor server initialized
I (590) EXAMPLE: ###### board_led_init() GPIO5 5 ON #######
I (2600) EXAMPLE: ###### board_led_init() GPIO5 5 OFF #######
I (94620) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (97070) BLE_MESH: No Health Server context provided
W (101040) BLE_MESH: No Health Server context provided
assert ke_task.c 177, param 00000000 00000000
E (106030) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (106030) task_wdt:  - IDLE (CPU 0)
E (106030) task_wdt: Tasks currently running:
E (106030) task_wdt: CPU 0: btController
E (106030) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4206AD06:0x3FC91DE00x403765CA:0x3FC91E00 0x40006B04:0x3FCE6ED0 0x42072D6B:0x3FCE6EF0 0x4000CB75:0x3FCE6F10 0x4002C3C9:0x3FCE6F30  |<-CORRUPTED

E (111030) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (111030) task_wdt:  - IDLE (CPU 0)
E (111030) task_wdt: Tasks currently running:
E (111030) task_wdt: CPU 0: btController
E (111030) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4206AD06:0x3FC91DE00x403765CA:0x3FC91E00 0x40006B04:0x3FCE6ED0 0x42072D6B:0x3FCE6EF0 0x4000CB75:0x3FCE6F10 0x4002C3C9:0x3FCE6F30  |<-CORRUPTED

E (116030) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (116030) task_wdt:  - IDLE (CPU 0)
E (116030) task_wdt: Tasks currently running:
E (116030) task_wdt: CPU 0: btController
E (116030) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4206AD06:0x3FC91DE00x403765CA:0x3FC91E00 0x40006B04:0x3FCE6ED0 0x42072D6B:0x3FCE6EF0 0x4000CB75:0x3FCE6F10 0x4002C3C9:0x3FCE6F30  |<-CORRUPTED

E (121030) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (121030) task_wdt:  - IDLE (CPU 0)
E (121030) task_wdt: Tasks currently running:
E (121030) task_wdt: CPU 0: btController
E (121030) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4206AD06:0x3FC91DE00x403765CA:0x3FC91E00 0x40006B04:0x3FCE6ED0 0x42072D6B:0x3FCE6EF0 0x4000CB75:0x3FCE6F10 0x4002C3C9:0x3FCE6F30  |<-CORRUPTED
]

Campou, 2021-06-22:
Hi @nitinbhosaleemb 

Please help to disable the menuconfig option [Support Duplicate Scan in BLE Mesh](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/bt/esp_ble_mesh/Kconfig.in#L10) and have a try.

Thanks.

nitinbhosaleemb, 2021-06-28:
@Campou yes it working now

Campou, 2021-06-29:
@nitinbhosaleemb Thanks for the feedback. We have fixed this issue internally, and the fix will be synchronized to Github soon.

closed 2021-09-13 by nitinbhosaleemb