espressif/esp-idf #7182: WiFi disconnect and right after reconnect. it happen periodically every ~30min

Rokachy, 2021-06-22:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version : v1
- Module or chip used:  ESP32-WROOM-32D
- IDF version :  4.3 release tag
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it) 
- Operating System:     Windows
- Using an IDE?:   No
- Power Supply:   5V

## Problem Description
Our app implement a BLE to WiFI gateway.
The Gateway listen to BLE devices advertisements (BLE scanning) and report the list of devices to system application via WiFi.

WiFi is disconnecting and right after reconnecting. it happen periodically every ~30min. Why Wifi is disconnecting?
We test it with v4.2.1 and v4.3 and its the same.

### Expected Behavior
WiFi stay connected.

### Actual Behavior
disconnecting and right after reconnecting
## Debug Logs

```
I (1842779) wifi:state: run -> init (1a0)
I (1842779) wifi:pm stop, total sleep time: 1508491543 us / 1840238933 us
```
```
I (3683949) wifi:state: run -> init (1a0)
I (3683949) wifi:pm stop, total sleep time: 1403121518 us / 1837964286 us
```
```
I (5523399) wifi:state: run -> init (1a0)
I (5523399) wifi:pm stop, total sleep time: 1247561271 us / 1839335170 us
```

nishanth-radja, 2021-06-24:
@Rokachy Thanks for reporting the issue,To debug and reproduce it we need a little more logs/details
1.From the bug ,I understand that the BT interface of ESP32 is continuously scanning while the WiFi is connected to the AP, But
   the wifi reconnects every 30 min.Is that correct.
2.What is the security of WiFi network,is it wpa2-psk?
3. Can we have a wifi sniffer capture during the issue.
4. Can you enable the logs, (supplicant logs and wifi logs,which can be enabled using menuconfig).

nishanth-radja, 2021-10-20:
@Rokachy capture will surely help,We can decode the capture if we know the SSID and the password of the network.

AxelLin, 2021-11-03:
> 1. I will try without BT and see if this help

@Rokachy 
Any update?

@nishanth-radja 
What is the meaing of "wifi:state: run -> init (**1a0**)"?

Rokachy, 2021-11-03:
I was busy with other priority tasks I will be back on this next week.

Rokachy, 2021-11-09:
Hi,

The "*1a0" *is the meaning for disassociated with unspecified reason (this
what I understand).
I took wirshark logs (without any data - just connected to cisco AP).
[image: image.png]
I see that the disassociation is sent from the Cisco AP to the ESP32.
I couldn't understand why? its every 30min.
The interesting thing is that this happens for most of the ESP32 sdks but
when I am using SD4.0.3 there isn't any disassociation disconnect from the
Cisco AP.
Any idea how to continue investigate this disconnection?

Thank you,
Yehuda

AxelLin, 2021-11-09:
> Hi, The "*1a0" *is the meaning for disassociated with unspecified reason (this what I understand). I took wirshark logs (without any data - just connected to cisco AP). [image: image.png]

@Rokachy 
Seems the file image.png was missed.
BTW, please upload pcap file to make it easier to check the log. (You probably need to rename file extension for uploading)

Rokachy, 2021-11-09:
rename extension to txt
[WiFiDisconnection-pcapng.txt](https://github.com/espressif/esp-idf/files/7503364/WiFiDisconnection-pcapng.txt)
The capture start 10 seconds before disconnection

Rokachy, 2021-11-11:
I disabled the BT and use only WiFi without power saving (power saving is mandatory if we use BT and WiFi), but  disconnection is still after ~30min :(.
I took another device that uses WiFi TI hardware and it stay connected for hours. 
The issue looks like a configuration at ESP32 since SDK 4.0.3 doesn't have the disconnection issue

nishanth-radja, 2021-11-16:
@Rokachy From the capture I see that the Cisco AP is sending a deauth packet with "Unspecified reason code". 
1. Do you see any error on the cisco controller.
2. Also what is the RSSI of the client seen on the AP, This can happen if the client is Far away from the AP.
3. Is there any data transfer happening or the client is idle,If so is there any idle time out configured on the AP.

Rokachy, 2021-11-16:
Thank you for your response.
I will check with our IT team and get back to you.

nishanth-radja, 2021-11-17:
Also pls share the commit ID you are using for esp32.That will help us to repro the issue.

MaxwellAlan, 2021-11-24:
hi @Rokachy , every 30mins seems like group key update time, can you help check the group key update time of your router?

Rokachy, 2021-11-25:
Yes, I will.

AxelLin, 2022-01-12:
> Yes, I will.
> […](#)
> On Wed, 24 Nov 2021 at 04:52, alanmaxwell ***@***.***> wrote: hi @Rokachy <https://github.com/Rokachy> , every 30mins seems like group key update time, can you help check the group key update time of your router? 

@Rokachy 
Any update?

AxelLin, 2022-01-15:
> hi @Rokachy , every 30mins seems like group key update time, can you help check the group key update time of your router?

The same question as https://github.com/espressif/esp-idf/issues/8236#issuecomment-1013127323

closed 2022-05-23 by Alvin1Zhang