espressif/esp-idf #7184: Add a config for omitting Ctrl+T in idf.py monitor

wangyz1997, 2021-06-23:
**Is your feature request related to a problem? Please describe.**

No.

**Describe the solution you'd like**

idf.py monitor have a shortcut to rebuild and flash new app, but this requires two steps: Ctrl+T and Ctrl+A. I know this is to coexist with console hotkeys, but in most cases serial console only acts as a log printer. So in this case, can you please add a config for omitting the Ctrl+T menu shorcut? This will be useful.

dobairoland, 2021-06-23:
Hi @wangyz1997. Thank you for the suggestion. Maybe this could be implemented in a way of customized shortcuts. But I'm not sure if anyone else would use it beside you. CTRL+T is with us since the initial release of IDF Monitor more than 4 years ago. I will think about this...

chegewara, 2021-06-23:
If i may add something, there is small issue with `CTRL + R` in VS code command line. It is shortcut that prevents reset esp32 in idf.py monitor.

wangyz1997, 2021-06-23:
> Hi @wangyz1997. Thank you for the suggestion. Maybe this could be implemented in a way of customized shortcuts. But I'm not sure if anyone else would use it beside you. CTRL+T is with us since the initial release of IDF Monitor more than 4 years ago. I will think about this...

Thanks for your reply. It would be much better with customized shortcuts.

closed 2023-10-02 by espressif-bot