espressif/esp-idf #7185: Added mqtt to cmakelists, but still undefined reference to functions

silardg, 2021-06-23:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): newest
- Build System:       idf.py
- Operating System:   macOS
- Using an IDE?:  VScode with esp-idf extension

## Problem Description
I have a component folder named MMQTT. 
MMQTT includes:
```

#include <functional>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

#include "esp_log.h"
#include "esp_heap_caps.h"
#include "esp_timer.h"

#include <Utilities.hpp>
#include "Constants.h"
#include "mqtt_client.h"
```

CMakeList.txt
```
idf_component_register(
    SRCS "MMQTT.cpp"
    INCLUDE_DIRS "."
    REQUIRES Utilities Constants mqtt esp_netif
)
```

Getting error:
```
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj):(.literal._ZN5MMQTT5startEv+0xc): undefined reference to `esp_mqtt_client_start'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj):(.literal._ZN5MMQTT12networkReadyEv+0x4): undefined reference to `esp_mqtt_client_init'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj):(.literal._ZN5MMQTT4stopEv+0x0): undefined reference to `esp_mqtt_client_stop'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj):(.literal._ZZN5MMQTTC4ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEjSt8functionIFvS5_S5_EEENUlPvE_4_FUNES9_+0x8): undefined reference to `esp_mqtt_client_publish'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj):(.literal._ZZN5MMQTTC4ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEjSt8functionIFvS5_S5_EEENUlPvE_4_FUNES9_+0xc): undefined reference to `esp_mqtt_client_subscribe'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj): in function `MMQTT::start()':
/Users/silardg/Documents/Development/pierre-firmware-espidf/build/../components/MMQTT/MMQTT.cpp:208: undefined reference to `esp_mqtt_client_start'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj): in function `MMQTT::networkReady()':
/Users/silardg/Documents/Development/pierre-firmware-espidf/build/../components/MMQTT/MMQTT.cpp:202: undefined reference to `esp_mqtt_client_init'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj): in function `MMQTT::stop()':
/Users/silardg/Documents/Development/pierre-firmware-espidf/build/../components/MMQTT/MMQTT.cpp:210: undefined reference to `esp_mqtt_client_stop'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/MMQTT/libMMQTT.a(MMQTT.cpp.obj): in function `operator()':
/Users/silardg/Documents/Development/pierre-firmware-espidf/build/../components/MMQTT/MMQTT.cpp:106: undefined reference to `esp_mqtt_client_publish'
/Users/silardg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/silardg/Documents/Development/pierre-firmware-espidf/build/../components/MMQTT/MMQTT.cpp:110: undefined reference to `esp_mqtt_client_subscribe'
```

silardg, 2021-06-23:
Did a `idf.py fullclean` and it worked.
First I had the file named MQTT.cpp and then renamed it to MMQTT.cpp, to have it different.
Does it matter if it's lower or upper case? Will it be registered the same?

david-cermak, 2021-06-23:
Could you please close the issue if it works now?

> First I had the file named MQTT.cpp and then renamed it to MMQTT.cpp, to have it different.

Filename `MQTT.cpp` should be ok, there's no `MQTT.cpp` anywhere in idf (even if it was, it's a private component source, so shouldn't cause any trouble)

> Does it matter if it's lower or upper case? Will it be registered the same?

It does matter (on MacOS) but we recommend using consistent case for filenames, otherwise it could cause strange issues when using the same project on windows.

closed 2021-06-23 by silardg