espressif/esp-idf #7186: WiFi Memory usage much higher when SPIRAM is enabled (ESP32S2)

DCSBL, 2021-06-23:
## Environment

- Development Kit: [ESP32-S2-HMI-DevKit-1](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s2-hmi-devkit-1)
- Kit version v1(?)
- Module or chip used: ESP32-S2-WROVER
- IDF version [v4.2.2](https://github.com/espressif/esp-idf/releases/tag/v4.2.2)
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: macOS
- (Windows only) environment type: -
- Using an IDE?: VSCode + ESP Extension
- Power Supply: USB

## Problem Description
When enabling `Component config > ESP32S2-specific > Support for external, SPI-connected RAM` in menuconfig (`ESP32S2_SPIRAM_SUPPORT`), the memory usage from WiFi increases a huge amount. (from 54KB assigned heap to 110KB)

I have no idea if this option needs to be enabled for my project, but in the [example project](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s2-hmi-devkit-1/examples/get-started/hello_world) it is enabled by default. I want to know if it is a bug or that this behaviour is intended.

## Context
I try to enable WiFi and LVGL (the graphics library) on the HMI DevKit, but It ran out of memory.  LVGL takes a lot of memory for the display buffer, but it seemed that WiFi also took a lot of memory for its own (about 110KB). After a bit of digging I've tried the [WiFi Station](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) example to remove any overhead from LVGL.

I have used this example to initialise WiFi on the HMI DevKit. I came to the conclusion that the only real difference is the `ESP32S2_SPIRAM_SUPPORT` option. When I turn this off it the memory usage is reduced to 54KB.

The moment that the memory is assigned seems to be in the `esp_wifi_init_internal` method, but that code is 'private' or precompiled. So I cannot look further. https://github.com/espressif/esp-idf/blob/7ab8f793ca5b026f37ae812bcc103e3aa698d164/components/esp_wifi/include/esp_private/wifi.h#L104

### Expected Behaviour
The memory usage should not change (that drastically).

### Actual Behaviour
Initialising WiFi consumes much more memory when `ESP32S2_SPIRAM_SUPPORT` is set.

### Steps to reproduce

1. Add some debug lines in the [wifi station example](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c):
```diff
void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    
+   heap_caps_print_heap_info((MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT));
    
    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
    
+   heap_caps_print_heap_info((MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT));
}
```
2. Run example, log result seen below:
```
Heap summary for capabilities 0x00000804:
  At 0x3ffd4650 len 162224 free 135884 allocated 25872 min_free 135884
    largest_free_block 135884 alloc_blocks 108 free_blocks 1 total_blocks 109
  At 0x3fffc000 len 14864 free 14828 allocated 0 min_free 14828
    largest_free_block 14828 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 150712 allocated 25872 min_free 150712 largest_free_block 135884
I (424) wifi station: ESP_WIFI_MODE_STA
Heap summary for capabilities 0x00000804:
  At 0x3ffd4650 len 162224 free 80988 allocated 80552 min_free 80968
    largest_free_block 80968 alloc_blocks 161 free_blocks 2 total_blocks 163
  At 0x3fffc000 len 14864 free 14828 allocated 0 min_free 14828
    largest_free_block 14828 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 95816 allocated 80552 min_free 95796 largest_free_block 80968
```
3. Enable `ESP32S2_SPIRAM_SUPPORT` and run example, log result seen below:
```
Heap summary for capabilities 0x00000804:
  At 0x3ffd4d00 len 32767 free 12340 allocated 20312 min_free 12340
    largest_free_block 12340 alloc_blocks 19 free_blocks 1 total_blocks 20
  At 0x3ffd4c68 len 160664 free 121948 allocated 38316 min_free 121948
    largest_free_block 121948 alloc_blocks 91 free_blocks 1 total_blocks 92
  At 0x3fffc000 len 14864 free 14828 allocated 0 min_free 14828
    largest_free_block 14828 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 149116 allocated 58628 min_free 149116 largest_free_block 121948
I (458) wifi station: ESP_WIFI_MODE_STA
Heap summary for capabilities 0x00000804:
  At 0x3ffd4d00 len 32767 free 32 allocated 32584 min_free 32
    largest_free_block 32 alloc_blocks 28 free_blocks 1 total_blocks 29
  At 0x3ffd4c68 len 160664 free 24960 allocated 134996 min_free 24940
    largest_free_block 24940 alloc_blocks 167 free_blocks 2 total_blocks 169
  At 0x3fffc000 len 14864 free 14828 allocated 0 min_free 14828
    largest_free_block 14828 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 39820 allocated 167580 min_free 39800 largest_free_block 24940
```

igrr, 2021-06-23:
This is probably due to the fact that a few Wi-Fi library configuration options have different default values depending on whether PSRAM is enabled or not.
For example, https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/esp_wifi/Kconfig#L152-L157
You can try adding `SPIRAM_TRY_ALLOCATE_WIFI_LWIP=n` to sdkconfig.defaults of your project, then regenerate sdkconfig (`rm sdkconfig; idf.py reconfigure`). This should take these options back to their non-PSRAM default values, resulting likely in slightly reduced Wi-Fi performance.

DCSBL, 2021-06-24:
Thanks, that makes sense. I have tested this and it still uses a lot of memory compared to when `ESP32S2_SPIRAM_SUPPORT` is disabled. But it improved the memory consumption of the project itself a bit, see below. Is this behaviour intended?

Can you please give me a reference of documentation page where I can read what I can configure to improve the memory usage?

Before enabling SPIRAM:
```
Heap summary for capabilities 0x00000804:
  At 0x3ffcf6d8 len 182568 free 165816 allocated 16500 min_free 165504
    largest_free_block 165816 alloc_blocks 54 free_blocks 1 total_blocks 55
  At 0x3fffc000 len 14864 free 14828 allocated 0 min_free 14828
    largest_free_block 14828 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 180644 allocated 16500 min_free 180332 largest_free_block 165816
I (409) wifi station: ESP_WIFI_MODE_STA
Heap summary for capabilities 0x00000804:
  At 0x3ffcf6d8 len 182568 free 125772 allocated 56048 min_free 122268
    largest_free_block 124480 alloc_blocks 169 free_blocks 10 total_blocks 179
  At 0x3fffc000 len 14864 free 14828 allocated 0 min_free 14828
    largest_free_block 14828 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 140600 allocated 56048 min_free 137096 largest_free_block 124480

***
was: free 95816 allocated 80552 min_free 95796 largest_free_block 80968
180644 - 140600 = 40044 (was 150712 - 95816 = 54896)
```

After enabling SPIRAM:
```
Heap summary for capabilities 0x00000804:
  At 0x3ffcfdf0 len 32767 free 19444 allocated 13224 min_free 19412
    largest_free_block 19444 alloc_blocks 15 free_blocks 1 total_blocks 16
  At 0x3ffcfd58 len 180904 free 144616 allocated 36088 min_free 144312
    largest_free_block 144616 alloc_blocks 41 free_blocks 1 total_blocks 42
  At 0x3fffc000 len 14864 free 14828 allocated 0 min_free 14828
    largest_free_block 14828 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 178888 allocated 49312 min_free 178552 largest_free_block 144616
I (877) wifi station: ESP_WIFI_MODE_STA
Heap summary for capabilities 0x00000804:
  At 0x3ffcfdf0 len 32767 free 32 allocated 32584 min_free 4
    largest_free_block 32 alloc_blocks 28 free_blocks 1 total_blocks 29
  At 0x3ffcfd58 len 180904 free 42504 allocated 137580 min_free 38380
    largest_free_block 40976 alloc_blocks 188 free_blocks 9 total_blocks 197
  At 0x3fffc000 len 14864 free 14828 allocated 0 min_free 14828
    largest_free_block 14828 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 57364 allocated 170164 min_free 53212 largest_free_block 40976

***
was: free 39820 allocated 167580 min_free 39800 largest_free_block 24940
178888 - 57364 = 121524 (was 149116 -  39820 = 109296)
```

chegewara, 2021-06-24:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/index.html

Can you check 1 thing please?
I believe that PSRAM is forcing wifi to use static buffers when internal memory allows to select between static and dynamic allocation.

DCSBL, 2021-06-24:
@chegewara
Thank you for the link. 
I think I don't fully understand you. How can I check this?

chegewara, 2021-06-24:
with `idf.py menuconfig`
`Component config  --->Wi-Fi  ---> Type of WiFi TX buffers (Static) `

igrr, 2021-06-24:
@DCSBL With regards to the logs you posted in https://github.com/espressif/esp-idf/issues/7186#issuecomment-867600769, could you please attach sdkconfig files for both of the cases?

DCSBL, 2021-06-24:
@chegewara 
This option is set to `static` and it is the only option indeed. When PSRAM is disabled `dynamic` will become available as well. So this behaviour is intended I think.

@igrr Here you go:

<details><summary>sdkconfig when `ESP32S2_SPIRAM_SUPPORT` is set</summary>

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32s2"
CONFIG_IDF_TARGET_ESP32S2=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0002

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32s2-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_INFO is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=0
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config

[... 2226 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

</details>

I think the most importent question for me is: '**Why** is SPIRAM enabled?`. I can't find an external RAM chip on the [DevKit](https://github.com/espressif/esp-dev-kits/blob/master/esp32-s2-hmi-devkit-1/docs/_static/board_func.jpg). But for having this answered I think I am at the wrong place.

DCSBL, 2021-06-24:
@chegewara 
This option is set to `static` and it is the only option indeed. When PSRAM is disabled `dynamic` will become available as well. So this behaviour is intended I think.

@igrr Here you go:

<details><summary>sdkconfig when `ESP32S2_SPIRAM_SUPPORT` is set</summary>

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32s2"
CONFIG_IDF_TARGET_ESP32S2=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0002

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32s2-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_INFO is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=0
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config

[... 2226 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

</details>

I think the most importent question for me is: '**Why** is SPIRAM enabled?`. I can't find an external RAM chip on the [DevKit](https://github.com/espressif/esp-dev-kits/blob/master/esp32-s2-hmi-devkit-1/docs/_static/board_func.jpg). But for having this answered I think I am at the wrong place.

igrr, 2021-06-24:
> I think the most importent question for me is: 'Why is SPIRAM enabled?`. I can't find an external RAM chip on the DevKit. But for having this answered I think I am at the wrong place.

It should be under the metal lid of the ESP32S2-WROVER module.

Doing a diff of these two sdkconfig files, the only seemingly relevant setting is `CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=32`, but looking at the description it doesn't seem it should affect memory usage when Wi-Fi is not started. Can you try lowering this value when PSRAM is enabled, to see if the memory usage is reduced?

One more thing, could you please explain how these 2 lines are printed/calculated?
```
was: free 95816 allocated 80552 min_free 95796 largest_free_block 80968
180644 - 140600 = 40044 (was 150712 - 95816 = 54896)
```
The numbers after `(was` don't show up in the previous "Totals" outputs, so I'm having trouble trying to interpret them.

DCSBL, 2021-06-24:
> It should be under the metal lid of the ESP32S2-WROVER module.

🤦 I thought that there was only a flash chip underneath it, but yeah, you are right. I've found the external RAM. I will look into it to somehow make use of the external RAM.

> Doing a diff of these two sdkconfig files, the only seemingly relevant setting is `CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=32`, but looking at the description it doesn't seem it should affect memory [...]

Memory usage is lowered when setting `CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=16` when PSRAM is enabled by ~27KB. It is still higher than with PSRAM disabled, but it is something.

> One more thing, could you please explain how these 2 lines are printed/calculated?

These lines after '***' were added by me manually. Sorry for the confusion. `was: free 95816 allocated 80552 min_free 95796 largest_free_block 80968` was copied from step 2 from [this post](https://github.com/espressif/esp-idf/issues/7186#issue-928373839). The calculation is [memory before wifi_init] - [memory after wifi_init] = [estimated memory used by wifi] from both cases.

igrr, 2021-06-24:
> Memory usage is lowered when setting CONFIG_ESP32_WIFI_CACHE_TX_BUFFER_NUM=16 when PSRAM is enabled by ~27KB. It is still higher than with PSRAM disabled, but it is something.

Looks like setting it to 0 will lower it by 54 kB compared to its default setting 32 (when PSRAM is enabled), which is almost exactly the difference you are observing between PSRAM- and non-PSRAM builds  (110-55=55kB). Which seems to explain where the memory is lost when enabling PSRAM!

DCSBL, 2021-06-24:
Seems great to bring in the some expertise 🙂! For the project I work on every KB that can be freed because some setting is 'wrong' is worth it. Maybe it is worth it to see if the value can be lowered without introducing (stability) problems.

MaxwellAlan, 2021-11-26:
hi @DCSBL sorry for reply late.
Of couse the default value can be lowered without introducing problems.
For more detail, you can refer to this link https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/wifi.html#using-psram .
Thanks.

AshUK, 2021-11-29:
@MaxwellAlan are there any suggestions for the following ?

- CACHE_TX_BUFFER_NUM
- MGMT_SBUF_NUM
- TX_BA_WIN

WIFI_DYNAMIC_RX_BUFFER_NUM differ significantly across different IDF versions.

How are TCP_SND_BUF_DEFAULT and TCP_WND_DEFAULT calculated, as they have a dependency on the MSS size.

DCSBL, 2021-12-15:
Hi, Thanks for the information. I was able to increase memory while still having a stable connection. IMO this issue can be closed.

MaxwellAlan, 2021-12-15:
hi @AshUK 
- CACHE_TX_BUFFER_NUM : this option not used when SPIRAM not enable. you can decrease this option.
- MGMT_SBUF_NUM : suggest not modify this option.
- TX_BA_WIN : this option not affect memory or performace so much, you can only keep the value with WIFI_DYNAMIC_TX_BUFFER_NUM
- WIFI_DYNAMIC_RX_BUFFER_NUM differ significantly across different IDF versions: may be we can sync all version with the same value.
- How are TCP_SND_BUF_DEFAULT and TCP_WND_DEFAULT calculated, as they have a dependency on the MSS size: actually this option do not need to consider MSS size, suggest keep this value according WiFi TX/RX Dynamic/Static buffer * 1.6KB / application tcp streams, such as config 16 TX dynamic buffer and application have 2 socket using tcp, then TCP_SND_BUF = 16 * 1.6KB / 2 = 12800 (no more than this value :) )

closed 2022-03-08 by Alvin1Zhang