espressif/esp-idf #7187: C++ build with nimble fails in idf v4.3

justintconroy, 2021-06-23:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32E]
- IDF version:            v4.3
- Build System:         [idf.py]
- Compiler version:  8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32 (git bash)].
- Using an IDE?: [Yes (vscode for development, but I build with idf.py in a terminal)]
- Power Supply:         [USB]

## Problem Description

I just tried upgrading from esp-idf v4.2.1 to esp-idf v4.3. My code doesn't build now. I started getting a lot of `error: invalid conversion from 'int' to 'esp_log_level_t'` now that I wasn't getting before in nimble libraries. I'm building my project as C++.

### Expected Behavior

Code that built in v4.2 should also build in v4.3 without issues.

### Actual Behavior

The build fails with conversion errors that would be fixed by a simple cast (though I'm not really sure where to put that). See build.log below.

## Debug Logs

- [build.log](https://github.com/espressif/esp-idf/files/6704913/build.log)

## Other items if possible

- [sdkconfig.txt](https://github.com/espressif/esp-idf/files/6704915/sdkconfig.txt)

justintconroy, 2021-06-23:
#7167 might be related.

vanwykc, 2021-07-02:
I encountered the same issue and applied this hot fix as a temporary work around until the gurus solve it properly :)

```
/*
 * Licensed to the Apache Software Foundation (ASF) under one
 * or more contributor license agreements.  See the NOTICE file
 * distributed with this work for additional information
 * regarding copyright ownership.  The ASF licenses this file
 * to you under the Apache License, Version 2.0 (the
 * "License"); you may not use this file except in compliance
 * with the License.  You may obtain a copy of the License at
 *
 *  http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing,
 * software distributed under the License is distributed on an
 * "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY
 * KIND, either express or implied.  See the License for the
 * specific language governing permissions and limitations
 * under the License.
 */

#ifndef H_LOG_COMMON_
#define H_LOG_COMMON_

#include "esp_log.h"
#include "log_common/ignore.h"

#ifdef __cplusplus
extern "C" {
#endif

#define LOG_LEVEL_DEBUG    ESP_LOG_DEBUG
#define LOG_LEVEL_INFO     ESP_LOG_INFO
#define LOG_LEVEL_WARN     ESP_LOG_WARN
#define LOG_LEVEL_ERROR    ESP_LOG_ERROR
#define LOG_LEVEL_CRITICAL ESP_LOG_ERROR
#define LOG_LEVEL_NONE     ESP_LOG_NONE
// Up to 7 custom log levels
#define LOG_LEVEL_MAX      (15)

#ifdef __cplusplus
}
#endif

#endif
```

RahulEspressif, 2021-07-02:
Hi @justintconroy  , sorry for delay. Can you please try attached patch and check if it works for you ? 
[error_conversion_macro.txt](https://github.com/espressif/esp-idf/files/6754741/error_conversion_macro.txt)

If you still observe error, kindly share your application code also that causes this issue.

RahulEspressif, 2021-07-02:
> I encountered the same issue and applied this hot fix as a temporary work around until the gurus solve it properly :)
> 
> I changed esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/log_common/log_common.h as follows:
> 
> ```
> /*
>  * Licensed to the Apache Software Foundation (ASF) under one
>  * or more contributor license agreements.  See the NOTICE file
>  * distributed with this work for additional information
>  * regarding copyright ownership.  The ASF licenses this file
>  * to you under the Apache License, Version 2.0 (the
>  * "License"); you may not use this file except in compliance
>  * with the License.  You may obtain a copy of the License at
>  *
>  *  http://www.apache.org/licenses/LICENSE-2.0
>  *
>  * Unless required by applicable law or agreed to in writing,
>  * software distributed under the License is distributed on an
>  * "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY
>  * KIND, either express or implied.  See the License for the
>  * specific language governing permissions and limitations
>  * under the License.
>  */
> 
> #ifndef H_LOG_COMMON_
> #define H_LOG_COMMON_
> 
> #include "esp_log.h"
> #include "log_common/ignore.h"
> 
> #ifdef __cplusplus
> extern "C" {
> #endif
> 
> #define LOG_LEVEL_DEBUG    ESP_LOG_DEBUG
> #define LOG_LEVEL_INFO     ESP_LOG_INFO
> #define LOG_LEVEL_WARN     ESP_LOG_WARN
> #define LOG_LEVEL_ERROR    ESP_LOG_ERROR
> #define LOG_LEVEL_CRITICAL ESP_LOG_ERROR
> #define LOG_LEVEL_NONE     ESP_LOG_NONE
> // Up to 7 custom log levels
> #define LOG_LEVEL_MAX      (15)
> 
> 
> #ifdef __cplusplus
> }
> #endif
> 
> #endif
> ```

Hi @vanwykc  , thanks. Indeed this seems to be the correct conversion. Thanks for your contribution.

justintconroy, 2021-07-07:
The patch fixes the build errors I reported. Unfortunately, I'm still getting build errors from [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp). I'll have to open up a ticket over there. I think this issue can be considered resolved as soon as the above patch is merged in.

rahult-github, 2021-10-01:
Fix is now incorporated and available on github

closed 2021-10-01 by rahult-github