espressif/esp-idf #7191: Guru Meditation Error: LoadProhibited in esp_efuse_mac_get_default

baWedekind, 2021-06-24:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.2.1
- Build System:         CMake
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes, VSCode with ESP-IDF extension
- Power Supply:         USB

## Problem Description

I am using the esp32 to send certain system information to a server via https (over wifi), including its mac address. When I try to fetch the mac address using `esp_efuse_mac_get_default`, I receive a `Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.`  
The same behaviour was observed using `esp_efuse_mac_get_custom` just to make sure.  
I am including code, core dump and stack trace.

### Expected Behavior
The mac address gets written onto the uint8_t * provided.

### Actual Behavior
Call to `esp_efuse_mac_get_default` results in a `LoadProhibited` error in `esp_efuse_get_field_size`.

### Steps to reproduce
Theoretically, simply call `esp_efuse_mac_get_default` or `esp_efuse_mac_get_custom`, although I can imagine these usually work.

My setup and wiring should have no bearing on this issue, but here is an image for the sake of completeness:
[image]

### Code to reproduce this issue

```c
#include <stdio.h>
#include <string.h>
#include <math.h>
#include <sys/time.h>
#include "nvs_flash.h"
#include "esp_sleep.h"

char* build_mutation_request(void)
{
    sint32_t i;
    uint8_t mac[8];
    char mac_str[13];
    char request[1046];

    ESP_ERROR_CHECK(esp_efuse_mac_get_default(mac));
    ESP_LOGI(TAG, "MAC ADDR got");
...
}

sint32_t app_main(void)
{
    int test;
    uint64_t time_slept;
    touch_reading_t *measurement_pointer;
    struct timeval start, wifi, result, end;
    esp_sleep_wakeup_cause_t wakeup_reason;

    gpio_set_direction(GPIO_NUM_25, GPIO_MODE_INPUT);
    gpio_set_direction(GPIO_NUM_26, GPIO_MODE_INPUT);
    ESP_LOGI(TAG, "Stock detect booted up\n");

    // As we use RTC only deep sleep, we must rely on the RTC clock for timing, which measures at 6.6667 us resolution
    // Note: RC oscillator is affected by temperature fluctuations
    // therefore, times are accurate to ms
    gettimeofday(&start, NULL);
    // time slept
[... 148 lines cut ...]
I (623) wifi:wifi certification version: v7.0
I (623) wifi:config NVS flash: enabled
I (633) wifi:config nano formating: disabled
I (633) wifi:Init data frame dynamic rx buffer num: 32
I (633) wifi:Init management frame dynamic rx buffer num: 32
I (643) wifi:Init management short buffer num: 32
I (643) wifi:Init dynamic tx buffer num: 32
I (653) wifi:Init static rx buffer size: 1600
I (653) wifi:Init static rx buffer num: 10
I (663) wifi:Init dynamic rx buffer num: 32
I (663) wifi_init: rx ba win: 6
I (663) wifi_init: tcpip mbox: 32
I (673) wifi_init: udp mbox: 6
I (673) wifi_init: tcp mbox: 6
I (683) wifi_init: tcp tx win: 5744
I (683) wifi_init: tcp rx win: 5744
I (683) wifi_init: tcp mss: 1440
I (693) wifi_init: WiFi IRAM OP enabled
I (693) wifi_init: WiFi RX IRAM OP enabled
I (703) phy_init: phy_version 4660,0162888,Dec 23 2020
I (803) wifi:mode : sta (3c:61:05:4c:43:08)
I (803) TAG: wifi_init_sta finished.
```

As you can see, the crash happens deep in the `esp_efuse_api` and try as I might, I have not been able to find someone with a similar problem, or a cause. I'm hoping this is simpler than I'm making it. I will be grateful for any help.

KonstantinKondrashov, 2021-06-25:
Hi @baWedekind!
I do not think that it is an issue with `esp_efuse_mac_get_default` or `esp_efuse_api `. The code where this happens is straightforward and there is a check of argument on NULL. I suspect that you need to increase the stack size of the main function. 
(Top) → Component config → Common ESP-related -> Main task stack size (default 3584)
Also you can take a look at the `~/esp/esp-idf/examples/system/base_mac_address` example which uses `esp_efuse_mac_get_default`.
Thanks.

baWedekind, 2021-06-25:
Thank you for your input @KonstantinKondrashov, I had seen that example, and from it, it seems that I am not calling `esp_efuse_mac_get_default` at the wrong time, so I checked your suggestion of increasing the main task stack size, although i had already increased it to 35760 to allow for the https client. 
The same error is still happening, in the same place, after increasing the stack size to 45760. I don't think the mac addresses are this big.

It is worth noting that the line where the illegal read happens is `bits_counter += field[i]->bit_count;`, which is after the NULL checks. `EXCVADDR: 0x06750677` also does not seem to suggest a NULL pointer, if I understand correctly.

KonstantinKondrashov, 2021-06-25:
@baWedekind Yes, that I am talking about. esp_efuse_get_field_size goes through ESP_EFUSE_MAC_FACTORY_CRC and counts bits. The last item should be NULL where this function should finish. I think something damage this.

```c
const esp_efuse_desc_t* ESP_EFUSE_MAC_FACTORY_CRC[] = {
    &MAC_FACTORY_CRC[0],    		// CRC8 for factory MAC address
    NULL
};
```

```c

esp_err_t esp_efuse_mac_get_default(uint8_t* mac){
     esp_efuse_read_field_blob(ESP_EFUSE_MAC_FACTORY, mac, 48);
     ...
    uint8_t efuse_crc;
    esp_efuse_read_field_blob(ESP_EFUSE_MAC_FACTORY_CRC, &efuse_crc, 8);
    ...
}

esp_efuse_read_field_blob -> esp_efuse_utility_process -> esp_efuse_get_field_size
```

baWedekind, 2021-06-25:
Is it possible for this to be due to the register hardware breaking?

KonstantinKondrashov, 2021-06-25:
@baWedekind  No, in the sequence of calls `esp_efuse_read_field_blob -> esp_efuse_utility_process -> esp_efuse_get_field_size` there is no reading hw regs yet (regs will be read farther). I think it is a software issue. You can try to call this function from another place to make sure that it works, or even run `esp_efuse_get_field_size(ESP_EFUSE_MAC_FACTORY_CRC)` and expect that it returns 8 as we can see in backtrace (`A2      : 0x00000008`).

baWedekind, 2021-06-25:
@KonstantinKondrashov Interestingly, the error does not happen when i call the function earlier in my code, before wifi has initialized. Could it be something to do with that? Either way, this should already allow me to know the mac address, and pass it to the part of the program where i need it.

baWedekind, 2021-06-25:
Success! This does indeed solve my problem, although ideally i would fetch the mac address once i need it, not on wakeup. Is there some way I can make this possible?

closed 2021-11-29 by Alvin1Zhang