espressif/esp-idf #7193: FASTLED with RainMaker + HomeKit timing issue

codspeed, 2021-06-25:
Hi there, I have been trying to integrate FASTLED with RainMaker+HomeKit, however the animations are not running smooth.
Is anyone aware of any ongoing issues or any speed or memory limitations which may restrict the full functionality of FASTLED.
##Environment
- Module or chip used:  [ESP32-WROOM-32D|
- IDF version: v4.4-dev-479-g1067b2870
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:  macOS
- Using an IDE?: Yes VS Code
- Power Supply:        external 5V
-LED Strip: WS2811
- Length of strip: 100 LEDs
Do let me know if in case anyone requires any further information.
## Problem Description
I am running Animation “Colour Wipe”, i.e. the selected Colour is shown on LED one by one till the complete strip is covered and is again erased one by one. However I can see some glitches i.e. when running colour RED I can see some LEDs flicker which annoys and tells the animation is not smooth.
### Expected Behaviour
The animation should run smooth as it does when only FastLED library is used.
### Actual Behaviour
When integrated with Rainmaker there are visible glitches and animation transition seems to be not fluid.

suda-morris, 2021-06-26:
The RMT driver is highly interrupt-driven, so if you want to drive more LEDs which means you need to transfer more bytes in a transaction, which in result a high possibility that the interrupt doesn't been serviced in time. One way to **reduce** the pain is to increase the hardware memory block used by one RMT channel. You can configure it by increase [mem_block_num](https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/rmt.h#L75) during initialization.

myhomesense, 2021-07-28:
Thank you for this. It seems to have solved most of the issue for us!

Alvin1Zhang, 2021-07-29:
@myhomesense Thanks for sharing the updates. @codspeed Would you please help share if any updates for this issue? Thanks.

codspeed, 2021-07-29:
Hi team,
This seemed to solved the issue. @Alvin1Zhang could we request previous comment be removed? Thanks.

Alvin1Zhang, 2021-07-29:
@codspeed Thanks for sharing the updates, glad to hear that the issue resolved. Could this issue be closed now? Thanks.

codspeed, 2021-07-29:
Yes, please delete comment: https://github.com/espressif/esp-idf/issues/7193#issuecomment-888751591

Thanks.

Alvin1Zhang, 2021-07-29:
@codspeed Have updated, will close, feel free to reopen.

closed 2021-07-29 by Alvin1Zhang