espressif/esp-idf #7194: When using the example_i2s_adc_dac, can't find 'storage'

sosktv, 2021-06-25:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt|Plain Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
When I using the example path:esp-idf\examples\peripherals\i2s_adc_dac, I found that the 'xTaskCreate(example_i2s_adc_dac, "example_i2s_adc_dac", 1024 * 2, NULL, 5, NULL);' did't work, and the terminal shows that 'E (385) ad/da: Partition error: can't find partition name: storage'. It seems that the function'esp_partition_find_first' can't find 'storage'. So I check 'partitions_adc_dac_example.csv', and found that the value of offset of the 'storage' is empty. So i think before I flash this demo, i should change the address of 'storage', but I don't know how to deal with the address of flash. 

//Detailed problem description goes here.

### Expected Behavior
[image]

### Actual Behavior
[image]

```

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

sosktv, 2021-06-25:
I am sorry that i have missing the progress that set the Partition Table by idf.py menuconfig. After Rename the CSV file, the example can successfully run.

Alvin1Zhang, 2021-06-28:
Thanks for reporting and good to hear the issue resolved, feel free to reopen.

closed 2021-06-28 by Alvin1Zhang