espressif/esp-idf #7195: esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/: build failed:Failed to create temporary file...

shengMT, 2021-06-25:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):
    //v4.2.1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt|PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
When compiling the sample, an error occurs.Failed info: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_aead/xchacha20poly1305/sodium/aead_xchacha20poly1305.c.obj: No such file or directory
 The log is as follows:

C:\Users\shengMT\Desktop\esp-idf\examples\bluetooth>cd esp_ble_mesh/ble_mesh_node

C:\Users\shengMT\Desktop\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node>cd onoff_server

C:\Users\shengMT\Desktop\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server>
C:\Users\shengMT\Desktop\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server>

C:\Users\shengMT\Desktop\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server>idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory c:\users\shengMT\desktop\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\users\shengMT\desktop\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server"...
-- Found Git: C:/Users/shengMT/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/shengMT/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/shengMT/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/shengMT/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/shengMT/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/shengMT/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/shengMT/Desktop/esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/sdkconfig
Loading defaults file C:/Users/shengMT/Desktop/esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/sdkconfig.defaults...
-- Found PythonInterp: C:/Users/shengMT/.espressif/python_env/idf4.2_py3.8_env/Scripts/python.exe (found version "3.8.7")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "onoff_server" version: v4.2.1-dirty
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/shengMT/Desktop/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py example_init expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/shengMT/Desktop/esp-idf/components/app_trace C:/Users/shengMT/Desktop/esp-idf/components/app_update C:/Users/shengMT/Desktop/esp-idf/components/asio C:/Users/shengMT/Desktop/esp-idf/components/bootloader C:/Users/shengMT/Desktop/esp-idf/components/bootloader_support C:/Users/shengMT/Desktop/esp-idf/components/bt C:/Users/shengMT/Desktop/esp-idf/components/cbor C:/Users/shengMT/Desktop/esp-idf/components/coap C:/Users/shengMT/Desktop/esp-idf/components/console C:/Users/shengMT/Desktop/esp-idf/components/cxx C:/Users/shengMT/Desktop/esp-idf/components/driver C:/Users/shengMT/Desktop/esp-idf/components/efuse C:/Users/shengMT/Desktop/esp-idf/components/esp-tls C:/Users/shengMT/Desktop/esp-idf/components/esp32 C:/Users/shengMT/Desktop/esp-idf/components/esp_adc_cal C:/Users/shengMT/Desktop/esp-idf/components/esp_common C:/Users/shengMT/Desktop/esp-idf/components/esp_eth C:/Users/shengMT/Desktop/esp-idf/components/esp_event C:/Users/shengMT/Desktop/esp-idf/components/esp_gdbstub C:/Users/shengMT/Desktop/esp-idf/components/esp_hid C:/Users/shengMT/Desktop/esp-idf/components/esp_http_client C:/Users/shengMT/Desktop/esp-idf/components/esp_http_server C:/Users/shengMT/Desktop/esp-idf/components/esp_https_ota C:/Users/shengMT/Desktop/esp-idf/components/esp_https_server C:/Users/shengMT/Desktop/esp-idf/components/esp_ipc C:/Users/shengMT/Desktop/esp-idf/components/esp_local_ctrl C:/Users/shengMT/Desktop/esp-idf/components/esp_netif C:/Users/shengMT/Desktop/esp-idf/components/esp_ringbuf C:/Users/shengMT/Desktop/esp-idf/components/esp_rom C:/Users/shengMT/Desktop/esp-idf/components/esp_serial_slave_link C:/Users/shengMT/Desktop/esp-idf/components/esp_system C:/Users/shengMT/Desktop/esp-idf/components/esp_timer C:/Users/shengMT/Desktop/esp-idf/components/esp_websocket_client C:/Users/shengMT/Desktop/esp-idf/components/esp_wifi C:/Users/shengMT/Desktop/esp-idf/components/espcoredump C:/Users/shengMT/Desktop/esp-idf/components/esptool_py C:/Users/shengMT/Desktop/esp-idf/examples/bluetooth/esp_ble_mesh/common_components/example_init C:/Users/shengMT/Desktop/esp-idf/components/expat C:/Users/shengMT/Desktop/esp-idf/components/fatfs C:/Users/shengMT/Desktop/esp-idf/components/freemodbus C:/Users/shengMT/Desktop/esp-idf/components/freertos C:/Users/shengMT/Desktop/esp-idf/components/heap C:/Users/shengMT/Desktop/esp-idf/components/idf_test C:/Users/shengMT/Desktop/esp-idf/components/jsmn C:/Users/shengMT/Desktop/esp-idf/components/json C:/Users/shengMT/Desktop/esp-idf/components/libsodium C:/Users/shengMT/Desktop/esp-idf/components/log C:/Users/shengMT/Desktop/esp-idf/components/lwip C:/Users/shengMT/Desktop/esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/main C:/Users/shengMT/Desktop/esp-idf/components/mbedtls C:/Users/shengMT/Desktop/esp-idf/components/mdns C:/Users/shengMT/Desktop/esp-idf/components/mqtt C:/Users/shengMT/Desktop/esp-idf/components/newlib C:/Users/shengMT/Desktop/esp-idf [... cut ...]
[1124/1284] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/port/porttimer.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\shengMT\Desktop\esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server>

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

atanisoft, 2021-06-25:
try moving the esp-idf tree from C:\Users\shengMT\Desktop\esp-idf to c:\. Windows has a path length limitation and it appears that you are exceeding that limit.

shengMT, 2021-06-28:
Thank you very much for your help, this issue has been resolved

shengMT, 2021-07-01:
> Thank you very much for your help, this issue has been resolved

I want to get the info of function`s backtrace, so I include the execinfo.h file.But it can not complie and the error info is "fatal error: execinfo.h: No such file or directory".
How can I include this head file or how to get the function`s backtrace?

> try moving the esp-idf tree from C:\Users\shengMT\Desktop\esp-idf to c:. Windows has a path length limitation and it appears that you are exceeding that limit.

I want to get the info of function`s backtrace, so I include the execinfo.h file.But it can not complie and the error info is "fatal error: execinfo.h: No such file or directory".
How can I include this head file or how to get the function`s backtrace?

shengMT, 2021-07-01:
I am sorry,this issue happened again today
issue log is :
[19/93] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.c.obj
FAILED: esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.c.obj
ccache D:\espressif\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DCONFIGURED -DHAVE_WEAK_SYMBOLS -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DNATIVE_LITTLE_ENDIAN -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -Iconfig -I../../../../../../components/libsodium/libsodium/src/libsodium/include -I../../../../../../components/libsodium/port_include -I../../../../../../components/libsodium/libsodium/src/libsodium/include/sodium -I../../../../../../components/libsodium/port_include/sodium -I../../../../../../components/libsodium/port -I../../../../../../components/newlib/platform_include -I../../../../../../components/freertos/include -I../../../../../../components/freertos/xtensa/include -I../../../../../../components/heap/include -I../../../../../../components/log/include -I../../../../../../components/lwip/include/apps -I../../../../../../components/lwip/include/apps/sntp -I../../../../../../components/lwip/lwip/src/include -I../../../../../../components/lwip/port/esp32/include -I../../../../../../components/lwip/port/esp32/include/arch -I../../../../../../components/lwip/port/esp32/tcp_isn -I../../../../../../components/soc/src/esp32/. -I../../../../../../components/soc/src/esp32/include -I../../../../../../components/soc/include -I../../../../../../components/esp_rom/include -I../../../../../../components/esp_common/include -I../../../../../../components/esp_system/include -I../../../../../../components/xtensa/include -I../../../../../../components/xtensa/esp32/include -I../../../../../../components/esp32/include -I../../../../../../components/driver/include -I../../../../../../components/driver/esp32/include -I../../../../../../components/esp_ringbuf/include -I../../../../../../components/efuse/include -I../../../../../../components/efuse/esp32/include -I../../../../../../components/espcoredump/include -I../../../../../../components/esp_timer/include -I../../../../../../components/esp_ipc/include -I../../../../../../components/soc/soc/esp32/include -I../../../../../../components/soc/soc/esp32/../include -I../../../../../../components/soc/soc/esp32/private_include -I../../../../../../components/vfs/include -I../../../../../../components/esp_wifi/include -I../../../../../../components/esp_wifi/esp32/include -I../../../../../../components/esp_event/include -I../../../../../../components/esp_netif/include -I../../../../../../components/esp_eth/include -I../../../../../../components/tcpip_adapter/include -I../../../../../../components/app_trace/include -I../../../../../../components/mbedtls/port/include -I../../../../../../components/mbedtls/mbedtls/include -I../../../../../../components/mbedtls/esp_crt_bundle/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2.1-dirty\" -DESP_PLATFORM -MD -MT esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.c.obj -MF esp-idf\libsodium\CMakeFiles\__idf_libsodium.dir\libsodium\src\libsodium\crypto_pwhash\scryptsalsa208sha256\nosse\pwhash_scryptsalsa208sha256_nosse.c.obj.d -o esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.c.obj   -c D:/esp-idf/components/libsodium/libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.c
ccache: error: Failed to create temporary file for esp-idf\libsodium\CMakeFiles\__idf_libsodium.dir\libsodium\src\libsodium\crypto_pwhash\scryptsalsa208sha256\nosse\pwhash_scryptsalsa208sha256_nosse.c.obj.d: No such file or directory
[32/93] Performing build step for 'bootloader'
ninja: no work to do.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

InfiniteYuan, 2021-09-26:
Please follow the steps below:
1. cd esp-idf
2. git submodule update -f --init --recursive
3. cd project path
4. rm -rf build
5. idf.py build

Alvin1Zhang, 2021-11-29:
@shengMT Thanks for reporting, would you please help share if any updates for the issue? Thanks.

sabdelmalik, 2022-01-19:
I've just had the same failure. Followed the steps above to update esp-idf and the problem is resolved. Thanks.

Alvin1Zhang, 2022-01-20:
@sabdelmalik Thanks for sharing the updates and good to hear the problem is resolved. @shengMT Thanks for reporting, will close due to short of feedback, feel free to reopen with more details.

closed 2022-01-20 by Alvin1Zhang