espressif/esp-idf #7196: Wi-Fi Provisioning Manager example won't scan WPA2+PMF/WPA3 networks properly

RadekHvizdos, 2021-06-25:
## Environment

- Development Kit:      ESP32-DevKitC
- IDF version: v4.2.2 / v4.4-dev-1594-g1d7068e4b
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Android ESP BLE Prov: v2.0.8 - e8734ca

## Problem Description

Wi-Fi Provisioning Manager example won't scan WPA2+PMF/WPA3 networks properly. The networks with PMF enabled are shown as open (without a password), which makes it impossible to provision via the application. The android app will not ask for a password and it will send a blank password to be configured on the target device.
It is possible to provision the device by clicking on "Join Other Network" which bypasses the faulty network scan, the credentials are then properly sent to the device and it is provisioned correctly.

The example code has `CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE=y` enabled so this should work.

### Expected Behavior

Networks with PMF/WPA3 should be shown correctly as encrypted and should prompt for a password to be entered.

### Actual Behavior

Networks with PMF/WPA3 are shown as unencrypted/open which breaks the provisioning process.

### Steps to reproduce

1. Enable WPA2 w/PMF or WPA3 on your WiFi AP
2. Compile and run the Wi-Fi Provisioning Manager example
3. Start ESP BLE Prov app on android device
4. Try to provision a new device, the listed networks with PMF will be shown as open
5. Try to add a network with PMF, the app will not ask for a password and the process will fail

[image]

## Debug Logs

```
I (766) app: Starting provisioning
I (776) phy_init: phy_version 4660,0162888,Dec 23 2020
W (776) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (936) wifi:mode : sta (30:ae:a4:1a:98:14)
I (946) BTDM_INIT: BT controller compile version [ba56601]
I (946) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1196) protocomm_nimble: BLE Host Task Started
I (1206) wifi_prov_mgr: Provisioning started with service name : PROV_1A9814
I (1206) app: Provisioning started
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
W (26916) wifi:Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!

I (43126) app: Received Wi-Fi credentials
        SSID     : REDACTED-iot
        Password :
I (47266) app: Disconnected. Connecting to the AP again...
E (47276) wifi_prov_mgr: STA Disconnected
E (47276) wifi_prov_mgr: Disconnect reason : 15
E (47276) wifi_prov_mgr: STA Auth Error
E (47276) app: Provisioning failed!
        Reason : Wi-Fi station authentication failed
        Please reset to factory and retry provisioning

```
[image]

nishanth-radja, 2021-06-30:
@RadekHvizdos Thank you for raising the issue,can you pls help we with the following details.
1.The AP model which is used in this case.
2.Sniffer capture on the channel on which the wpa3 AP is beaconing.Pls take the capture when the Esp32 is scanning.

Alvin1Zhang, 2021-07-28:
Thanks for reporting and will close due to short of feedback, feel free to reopen with more updates.

closed 2021-07-28 by Alvin1Zhang