espressif/esp-idf #7198: Implement fstat for LwIP VFS

atanisoft, 2021-06-26:
**Is your feature request related to a problem? Please describe.**

The current LwIP VFS code does not implement a handler for `fstat` causing some cross-platform code to fail unnecessarily on the ESP32.

**Describe the solution you'd like**

An implementation that sets the mode flag to show the file descriptor is a socket.

**Describe alternatives you've considered**

Disabling cross-platform validations in the library I'm using is the only option I have at this time.

**Additional context**

This code will fail on the ESP32:
```
int fd = socket(PF_INET, SOCK_STREAM, IPPROTO_TCP);
struct stat stat;
fstat(fd, &stat);
assert(S_ISSOCK(stat.st_mode));
```

david-cermak, 2021-08-18:
@atanisoft Thanks for this feature request! This would help when writing more platform agnostic code.

Although `fstat()` sets other fields, as well, we're mostly interested in the file descriptor type, so the `st_mode`. The simplest solution would be:
[lwip-Add-partial-fstat-vfs-support-to-set-st_mode.patch.txt](https://github.com/espressif/esp-idf/files/7008254/lwip-Add-partial-fstat-vfs-support-to-set-st_mode.patch.txt)

This fixes the sample code above, but maybe there's more... Do you see any trouble with this partial/naive implementation?

atanisoft, 2021-08-18:
@david-cermak I think the minimal implementation is fine to start with. It can be extended if/when other issues come up.

closed 2021-08-24 by espressif-bot