espressif/esp-idf #7199: Using critical sections generates `Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).`

neilbalch, 2021-06-28:
## Environment

- Development Kit: [ESP32-S2-DevKitM-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-devkitm-1-v1.html)
- Module or chip used: [ESP32-S2-MINI-1](https://www.espressif.com/sites/default/files/documentation/esp32-s2-mini-1_esp32-s2-mini-1u_datasheet_en.pdf)
- IDF version (run ``git describe --tags`` to find it): Unknown / not clear (_maybe `esp32s2-rc4-20191025`?_)
- Build System: PlatformIO with the Arduino framework
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): Unknown / not clear
- Operating System: Windows 10
- Using an IDE?: Platform IO in VSCode
- Power Supply: USB

## Problem Description

I’m trying to follow [this tutorial](https://www.youtube.com/watch?v=C2xKhxROmhA&list=PLEBQazB0HUyQ4hAPU1cJED6t3DU0h34bz&index=11) explaining how to avoid priority inversion using critical sections. As such, I'm working with the code [here](https://github.com/ShawnHymel/introduction-to-rtos/blob/main/11-priority-inversion/esp32-freertos-11-solution-critical-section/esp32-freertos-11-solution-critical-section.ino).

### Expected Behavior

As seen in [the video](https://www.youtube.com/watch?v=C2xKhxROmhA&list=PLEBQazB0HUyQ4hAPU1cJED6t3DU0h34bz&index=11) at 11:18, I'd expect the priority inversion to be gone, with "Task H" taking <<5s to wait for the lock.

### Actual Behavior

However, something I haven't been able to trace completely is making the processor crash and reset with the following output:

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x424
load:0x4004c000,len:0x844
load:0x40050000,len:0x2460
entry 0x4004c180
�
--- FreeRTOS Priority Inversion Spin Lock / Critical Section ---
Task L trying to take lock...
Task L got lock. Spent 0 ms waiting for lock. Doing some work...
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40080ff4  PS      : 0x00060d34  A0      : 0x00000000  A1      : 0x3ffc7730
A2      : 0x3ffc0844  A3      : 0x000003eb  A4      : 0x3ffbe8d4  A5      : 0x3ffbe8c8  
A6      : 0x0000000a  A7      : 0x3f000120  A8      : 0x000000fa  A9      : 0x3ffc7710
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x4008136c  A13     : 0x3ffc70d0  
A14     : 0x3f4c0058  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4008136c  LEND    : 0x3ffc70d0  LCOUNT  : 0x40026008

Backtrace:0x40080ff1:0x3ffc7730

ELF file SHA256: 0000000000000000

Rebooting...
```

Please let me know if this is insufficient information to help me with this issue. Thanks!

Spritetm, 2021-06-28:
The issue here is that in a RTOS, holding a critical section for any non-trivial amount of time is generally seen as a Bad Thing, as it stops pre-emption from happening. (Interrupts won't get fired, task switches don't happen, in other words any sort of direct response to events will not happen anymore.) As such, we assume that if this happens for any non-trivial amount of time, something horrible has gone wrong, and we need to protect against this using a watchdog which resets the system to get it back into a working state. Well, holding a critical section for a non-trivial time is exactly what you're doing, and a reset caused by the interrupt watchdog is exactly what you're seeing.

projectgus, 2021-06-29:
Hi @neilbalch,

To follow up on @Spritetm's comment:

If you're sure that you want to spend this time with interrupts disabled then you can [disable the interrupt watchdog in your project config](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-int-wdt). For a simple learning project this is probably fine, if you enter a critical section for a very long period in a more complex project (especially one with BT or Wi-Fi) then things will probably not work as desired.

Alternatively, you can modify your example to decrease the amount of time it spends in each critical section.

You can read more about the Watchdog feature in the [Watchdogs section of the ESP-IDF Programmers Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html). (The "Interrupt Watchdog" is an ESP-IDF feature not a FreeRTOS feature.)

I'm going to close this as it's not a bug in ESP-IDF, if you have more questions about managing critical sections in ESP-IDF then suggest posting on our discussion forum https://esp32.com

closed 2021-06-29 by projectgus

neilbalch, 2021-06-29:
Oh, interesting! As you've probably guessed, I'm quite new to FreeRTOS and the ESP32, so this is all new to me and I really appreciate your calm and detailed answers. Do either of you have any idea why the code I linked to would have worked for him on a vanilla ESP-IDF configuration but mine didn't?

Spritetm, 2021-06-29:
I honestly have no idea. If he hangs up the CPU in a critical section for 250ms, he also should have had the interrupt watchdog cause a reset.