espressif/esp-idf #7203: Quickly connecting two devices via BT SPP creates a timer collision

m-bruck, 2021-06-28:
- Development Kit: none
- Module or chip used:  ESP32
- IDF version: v4.3
- Build System: idf.py
- Compiler version: 2020r3
- Operating System: Windows, ESP Command Prompt
- Using an IDE?: No

This happens when connecting from two devices roughly at the same time.

Original error message:
```
E (489389) BT_OSI: alarm_set failed to start timer, err 0x103
E (489409) BT_OSI: alarm_set failed to start timer, err 0x103
```

These timers are set in `bta_dm_adjust_roles()` by
`bta_sys_start_timer(&bta_dm_cb.switch_delay_timer[i], 0, 500);`

`bta_dm_adjust_roles()` seems to process all the devices each time it is called and doesn't check if that timer is already active.

closed 2021-08-24 by espressif-bot

pad52, 2022-11-04:
please backport this fix in 4.3 branch

BetterJincheng, 2022-11-24:
@pad52  Already backported by 30660bd6e06f23165f5da8496dfcb25bd72ac642