espressif/esp-idf #7204: Build failed on dirent.h inclusion (C linkage, seems need linkage guard )

kangbudz, 2021-06-29:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER|ESP32]
- IDF version (run ``git describe --tags`` to find it):
     v4.4-dev-1850-gbf16c0d82b
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No)]
- Power Supply:         [USB]

## Problem Description

Build failed with error error: conflicting declaration of 'int closedir(DIR*)' with 'C' linkage
components/newlib/platform_include/sys/dirent.h
Problem occured on dirent.h inclusion

### Expected Behavior
No Error
### Actual Behavior
Error on Build
### Steps to reproduce
Just Build

### Code to reproduce this issue

```cpp
#include "esp_vfs.h"
#include "dirent.h"

#ifdef CONFIG_IDF_TARGET_ESP32
#include "driver/sdmmc_host.h"
#endif

```

## Debug Logs

```
In file included from ../components/LogStorage/SDCard.cpp:22:
c:\users\fmlxfirmware\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\dirent.h:72:7: error: conflicting declaration of 'void seekdir(DIR*, long int)' with 'C' linkage
```

## Other items if possible

Add C linkage guard on dirent.h solve the problem
Modify dirent.h files

```cpp
#pragma once

#include <stddef.h>
#include <stdint.h>

#ifdef __cplusplus
extern "C" {
#endif

/**
 * This header file provides POSIX-compatible definitions of directory
 * access functions and related data types.
 * See http://pubs.opengroup.org/onlinepubs/7908799/xsh/dirent.h.html
 * for reference.
 */

/**
 * @brief Opaque directory structure
 */
typedef struct {
    uint16_t dd_vfs_idx; /*!< VFS index, not to be used by applications */
    uint16_t dd_rsv;     /*!< field reserved for future extension */
    /* remaining fields are defined by VFS implementation */
} DIR;

/**
 * @brief Directory entry structure
 */
struct dirent {
    int d_ino;          /*!< file number */
    uint8_t d_type;     /*!< not defined in POSIX, but present in BSD and Linux */
#define DT_UNKNOWN  0
#define DT_REG      1
#define DT_DIR      2
#if __BSD_VISIBLE
#define MAXNAMLEN 255
    char d_name[MAXNAMLEN+1];   /*!< zero-terminated file name */
#else
    char d_name[256];
#endif
};

DIR* opendir(const char* name);
struct dirent* readdir(DIR* pdir);
long telldir(DIR* pdir);
void seekdir(DIR* pdir, long loc);
void rewinddir(DIR* pdir);
int closedir(DIR* pdir);
int readdir_r(DIR* pdir, struct dirent* entry, struct dirent** out_dirent);

#ifdef __cplusplus
}
#endif

```

edouardreg, 2021-07-05:
Hello, I have the same problem. My inclusions use dirent.h in .c ans .cpp files...

kangbudz, 2021-07-05:
add Cplusplus guard to fix the issue for build in cpp

edouardreg, 2021-07-05:
Thanks a lot it's OK now !

void-spark, 2021-07-06:
I'm recently getting "undefined reference to `opendir(char const*)'", could this be related somehow?

It's in a c++ file with these includes:
// VFS with SD over SPI
#include "driver/sdspi_host.h"
#include "esp_vfs.h"
#include "esp_vfs_fat.h"
#include "sdmmc_cmd.h"

void-spark, 2021-07-07:
Could it be https://github.com/espressif/esp-idf/blob/master/components/fatfs/vfs/vfs_fat.c also needs sys/dirent.h instead on dirent.h? (Related to https://github.com/espressif/esp-idf/commit/b14830c5c044561b49fd6da12624701362e85f7c#diff-94e5d65f3e0877ca06ebeffb4b561df0d1e41fe7bff2704782478451997d7bde)

void-spark, 2021-07-07:
Well, I have my answer, it's the same issue.
Only I thought https://github.com/espressif/esp-idf/blob/master/components/newlib/platform_include/sys/dirent.h had/would have guards, but it doesn't. Adding guards in that file solved my issue (but of course there should be a fix in IDF)

closed 2021-07-16 by espressif-bot

justintconroy, 2021-09-09:
This is still broken in release v4.3.1 from today. It looks like the fix didn't get included in that release...

igrr, 2021-09-09:
@justintconroy Sorry about that! Have just merged, should appear in release/v4.3 branch soon.

justintconroy, 2021-09-09:
@igrr Can we expect a v4.3.2 release soon?