espressif/esp-idf #7205: esp-netif_lwip-ppp: pppos_input_tcpip failed with -1

achao1104, 2021-06-29:
I transplanted https://github.com/zztiger123/idf-simple-demo/tree/main/example/wifi/sta_ap code to pppos example, and most of part works well, but it raised exception: esp-netif_lwip-ppp: pppos_input_tcpip failed with -1

zztiger123, 2021-06-29:
pppos_input_tcpip failed with -1， -1 may refer to Out of memory error, could you please call esp_get_free_heap_size() to check the free heap?

achao1104, 2021-06-29:
I checked by calling esp_get_free_heap_size() as below, there are still 2070652 bytes left when the erro raised, BTW, I am using esp32-s2-wrover and pppos sample(no other business codes)
void esp_netif_lwip_ppp_input(void *ppp_ctx, void *buffer, size_t len, void *eb)
{
    struct lwip_ppp_ctx * obj = ppp_ctx;
    err_t ret = pppos_input_tcpip(obj->ppp, buffer, len);

    if (ret != ERR_OK) {
    	printf("a2 %u.",esp_get_free_heap_size());
        ESP_LOGE(TAG, "pppos_input_tcpip failed with %d", ret);
    }
}

david-cermak, 2021-07-01:
Hi @achao1104 

Please note the the `pppos_input_tcpip()` is trying to allocate a pbuf (=lwip structure to hold packets) from a memory pool rather than the heap.
If I understand the application correctly, it sets up an AP to provide the pppos connectivity from a modem device. So I think this setup might indeed pose a significant load to the network (especially when the AP's clients opening browser windows) and generate a lot of pbufs. There might be some ways to mitigate the problem by e.g.:
* allocate bigger pbufs by setting the rx timeout to a value higher https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/components/modem/src/esp_modem.c#L514
* increase the pbuf mempool size by setting `PBUF_POOL_SIZE`

achao1104, 2021-07-15:
The change I made :
1) esp-idf-v4.2.1\components\lwip\lwip\src\include\lwip\opt.h 
    PBUF_POOL_SIZE 400
2) protocols\pppos_client\components\modem\src\esp_modem.c
   res = uart_set_rx_timeout(esp_dte->uart_port, 100);
It looks better but still reports error as  below:
E (60164) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (60254) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (60264) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (61974) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (61984) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1

david-cermak, 2021-07-23:
@achao1104 I guess we cannot help much here,  this setup simply works very slow as the pppos clients talks 115kbaud to the modem. We could give some guidance on what you could try to mitigate these failures. You can also perhaps increase the baud rate of the pppos session and move the UART handlers to RAM, which should help, too, but would never work smooth.
Anyway, if the lwip ppp fails to allocate the input pbuf, it's going to try again later, so the connection should work in the end.

leeebo, 2021-07-31:
@david-cermak Same problem `esp-netif_lwip-ppp: pppos_input_tcpip failed with -1`, tried to increase PBUF_POOL_SIZE not help. **But when enable PSRAM and `Try to allocate memories of WiFi and LWIP in SPIRAM firstly. If failed, allocate internal memory` it works** (using default PBUF_POOL_SIZE), no error message anymore. 

Can you help explain it? ( From [lwip options header](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/lwipopts.h#L86), lwip stack in IDF seems using malloc/free/realloc provided by your C-library instead of the lwip internal allocator, right?  )

achao1104, 2021-08-03:
> @achao1104 I guess we cannot help much here, this setup simply works very slow as the pppos clients talks 115kbaud to the modem. We could give some guidance on what you could try to mitigate these [...]

After increasing the baud rate, the issue seems changed as below, could you help?
W (88056) esp-modem: HW FIFO Overflow
W (88066) esp-modem: HW FIFO Overflow
W (88076) esp-modem: HW FIFO Overflow
W (88086) esp-modem: Ring Buffer Full
W (88096) esp-modem: Ring Buffer Full
E (88096) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (88106) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
W (88106) esp-modem: Ring Buffer Full
E (88116) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (88116) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
W (88136) esp-modem: Ring Buffer Full
W (88236) esp-modem: Ring Buffer Full
W (88246) esp-modem: Ring Buffer Full
W (88266) esp-modem: Ring Buffer Full
W (88286) esp-modem: Ring Buffer Full
W (88356) esp-modem: HW FIFO Overflow
W (88366) esp-modem: HW FIFO Overflow

david-cermak, 2021-08-03:
@leeebo I would just guess that allocating from PSRAM might slow the AP side just a little bit to kind of load-balance the NAT-ing between the two very different bandwidths. Otherwise I'm afraid I cannot explain this, since the ppp frames are allocated from the pool.
Yes, the lwip uses system malloc/free (by means of `MEM_MALLOC()` macros which translates to the malloc with configured capabilities, that preferred PSRAM in your config settings), but from the `heap` component.

@achao1104 In this setup I'd suggest adjusting the UART queue sizes and lowering the rx timeout (this goes against the first requirement though), but most of all, moving the UART ISR to IRAM should help here.

david-cermak, 2021-08-30:
@achao1104 Any update? Did the [`CONFIG_UART_ISR_IN_IRAM`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-uart-isr-in-iram) help? Can we close the issue?

closed 2021-08-30 by achao1104

leeebo, 2023-04-17:
@david-cermak @achao1104 I found the `esp-netif_lwip-ppp: pppos_input_tcpip failed with -1` happends in my case caused by `sys_mbox_trypost` in `tcpip_inpkt` sometimes return `ERR_MEM`, increase `LWIP_TCPIP_RECVMBOX_SIZE` from 32 to 64 can help.

boraozgen, 2023-09-04:
@david-cermak I could not find a better place to discuss this, please direct me if this does not belong here.

I am evaluating the Wi-Fi to LTE routing using the ap_to_pppos example. I would like to achieve the maximum throughput over this setup. When I perform a speed test on the client, I get the previously mentioned `pppos_input_tcpip failed with -1` errors (only during the upload, download works fine). I could not get rid of the errors by increasing `LWIP_TCPIP_RECVMBOX_SIZE` to 64. What other parameters can be tweaked to increase the performance of the lwip stack?

Furthermore I tried increasing the baud rate to achieve higher throughput (flow control enabled). I observed a linear increase up to 0.25 Mbps with 460800 baud. When I increase the baud rate further, the throughput drops to <0.20 Mbps. Do you have an idea why this could happen? The CPU might be getting overloaded, but I am not sure since I have not yet set up the debugger to trace the application. Did you test the throughput limits of the AP to PPPoS routing? What are the possible bottlenecks?

david-cermak, 2023-09-11:
> I could not find a better place to discuss this

Performance of IDF's PPP channel is an important topic. User/modem specific layers around PPPoS are developed separately in https://github.com/espressif/esp-protocols, as `esp_modem` component. You can create an issue there.

I've already created an `iperf` [test application](https://github.com/espressif/esp-protocols/pull/354) and the results show that on 921600 bauds, we get get around 0.7Mbits (throughput decreases with baudrate, as expected, e.g. on 460800, I was at about 0.3Mbits)

> I am evaluating the Wi-Fi to LTE routing using the ap_to_pppos example

I also tested this example (with two different modems) and got very different results (~0.12 Mbits on BG96 and ~0.24 on SIM7600), but testing with linux `pppd` with masquerading internet access showed around 0.65 Mbits on the ESP32's AP side.
Didn't find anything wrong or suspicious with the application itself, no thread starving or consuming  too much CPU's time. I'd expect the `ap_to_pppos` NAT address translation would take some time, but it's still negligible on such low network speeds.
We could route packets via L2, using bridges or simple forwarders, but the L3/NAT doesn't seem to be the bottleneck.

> I could not get rid of the errors by increasing LWIP_TCPIP_RECVMBOX_SIZE

Increasing UART task priority might help, or you can configure lwIP to use core-locking and accept packets by directly from input threads enabling [`CONFIG_LWIP_TCPIP_CORE_LOCKING_INPUT`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-tcpip-core-locking-input)

> What are the possible bottlenecks?

I'd still like to investigate why the throughput is different on real devices, compared to a generic `pppd` server. Maybe it's just something with the device settings, like Quality of Service, Bands, Access technology, etc.

AD-Codex, 2023-09-20:
using EC25-E with esp32-s3 also gave the same error " esp-netif_lwip-ppp: pppos_input_tcpip failed with -1 " . Also speed is only 8kbps. Any improvements for this issue