espressif/esp-idf #7206: ETH2AP and MQTT connection using TCP/IP at same time

sowmimalu, 2021-06-29:
Hai,
Module or chip used: ESP32-WROVER-B
IDF version: v4.1
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Windows
 I have been testing ethernet bridging using the eth2ap code sample , working with ETHERNET LINK UP and also establishing a AP MODE - able to  mobile and other devices with AP MODE.
by defualt in ETH2AP example not initialized any TCP/IP 
but same time i am trying to use MQTT also , i had added  TCP/IP  api  [attached the functions below which i have modified]
 After adding of TCP /IP  - AP MODE connection is not getting IP showing obtaining ip address. not connecting.
MQTT is  connecting. is there any way to connect AP mode at same time by enabling TCP/IP stack ?

#include <string.h>
#include <stdlib.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_eth.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "esp_private/wifi.h"
#include "driver/gpio.h"
#include "esp_netif.h"
//mqtt
#include "mqtt_client.h"

static const char *TAG = "eth_example";
static esp_eth_handle_t s_eth_handle = NULL;
static xQueueHandle flow_control_queue = NULL;
static bool s_sta_is_connected = false;
static bool s_ethernet_is_connected = false;
static uint8_t s_eth_mac[6];

#define FLOW_CONTROL_QUEUE_TIMEOUT_MS (100)
#define FLOW_CONTROL_QUEUE_LENGTH (40)
#define FLOW_CONTROL_WIFI_SEND_TIMEOUT_MS (100)

typedef struct {
    void *packet;
    uint16_t length;
} flow_control_msg_t;

esp_mqtt_client_handle_t client;
esp_netif_t *esp_netif;

char *payload = {"hello welcome"};

static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event);

esp_mqtt_client_config_t mqtt_cfg = { .uri = "mqtt://"xx.xx.xx.xx,

		.username = "", .password = "", .event_handle = mqtt_event_handler,
		};
void tw_mqtt_start();
#define PIN_PHY_POWER 2
esp_netif_t *eth_netif;

// Forward packets from Wi-Fi to Ethernet
static esp_err_t pkt_wifi2eth(void *buffer, uint16_t len, void *eb)
{
    if (s_ethernet_is_connected) {
        if (esp_eth_transmit(s_eth_handle, buffer, len) != ESP_OK) {
            ESP_LOGE(TAG, "Ethernet send packet failed");
        }
    }
    esp_wifi_internal_free_rx_buffer(eb);
    return ESP_OK;
}
[... 280 lines cut ...]
}

void app_main(void)
{
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
         ESP_ERROR_CHECK(ret);
    // Initialize TCP/IP network interface (should be called only once in application)
       ESP_ERROR_CHECK(esp_netif_init());
       ESP_ERROR_CHECK(esp_event_loop_create_default());

     esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
     eth_netif = esp_netif_new(&cfg);
     // Set default handlers to process TCP/IP stuffs
     ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));

    ESP_ERROR_CHECK(initialize_flow_control());

    initialize_ethernet();
    initialize_wifi();
    xTaskCreate(mqtt_task, "mqtt_task", 4096, NULL, 5, NULL);
}

david-cermak, 2021-07-09:
Hi @sowmimalu 

To achieve this, you'd have to setup some kind of switching from the ethernet interface to either AP or the internal network stacks. This could be implemented either on the network layer in the TCP/IP stack directly, or directly on the link layer, similar to the `eth2ap` example forking the packets to and from the ethernet driver based on some rules. The former is easier, the latter would be faster.

### Layer 2 option

This might be a bit difficult to implement, but should work faster, causing minimal performance drops compared to the original `eth2ap` example. The principle is to combine the basic ethernet example with the `eth2ap`, and creating a custom `eth-netif-glue` layer (based on [this](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_netif_glue.c)), which would route packets either to the wifi AP, or the network stack. The complex part comes when defining the switching rules. The naive approach I think would be to pass all packets from the netif to the ethernet and filter the incoming packets based on the HW address, routing the netif HW address to the TCPIP stack and everything else to the wifi AP, but passing the broadcasts and multicasts to both.

### Layer 3 option

Everything is ready for this option, just need to enable
```
CONFIG_LWIP_L2_TO_L3_COPY=y
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y
```
and setup and enable the napt in your application (you can check [this](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network) example to see how the napt could be configured and enabled).
This option is available only from `v4.2`, so it's not an option if you want to stick with `v4.1`. This also means that all packets would have to come and go over the network stack (to have their IP header translated) so this option would be much slower.

sowmimalu, 2021-07-09:
Thanks for the reply... is Layer 3 option available on  v4.2 or/ both options i have to try on v4.2 ?

in IDF 4.1 - used the below options for STA+AP mode and succesfully connected with internet access.
CONFIG_LWIP_L2_TO_L3_COPY=y
CONFIG_LWIP_IP_FORWARD=y
#CONFIG_LWIP_IPV4_NAPT=y

Layer 2 option
This might be a bit difficult to implement, but should work faster, causing minimal performance drops compared to the original eth2ap example. The principle is to combine the basic ethernet example with the eth2ap, and creating a custom eth-netif-glue layer (based on this), which would route packets either to the wifi AP, or the network stack. The complex part comes when defining the switching rules. The naive approach I think would be to pass all packets from the netif to the ethernet and filter the incoming packets based on the HW address, routing the netif HW address to the TCPIP stack and everything else to the wifi AP, but passing the broadcasts and multicasts to both.

sowmimalu, 2021-07-09:
Thanks for the reply... is Layer 3 option available on v4.2 or/ both options i have to try on v4.2 ?

in IDF 4.1 itself - used the below options for STA+AP mode and succesfully connected with internet access.
CONFIG_LWIP_L2_TO_L3_COPY=y
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y

Are you suggesting the same for ETH2AP ?

Layer 2 option
This might be a bit difficult to implement, but should work faster, causing minimal performance drops compared to the original eth2ap example. The principle is to combine the basic ethernet example with the eth2ap, and creating a custom eth-netif-glue layer (based on this), which would route packets either to the wifi AP, or the network stack. The complex part comes when defining the switching rules. The naive approach I think would be to pass all packets from the netif to the ethernet and filter the incoming packets based on the HW address, routing the netif HW address to the TCPIP stack and everything else to the wifi AP, but passing the broadcasts and multicasts to both.

 The principle is to combine the basic ethernet example with the eth2ap ---- have combined the functionalities and tried and attached the file in above mentioned query. https://github.com/espressif/esp-idf/files/6733885/Eth2ap.txt is this the correct way ?

sowmimalu, 2021-07-09:
Thanks for the reply... is Layer 3 option available on v4.2 or/ both options i have to try on v4.2 ?

in IDF 4.1 itself - used the below options for STA+AP mode and succesfully connected with internet access.
CONFIG_LWIP_L2_TO_L3_COPY=y
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y

Are you suggesting the same for ETH2AP ?

Layer 2 option
This might be a bit difficult to implement, but should work faster, causing minimal performance drops compared to the original eth2ap example. The principle is to combine the basic ethernet example with the eth2ap, and creating a custom eth-netif-glue layer (based on this), which would route packets either to the wifi AP, or the network stack. The complex part comes when defining the switching rules. The naive approach I think would be to pass all packets from the netif to the ethernet and filter the incoming packets based on the HW address, routing the netif HW address to the TCPIP stack and everything else to the wifi AP, but passing the broadcasts and multicasts to both.

The principle is to combine the basic ethernet example with the eth2ap ---- have combined the functionalities and tried and attached the file in above mentioned query. https://github.com/espressif/esp-idf/files/6733885/Eth2ap.txt is this the correct way ?

Thanks,

david-cermak, 2021-07-09:
@sowmimalu 

> is Layer 3 option available on v4.2 or/ both options i have to try on v4.2 ?

The layer 3 option is available from `v4.2` and higher (since NAPT was the new feature on 4.2), layer 2 option should work on `v4.1`, too.

> The principle is to combine the basic ethernet example with the eth2ap ---- have combined the functionalities and tried and attached the file in above mentioned query. https://github.com/espressif/esp-idf/files/6733885/Eth2ap.txt is this the correct way ?

Almost, but you'd have to implement your own ethernet glue layer.

Please note the important section about bridging mentioned above:

> the complex part comes when defining the switching rules.

sowmimalu, 2021-07-12:
thanks for the response,
esp_eth_ioctl(s_eth_handle, ETH_CMD_S_PROMISCUOUS, (void *)true); 
esp_netif_attach(eth_netif, esp_eth_new_netif_glue(s_eth_handle)); 
esp_eth_update_input_path(s_eth_handle, pkt_eth2wifi, (void *)true); //change the input path to fn pkt_eth2wifi
esp_eth_start(s_eth_handle);

followed the steps after ethernet driver install ,mqtt gets connected but unable to get Access point to connect with device / or mobile. attached the packets  to the input path of ethernet. how to route the packets to TCPIP stack ?

sowmimalu, 2021-07-13:
hai ,
 i had tried out idf 4.3 by enabling the following config
CONFIG_LWIP_L2_TO_L3_COPY=y
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y
and  by enabling the NAPT in my appllication.

ESP_ERROR_CHECK(esp_eth_driver_install(&config, &s_eth_handle));
   esp_eth_ioctl(s_eth_handle, ETH_CMD_S_PROMISCUOUS, (void *)true);
  esp_netif_attach(eth_netif, esp_eth_new_netif_glue(s_eth_handle));
	#if IP_NAPT
  u32_t napt_netif_ip = 0xC0A80101; // Set to ip address of softAP netif (Default is 192.168.4.1)
  ip_napt_enable(htonl(napt_netif_ip), 1);
  ESP_LOGI(TAG, "NAT is enabled");
     #endif
 //esp_eth_update_input_path(s_eth_handle, pkt_eth2wifi, (void *)true); //change the input path to fn pkt_eth2wifi
	esp_eth_start(s_eth_handle);

but still not connecting MQTT what i have missed here ?

sowmimalu, 2021-07-21:
could you share few information on these , awaiting for reply!

Thanks

david-cermak, 2021-07-22:
Hi @sowmimalu 

Sorry for the late reply. To enable the NAPT, please add this code to the `app_main()` of the basic eth example,after getting the IP:
```cpp
    xEventGroupWaitBits(event_group, GOT_IP_EVENT, pdTRUE, pdTRUE, portMAX_DELAY); // set this flag on getting an IP from eth interface

    esp_netif_t *ap_netif = esp_netif_create_default_wifi_ap();
    assert(ap_netif);
    esp_netif_dns_info_t dns;
    ESP_ERROR_CHECK(esp_netif_get_dns_info(eth_netif, ESP_NETIF_DNS_MAIN, &dns));
    set_dhcps_dns(ap_netif, dns.ip.u_addr.ip4.addr);

    wifi_init_softap();
    ip_napt_enable(_g_esp_netif_soft_ap_ip.ip.addr, 1);
```

where:
* ` wifi_init_softap();` is the standard [softAP](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c#L47-L82) init, but don't forget to remove [these](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c#L49-L51) netif and event loop inits, add add [the nvs init](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/softAP/main/softap_example_main.c#L87-L92)
* `set_dhcps_dns()` is defined as:
```cpp
static esp_err_t set_dhcps_dns(esp_netif_t *netif, uint32_t addr)
{
    esp_netif_dns_info_t dns;
    dns.ip.u_addr.ip4.addr = addr;
    dns.ip.type = ESP_IPADDR_TYPE_V4;
    dhcps_offer_t dhcps_dns_value = OFFER_DNS;
    ESP_ERROR_CHECK(esp_netif_dhcps_option(netif, ESP_NETIF_OP_SET, ESP_NETIF_DOMAIN_NAME_SERVER, &dhcps_dns_value, sizeof(dhcps_dns_value)));
    ESP_ERROR_CHECK(esp_netif_set_dns_info(netif, ESP_NETIF_DNS_MAIN, &dns));
    return ESP_OK;
}
```

in addition to that, we have to make the packet's IP header modifiable , and not only for the WiFi, but also for the ethernet interface. It's possible to create a new I/O function and modify the eth-netif, but the easiest way is to apply this patch:
https://github.com/espressif/esp-lwip/issues/20#issuecomment-680812131

I can post the full working project somewhere if it helps.

sowmimalu, 2021-07-22:
thank you so much for the reply..

added in Basic ethernet example  and espressif/esp-lwip#20 (comment) the mentioned patch file to replace instead of existing function if it so i  have added but getting few errors , getting little bit confused of replacing the patch file.

the below is my function gets replaced ..

void ethernetif_input(void *h, void *buffer, size_t len, void *eb)
{
    struct netif *netif = h;
    struct pbuf *p;
	
	  if (p == NULL) {
		  
		  return;
		  
	  }
 p->payload = buffer;
#if ESP_LWIP
     p->l2_owner = netif;
     p->l2_buf = buffer;
#endif

     /* full packet send to tcpip_thread to process */
     if (unlikely(netif->input(p, netif) != ERR_OK)) {
         LWIP_DEBUGF(NETIF_DEBUG, ("ethernetif_input: IP input error\n"));
 err_t ethernetif_init(struct netif *netif)
     netif->output_ip6 = ethip6_output;
 #endif /* LWIP_IPV6 */
     netif->linkoutput = ethernet_low_level_output;
#if !ESP_L2_TO_L3_COPY
     netif->l2_buffer_free_notify = ethernet_free_rx_buf_l2;
#endif // !ESP_L2_TO_L3_COPY
 
     ethernet_low_level_init(netif);
 }

david-cermak, 2021-07-22:
Added the example here https://github.com/david-cermak/eth-ap-nat/tree/main/eth-ap-idf4.3

sowmimalu, 2021-07-22:
Replaced the patch file .... its worked thank you so much !! can i  make use this patch file in idf 4.1 by enabling NAPT .

david-cermak, 2021-07-23:
For setting static IP, please refer to the newly added example `static_ip`, it's configured in [example_set_static_ip()](https://github.com/espressif/esp-idf/blob/master/examples/protocols/static_ip/main/static_ip_example_main.c#L44-L60) .

Your code look more or less correct. The trouble you're seeing might be that the example sets a global dns server, while the AP uses the dns server taken from settings of the ethernet settings (provided you've followed the [code shared here](https://github.com/espressif/esp-idf/issues/7206#issuecomment-884847375).

> ! can i make use this patch file in idf 4.1
No this won't work on `v4.1`, as the napt feature is not available. The patch only updated the *default* ethernet per our needs, it's not needed if we configure the `esp_netif` to use a *custom* ethernet interface as it's done in the [eth-ap-napt](https://github.com/david-cermak/eth-ap-nat) project (the works on standard -- unpatched  -- `release/v4.3`)

david-cermak, 2021-07-27:
@sowmimalu Added the L2 option for completeness: https://github.com/david-cermak/eth-ap-nat/tree/main/eth-ap-L2

Could you please close the issue if this helps resolve the problem?

sowmimalu, 2021-07-27:
hai  @david-cermak , 
thank you so much ,  L2 layer option on idf 4.1 working  and L3 on idf 4.3 working  ..which would be the best for network speed to be connected on multiple device ?

david-cermak, 2021-07-27:
@sowmimalu As mentioned in [the initial post](https://github.com/espressif/esp-idf/issues/7206#issuecomment-876939873) the L2 option is faster, whilst the L3 option is better for advanced networking (L2 is okay for simple TCP communication, like the mqtt client as required, but won't work great with multicasts and the AP won't see the ETH netifs...)

> best for network speed to be connected on multiple device ?

If multiple devices are stations to be connected to AP, then the same as network performance, e.g. L2 option.

sowmimalu, 2021-07-28:
@david-cermak  Thank you so much ,  closing this now..

closed 2021-07-28 by sowmimalu

pr200sd, 2023-10-23:
@david-cermak Hello. While studying your examples for network interfaces, I learned a lot of interesting things, but I just can’t solve one problem. I need a code example for the following configuration: esp32 connects to the router via STA and gets access to the Internet, and via ETHERNET I want to distribute the Internet, i.e. this is eth2ap just the other way around, but in order to be able to configure ssid and pass I need on the port ethernet or including AP to run a WEB server and also be able to have access to the Internet in addition to the STA-ETH gateway, is it possible to organize this in some way, if so. then I would be grateful for an example similar to those you posted here: https://github.com/david-cermak/eth-ap-nat/

david-cermak, 2023-10-24:
Hi @pr200sd 

You can check this IDF example: https://github.com/espressif/esp-idf/tree/master/examples/network/sta2eth
It seems to me that it has everything you need, including WiFi configuration and/or provisioning.

pr200sd, 2023-10-24:
Thank you, it looks like everything is there, but I suspect that this is for esp-idf > 4.4, I’ll try to figure it out.