espressif/esp-idf #7209: ULP RISCV wakes main CPU immediately after first wakeup

bbrother, 2021-06-29:
## Environment

- Development Kit:      SP32-S2-Kaluga-1 Kit v1.3
- Kit version (for WroverKit/PicoKit/DevKitC): v1.3
- Module or chip used:  ESP32-S2-WROVER
- IDF version: v4.3
- Build System:     idf.py
- Compiler version: 8.4.0
- Operating System:     Linux (using v4.3 docker container)
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
I am trying to get a ULP example working where the ULP runs on a wakeup period and then wakes up the main CPU when certain conditions are satisfied. Eventually it will be based on GPIO but for now I cannot get even a basic simple counter example to work.

On the first iteration it wakes up after the counter reaches the specified value but after that it wakes up immediately continuously. 

This appears to be similar to other issues which I found but it doesn't appear there has ever been a resolution.

### Expected Behavior
It wakes up once after counter reaches 100 and then due to flag does not wake up again

### Actual Behavior
It wakes up once after counter reaches 100 and then continues to wake up almost immediately again and again
### Steps to reproduce

1. start with ulp_riscv example code
2. Replace ulp_riscv_example_main.c with code from gist
3. Replace ulp/main.c with code from gist

### Code to reproduce this issue
https://gist.github.com/bbrother/58b871e9861b2a443e9a94292d8aabc6

## Debug Logs

```
--- idf_monitor on /dev/ttyUSB2 115200 ---              
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---                                                                    
ESP-ROM:esp32s2-rc4-20191025                            
Build:Oct 25 2019                                        
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee       
mode:DIO, clock div:1                          
load:0x3ffe6100,len:0x1290
load:0x4004c000,len:0x8d4                      
load:0x40050000,len:0x2984
entry 0x4004c1a8         
W (75) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Wakeup info 00000000 00000000 00000000 00000000 00000000
Not a ULP-RISC-V wakeup, initializing it! 0000 -557860018 44E481B7                                                                       
Entering in deep sleep 570987                           
                                                                                                                                         
ESP-ROM:esp32s2-rc4-20191025                            
Build:Oct 25 2019                                        
rst:0x5 (DSLEEP),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee       
mode:DIO, clock div:1                          
load:0x3ffe6100,len:0x1290
load:0x4004c000,len:0x8d4                   
load:0x40050000,len:0x2984                         
entry 0x4004c1a8         
W (73) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Wakeup info 00100000 00000000 00000000 00000800 00000000                                                         
ULP-RISC-V woke up the main CPU! 0006 103 00000FEC test 1                                                                                
Entering in deep sleep 3285118                          
                                                                                                                 
ESP-ROM:esp32s2-rc4-20191025                                                                                     
Build:Oct 25 2019                                   
rst:0x5 (DSLEEP),boot:0xa (SPI_FAST_FLASH_BOOT)  
SPIWP:0xee             
mode:DIO, clock div:1                                         
load:0x3ffe6100,len:0x1290                          
load:0x4004c000,len:0x8d4                        
load:0x40050000,len:0x2984                                                
entry 0x4004c1a8                                                           
W (73) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Wakeup info 00100000 00000000 00000000 00000800 00000000
ULP-RISC-V woke up the main CPU! 0006 150 00000FEC test 1
Entering in deep sleep 4857130
```

negativekelvin, 2021-06-29:
https://github.com/espressif/esp-idf/commit/70cab5bd814716436879dabede4977a0e6a73626

projectgus, 2021-07-01:
Hi @bbrother,

Thanks for the detailed report. The fix @negativekelvin links to (for #6654) has been backported to release/v4.3 branch in commit 0e904b3f7e5535a711bc310a6aa1b03c68906323. It will be included in the v4.3.1 release.

In the meantime, to get the fix you have two options:

1. Check out the release/v4.3 prerelease branch. ([explanation about release branches here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html)).
2. `git cherry-pick 0e904b3` in your IDF v4.3 working directory.

bbrother, 2021-07-01:
Hi @projectgus  thank you for the quick response.  @negativekelvin thanks for the patch information.

-Brett

projectgus, 2021-07-01:
Hi @bbrother,

No problems. If you get a chance to try it and it works then please let us know and the issue can be closed.

bbrother, 2021-07-01:
Hi @projectgus,

This did fix the issue but exposed another one.  After approximately 3400 wakeups of the ULP RISCV (without waking up app processor) I get a wakeup of the app processor with reason ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG.  I can close this and file another bug or keep this open.  The code provided in the issue still can create this bug just requires waiting a little longer.

Do you want me to close this and create a new issue?

```
Running idf_monitor in directory /project                                                                                                                                                                   
Executing "/opt/esp/python_env/idf4.3_py3.6_env/bin/python /opt/esp/idf/tools/idf_monitor.py -p /dev/ttyUSB1 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- /project/build/ulp-riscv-example.elf -m '/opt/
esp/python_env/idf4.3_py3.6_env/bin/python' '/opt/esp/idf/tools/idf.py'"...                                                                                                                                 
--- idf_monitor on /dev/ttyUSB1 115200 ---                                                                                                                                                                  
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---                                                                                                                                       
ESP-ROM:esp32s2-rc4-20191025                                                                                                                                                                                
Build:Oct 25 2019                                                                                                                                                                                           
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)                                                                                                                                                            
SPIWP:0xee                                                                                                                                                                                                  
mode:DIO, clock div:1                                                                                                                                                                                       
load:0x3ffe6100,len:0x12a8                                                                                                                                                                                  
load:0x4004c000,len:0x8f0                                                                                                                                                                                   
load:0x40050000,len:0x2984                                                                                                                                                                                  
entry 0x4004c1b0                                                                                                                                                                                            
W (72) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.                                                                   
Wakeup info 00000000 00000000 00000000 00000000 00000000                                                                                                                                                    
Not a ULP-RISC-V wakeup, initializing it! 0000 -1614824625 44A480B7                                                                                                                                         
Entering in deep sleep 549881                                                                                                                                                                               
                                                                                                                                                                                                            
ESP-ROM:esp32s2-rc4-20191025                                                                                                                                                                                
Build:Oct 25 2019                                                                                                                                                                                           
rst:0x5 (DSLEEP),boot:0xb (SPI_FAST_FLASH_BOOT)                                                                                                                                                             
SPIWP:0xee                                                                                                                                                                                                  
mode:DIO, clock div:1                                                                                                                                                                                       
[... cut ...]
Build:Oct 25 2019                                                                                                                                                                                           
rst:0x5 (DSLEEP),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x12a8
load:0x4004c000,len:0x8f0
load:0x40050000,len:0x2984
entry 0x4004c1b0
W (71) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Wakeup info 00100000 00000000 00000000 00002000 00000000
Not a ULP-RISC-V wakeup, initializing it! 000B 688 00000FEC
Entering in deep sleep 17223353
```

projectgus, 2021-07-02:
Hi @bbrother,

Thanks for letting us know about this new issue, that's not something we've seen. If you don't mind putting these details into a new issue, it will make it easier for us to follow up.

bbrother, 2021-07-02:
This was resolved for me with commit 0e904b3

closed 2021-07-02 by bbrother