espressif/esp-idf #7210: driver sdspi

gracianomor, 2021-06-29:
Hello. I'm new in this discouus.

I'am using the driver SDSPI by esp-tool and when I off my device and I up I have this issue:

I (315) spi_flash: detected chip: generic
I (318) spi_flash: flash io: dio
W (322) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (336) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (346) example: Initializing SD card
I (346) example: Using SPI peripheral
I (356) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (366) sdspi_transaction: cmd=52, R1 response: command not supported
I (416) sdspi_transaction: cmd=5, R1 response: command not supported
E (1416) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (1416) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (1416) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (1426) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.

can some one help me, please?

and I have another question.

in this driver, How can I to open a file?

I'm using an task to open the file but I have this issue;

E (1126) example: Failed to open file for writing
E (1136) FreeRTOS: FreeRTOS Task "Prueba_sdcard_t" should not return, Aborting now!

Thank you

dobairoland, 2021-06-30:
Hi @gracianomor. What development board are you using? If it is custom made then can you please share the schematics connecting the SD card to ESP32?

Have you tried the following example?
https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdspi

Please also share the version of ESP-IDF you are using,

tom-borcin, 2021-07-30:
Hi @gracianomor, please help us to resolve this issue by answering the questions @dobairoland asked. Thank you!

pacucha42, 2021-11-01:
@gracianomor - thank you for reporting the issue, glad to know you have solved the issue. Based on the error code, it's related to the card's startup sequence (OCR command should return supported voltage range). It can be hardware incompatibility or something specific to your card - if the issue reappears, please, try different SD card first. It it wouldn't help, reopen this bug report to let us know.

closed 2021-11-01 by pacucha42