espressif/esp-idf #7211: HID close device bug

guo-max, 2021-06-30:
```c++
--- a/components/bt/host/bluedroid/bta/hh/bta_hh_act.c
+++ b/components/bt/host/bluedroid/bta/hh/bta_hh_act.c
@@ -494,8 +494,7 @@ void bta_hh_api_disc_act(tBTA_HH_DEV_CB *p_cb, tBTA_HH_DATA *p_data)
         disc_dat.status = BTA_HH_ERR;

         status = HID_HostCloseDev(disc_dat.handle);

-        if (status) {
+        if (status == HID_SUCCESS) {
             (* bta_hh_cb.p_cback)(BTA_HH_CLOSE_EVT, (tBTA_HH *)&disc_dat);
         }
     }
```
Without error the HID_HostCloseDev() will return HID_SUCCESS, and HID_SUCCESS equal 0. So HID_HostCloseDev() return 0 when success and it never post the close event. 
```C++
enum {
    HID_SUCCESS,
    HID_ERR_NOT_REGISTERED,
    HID_ERR_ALREADY_REGISTERED,
    HID_ERR_NO_RESOURCES,
    HID_ERR_NO_CONNECTION,
    HID_ERR_INVALID_PARAM,
    HID_ERR_UNSUPPORTED,
    HID_ERR_UNKNOWN_COMMAND,
    HID_ERR_CONGESTED,
    HID_ERR_CONN_IN_PROCESS,
    HID_ERR_ALREADY_CONN,
    HID_ERR_DISCONNECTING,
    HID_ERR_SET_CONNABLE_FAIL,
    /* Device specific error codes */
    HID_ERR_HOST_UNKNOWN,
    HID_ERR_L2CAP_FAILED,
    HID_ERR_AUTH_FAILED,
    HID_ERR_SDP_BUSY,
    HID_ERR_GATT,

    HID_ERR_INVALID = 0xFF
};
```

boblane1, 2021-08-20:
Hi @guo-max, 

The function `bta_hh_api_disc_act` only sends the command to the peer device. After receiving the disconnect indication on L2CAP channel in function 'hidh_l2cif_disconnect_ind', it will trigger the state machine and post the close event.

Thanks.

guo-max, 2021-08-20:
Hi @boblane1 ,
So in the function` bta_hh_api_disc_act `,
```
      status = HID_HostCloseDev(disc_dat.handle);
      if (status) {
          (* bta_hh_cb.p_cback)(BTA_HH_CLOSE_EVT, (tBTA_HH *)&disc_dat);
      }
```
If the `status` equal any error, it will post a `BTA_HH_CLOSE_EVT` callback event?

Based on my understanding, I think it should post a `BTA_HH_CLOSE_EVT` event only when peer device successfully closed. 
And when the peer device successfully closed by function  `HID_HostCloseDev(disc_dat.handle);`, the function will return a status `HID_SUCCESS`. But the enum `HID_SUCCESS` equal `0`, 
```
 if (status) {
          (* bta_hh_cb.p_cback)(BTA_HH_CLOSE_EVT, (tBTA_HH *)&disc_dat);
      }
```
it will not trigger the `BTA_HH_CLOSE_EVT` event post. It only trigger the event post when status equal any error.

boblane1, 2021-08-23:
Hi @guo-max,

Yes, your understanding is right! `HID_HostCloseDev` only sends a close request to the peer device, the return status refects the result of sending action.

Thanks.

guo-max, 2021-08-23:
Hi @boblane1 ,
The returned status for successfully closed a peer device is `HID_SUCCESS` which is an **enum** and **equal 0**. 
So it will not trigger the `BTA_HH_CLOSE_EVT` event post by the following code. 
```
 if (status) {
          (* bta_hh_cb.p_cback)(BTA_HH_CLOSE_EVT, (tBTA_HH *)&disc_dat);
      }
```
And my fix is 
```
if (status == HID_SUCCESS) {
             (* bta_hh_cb.p_cback)(BTA_HH_CLOSE_EVT, (tBTA_HH *)&disc_dat);
         }
```

boblane1, 2021-08-24:
Hi @guo-max，

`HID_HostCloseDev` only sends a disconnect request to the peer device, the return status reflects the result of sending action. When the HID host receives the disconnect confirm from the HID device(the details in function `hidh_l2cif_disconnect_cfm`), it will generate the event `BTA_HH_CLOSE_EVT`. So I think it's better not report `BTA_HH_CLOSE_EVT` until receiving the disconnect confirm. How do you think?

Thanks.

guo-max, 2021-08-24:
Hi @boblane1 ,
Thanks for the explanation. 
Here is my understanding :
For any error returned from `HID_HostCloseDev`, post the `BTA_HH_CLOSE_EVT` from 
```
if (status) {
          (* bta_hh_cb.p_cback)(BTA_HH_CLOSE_EVT, (tBTA_HH *)&disc_dat);
      }
```
If `HID_HostCloseDev` returned `HID_SUCCESS `, the `BTA_HH_CLOSE_EVT` will posted from `hidh_l2cif_disconnect_cfm`.

In my case, I did not get the close event for some reasons. And if I changed it to `if (status == HID_SUCCESS)` I can get the close event callback. I will take a look again.

Alvin1Zhang, 2021-11-26:
@guo-max Thanks for reporting, would you please help share if any updates for the issue? Thanks.

guo-max, 2021-11-29:
@Alvin1Zhang I have tested with the current master branch and everything works. Thanks

closed 2021-11-30 by Alvin1Zhang