espressif/esp-idf #7212: Master: Hitting "recursive mutexes make no sense in ISR context" at various places

chipweinberger, 2021-06-30:
## Environment

- Development Kit:      ESP32-S2
- IDF version: v4.4-dev-1849-g8e3e65a47
- Build System:  idf.py
- Compiler version: xtensa-esp32s2-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS
- Using an IDE?:  Visual Code
- Power Supply:  USB

## Problem Description

At various places while running, usually in vprintf, or malloc, I hit this issue where we cannot get a recursive mutex.

*I've also ran this code with Master from April 28th (9d34a1cd), and hit the same issue.*

**What causes this error to happen?** Heap corruption? (I've tried turning heap corruption detection on/off). Running our of memory? (I've tried mallocing almost nothing). Something else?

### Logs

```
`I (25714) pd localtime: settimeofday updated 1645031095
I (25714) pd midi file: create midi file /sd/JAMC/2022/Chip's_Jamcorder__JMX00050__2022-02-16.mid
I (25724) pd midi file: file does not already exist. good.

abort() was called at PC 0x40026e7f on core 0
0x40026e7f: lock_acquire_generic at /Volumes/User/MBP-Google-Drive/superpiano/esp/esp-idf/components/newlib/locks.c:139`
```

### Backtrace
```

`0x400850c3: panic_abort at /esp/esp-idf/components/esp_system/panic.c:379

0x4002daf9: esp_system_abort at /esp/esp-idf/components/esp_system/esp_system.c:126

0x400359e6: abort at /esp/esp-idf/components/newlib/abort.c:46

0x40026e7f: lock_acquire_generic at /esp/esp-idf/components/newlib/locks.c:139

0x40026fc9: _lock_acquire_recursive at /esp/esp-idf/components/newlib/locks.c:167

0x40033373: block_locate_free at /esp/esp-idf/components/heap/heap_tlsf.c:447
 (inlined by) tlsf_malloc at /esp/esp-idf/components/heap/heap_tlsf.c:757

0x40034423: multi_heap_malloc_impl at /esp/esp-idf/components/heap/multi_heap.c:197

0x40025b01: heap_caps_malloc at /esp/esp-idf/components/heap/heap_caps.c:145

0x40025b35: heap_caps_malloc_default at /esp/esp-idf/components/heap/heap_caps.c:177

0x400359f1: malloc at/esp/esp-idf/components/newlib/heap.c:31

0x400be7ac: mem_malloc at /esp/esp-idf/components/lwip/lwip/src/core/mem.c:237

0x400be82b: do_memp_malloc_pool at /esp-idf/components/lwip/lwip/src/core/memp.c:254

0x400be899: memp_malloc at /esp/esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x400bf3b9: pbuf_alloc_reference at /esp/esp-idf/components/lwip/lwip/src/core/pbuf.c:336 (discriminator 2)

0x400bf591: pbuf_alloc at /esp/esp-idf/components/lwip/lwip/src/core/pbuf.c:237

0x400d0092: wlanif_input at /esp/esp-idf/components/lwip/port/esp32/netif/wlanif.c:197

0x401285d5: esp_netif_receive at /esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:858

0x400ba295: wifi_sta_receive at /esp/esp-idf/components/esp_wifi/src/wifi_netif.c:47

0x400357a2: sta_input at ??:?

0x40035855: sta_rx_cb at ??:?

0x40038ddd: ppRxPkt at ??:?

0x4003723b: ppTask at ??:?

0x40030aad: vPortTaskWrapper at /esp/esp-idf/components/freertos/port/xtensa/port.c:168
`

```

chipweinberger, 2021-06-30:
Rolled back to 4.2.2.

With **'Heap corruption detection' = Comprehensive** on, I'm seeing:

**CORRUPT HEAP: Bad tail at 0x3fff34ac. Expected 0xbaad5678 got 0xbaad5600**

Very helpful.

But still leaves a question of why does this stack trace so often fail here? And is there anything we can do to make it more clear to the user that this could be a heap corruption issue? The comment about "ISR context" is super misleading.

atanisoft, 2021-06-30:
@chipweinberger This looks more like a heap corruption issue as the TLSF heap allocator found a suitable block and when it asserted that the block size was larger than the requested size it aborted:
https://github.com/espressif/esp-idf/blob/8e3e65a47b7d9b5dc4f52eb56660a748fda1884e/components/heap/heap_tlsf.c#L445-L449

Are you possibly using a pointer after freeing it?

chipweinberger, 2021-06-30:
It was indeed heap corruption. Strlen on a non-null terminated string caused it :)

We should still add some comments to the code to help people that hit this backtrack.

projectgus, 2021-07-01:
Hi @chipweinberger,

Glad you figured out the root cause.

Totally agree that this is unclear. What's happening is that the heap implementation is detecting corruption and failing an assert while interrupts are disabled, then the assertion failure routine is trying to print something via normal stdout and that is triggering the "Recursive mutexes make no sense in ISR context" crash because it's not valid to call printf() when interrupts are disabled - it's a crash within a crash.

We'll fix this so the error message is not so confusing.

Alvin1Zhang, 2021-08-09:
Thanks for reporting and sorry for slow turnaround, fix is available https://github.com/espressif/esp-idf/commit/7b5f731cce4aae87c4a941bc9737b3ecb9125c0d. Feel free to reopen.

closed 2021-08-09 by Alvin1Zhang

chipweinberger, 2021-08-11:
What was the fix?

baldhead69, 2021-09-30:
Hi,

I'm having a similar problem.

```c
abort() was called at PC 0x40085a07 on core 1
0x40085a07: lock_acquire_generic at C:/esp-idf/components/newlib/locks.c:139

Backtrace:0x400d3a3e:0x3fff03100x4008eb31:0x3fff0330 0x400967fa:0x3fff0350 0x40085a07:0x3fff03c0 0x40085b65:0x3fff03f0 0x40085c01:0x3fff0410 0x40170241:0x3fff0440 0x4016babf:0x3fff0700 0x4016ba3a:0x3fff0750 0x4009484a:0x3fff0780 0x400951c2:0x3fff07a0 0x4008218a:0x3fff07c0 0x40096825:0x3fff07e0 0x4011a3a1:0x3fff0800 0x4011adaa:0x3fff0820 0x4011a385:0x3fff0840 0x400ebf54:0x3fff0860 0x4011c2c1:0x3fff0880 0x4011c305:0x3fff08a0 0x4011c3f4:0x3fff08c0 0x4011c636:0x3fff08e0 0x4011ae95:0x3fff0900 0x4011b062:0x3fff0940 0x4011b0e6:0x3fff0990 0x40092b8e:0x3fff09b0
0x400d3a3e: panic_abort at C:/esp-idf/components/esp_system/panic.c:391

0x4008eb31: esp_system_abort at C:/esp-idf/components/esp_system/esp_system.c:129

0x400967fa: abort at C:/esp-idf/components/newlib/abort.c:46

0x40085a07: lock_acquire_generic at C:/esp-idf/components/newlib/locks.c:139

0x40085b65: _lock_acquire_recursive at C:/esp-idf/components/newlib/locks.c:167

0x4011c305: httpd_sess_clear_ctx at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:251

0x4011c3f4: httpd_sess_delete at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:365

0x4011c636: httpd_sess_close at C:/esp-idf/components/esp_http_server/src/httpd_sess.c:145

0x4011ae95: httpd_process_ctrl_msg at C:/esp-idf/components/esp_http_server/src/httpd_main.c:150 (discriminator 15)

0x4011b062: httpd_server at C:/esp-idf/components/esp_http_server/src/httpd_main.c:215 (discriminator 15)

0x4011b0e6: httpd_thread at C:/esp-idf/components/esp_http_server/src/httpd_main.c:250 (discriminator 15)

0x40092b8e: vPortTaskWrapper at C:/esp-idf/components/freertos/port/xtensa/port.c:168
```
idf version: v4.4-dev-2359-g58022f859

chipweinberger, 2022-12-06:
To answer my own question, this was the fix:

```
/* Overriding assert function so that whenever assert is called from critical section,
 * it does not lead to a crash of its own.
 */
void __attribute__((noreturn)) __assert_func(const char *file, int line, const char *func, const char *expr)
{
#if CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_SILENT
    char buff[sizeof(ASSERT_STR) + 11 + 1] = ASSERT_STR;

    ra_to_str(&buff[sizeof(ASSERT_STR) - 1]);

    esp_system_abort(buff);
#else
    char addr[11] = { 0 };
    char buff[200];
    char lbuf[5];
    uint32_t rem_len = sizeof(buff) - 1;
    uint32_t off = 0;

    itoa(line, lbuf, 10);

    if (!spi_flash_cache_enabled()) {
       if (esp_ptr_in_drom(file)) {
           file = CACHE_DISABLED_STR;
       }

       if (esp_ptr_in_drom(func)) {
           ra_to_str(addr);
           func = addr;
       }

       if (esp_ptr_in_drom(expr)) {
           expr = CACHE_DISABLED_STR;
       }
    }

    const char *str[] = {ASSERT_STR, func ? func : "\b", " ", file, ":", lbuf, " (", expr, ")"};

    for (int i = 0; i < sizeof(str) / sizeof(str[0]); i++) {
        uint32_t len = strlen(str[i]);
        uint32_t cpy_len = MIN(len, rem_len);
        memcpy(buff + off, str[i], cpy_len);
        rem_len -= cpy_len;
        off += cpy_len;
        if (rem_len == 0) {
            break;
        }
    }
    buff[off] = '\0';
    esp_system_abort(buff);
#endif  /* CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_SILENT */
}
```