espressif/esp-idf #7213: ESP32S2 Module doesn't work when placed on the PCB

Shravani100, 2021-06-30:
## Environment

- Development Kit:  ESP32S2  Saola - 1M
- Kit version :
- Module or chip used: ESP32S2 WROOM
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v4.2-238-g8cd16b60f-dirty

  
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux Virtual Box
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply:  3.3V External

## Problem Description

I have developed a PCB with the ESP32S2 Module mounted on it. I am flashing the module from ESP32Prog Programmer. I am flashing a simple blink example from the get-started folder. It get flashed successfully but when running monitor command it does not work. While flashing status of the IO0 pin is Low and while executing its status is High. I am giving supply to the board from ESP32 Programmer while flashing the module. I have also given the supply from the PCB for the execution of the software, but no luck.

Please guide If I am doing anything wrong. Please help to solve this.

dobairoland, 2021-06-30:
Hi @Shravani100. I can suggest to use the [Circuit Schematic & PCB Design Review](https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review) support page for help.

Shravani100, 2021-06-30:
Hi @dobairoland. Thank you for the suggestion. I solved this issue. I was connecting wrong capacitor value for the RC Circuit at the EN Pin. I changed the value to 1uF and problem solved.

Shravani100, 2021-06-30:
@dobairoland But whenever I flash the code. It is not showing output on monitor. Do you know the reason?

dobairoland, 2021-06-30:
@Shravani100 During flashing the chip in bootloader mode and the application is not running. So it doesn't show anything.

closed 2021-08-09 by Alvin1Zhang