espressif/esp-idf #7214: Memory overflow due to replay message from mesh-server

sumathi98, 2021-06-30:
## Environment

Development Kit : ESP32-DevKitC
Module or chip used: ESP32-WROOM-32U
IDF version : v4.4.0
Build System : idf.py
Compiler version : 8.4.0
Operating System : Linux
Using an IDE? : No
Power Supply : external 5V

## Problem Description

I'm developing a BLE Mesh Network. Iam using onoff_client example as the base code. I have implemented both onoff_client model  and onoff_server model on same node. Client model for controlling other server nodes and Server model for receiving heartbeat (like I will be publishing a pin state every 5s) from other server nodes. It is working fine for sometime, after sometime it results in out of memory and wifi disconnection.
I don't see any messages for server node is stored in memory. If it is getting stored how to disable it? 
How to overcome this issue?  

## Debug Logs

<27>[0;32mI (163130) EXAMPLE: sent publish successful, msg_id=33588<27>[0m<\r><\n>
<27>[0;31mE (163270) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (163270) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (163310) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (163310) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (163310) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (163310) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (164000) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (164000) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (164000) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (164000) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (164060) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (164060) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (164060) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (164060) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (164240) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (164240) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (164240) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (164240) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;32mI (164550) example_connect: Wi-Fi disconnected, trying to reconnect...<27>[0m<\r><\n>
<27>[0;32mI (167690) example_connect: Wi-Fi disconnected, trying to reconnect...<27>[0m<\r><\n>
<27>[0;31mE (168080) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (168080) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (168080) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (168080) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (168140) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (168140) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (168140) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (168140) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (169100) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (169100) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (169100) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (169100) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (169160) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (169160) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (169160) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (169160) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (169320) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (169320) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (169320) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (169320) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;32mI (170830) example_connect: Wi-Fi disconnected, trying to reconnect...<27>[0m<\r><\n>
<27>[0;31mE (172920) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (172920) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (172920) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (172920) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (172960) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (172960) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (172960) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (172960) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (173140) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (173140) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (173140) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (173140) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;32mI (173970) example_connect: Wi-Fi disconnected, trying to reconnect...<27>[0m<\r><\n>
<27>[0;31mE (174180) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (174180) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (174180) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (174180) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (174360) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (174360) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (174360) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (174360) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (174420) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (174420) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (174420) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (174420) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;32mI (174650) EXAMPLE: Other event id:7<27>[0m<\r><\n>
<27>[0;31mE (174650) TRANSPORT_BASE: Failed to initialize new connection object<27>[0m<\r><\n>
<27>[0;31mE (174650) MQTT_CLIENT: Error transport connect<27>[0m<\r><\n>
<27>[0;32mI (174650) EXAMPLE: MQTT_EVENT_ERROR<27>[0m<\r><\n>
<27>[0;31mE (174660) EXAMPLE: Last error reported from esp-tls: 0x101<27>[0m<\r><\n>
<27>[0;32mI (174660) EXAMPLE: Last errno string (Success)<27>[0m<\r><\n>
<27>[0;32mI (174670) EXAMPLE: MQTT_EVENT_DISCONNECTED<27>[0m<\r><\n>
<27>[0;32mI (177110) example_connect: Wi-Fi disconnected, trying to reconnect...<27>[0m<\r><\n>
<27>[0;31mE (177780) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (177780) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (177780) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (177780) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (178000) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (178000) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (178000) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (178000) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (178020) BLE_MESH: Failed to set mesh/rpl/0062 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (178020) BLE_MESH: Failed to store RPL entry 0x0062<27>[0m<\r><\n>
<27>[0;31mE (178020) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (178030) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (179280) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (179280) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (179280) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
W (179280) wifi:m f probe req l=0<\r><\n>
<\r><\n>
<27>[0;31mE (179290) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (179440) BLE_MESH: Failed to set mesh/rpl/0064 data (err 257)<27>[0m<\r><\n>
<27>[0;31mE (179440) BLE_MESH: Failed to store RPL entry 0x0064<27>[0m<\r><\n>
<27>[0;31mE (179440) BLE_MESH: bt_mesh_alloc_buf, Out of memory<27>[0m<\r><\n>
<27>[0;31mE (179440) BLE_MESH: sdu_recv, Out of memory<27>[0m<\r><\n>

sumathi98, 2021-07-06:
Can I please get any update on above issue..

JiaLWang, 2021-08-17:
Hi @sumathi98 , Could you please provide the projects to reproduce the issue?

Alvin1Zhang, 2024-12-23:
Thanks for reporting, will close due to the issue can't be reproduced and also release/4.4 has been end of life, suggest to update to much newer versions, feel free to reopen with more updates. Thanks.

closed 2024-12-23 by Alvin1Zhang