espressif/esp-idf #7215: ESP32 Pub/Sub in BLE MESH : Error "E (572967) BLE_MESH: Too short SDU + MIC (len 4)"

nitinbhosaleemb, 2021-06-30:
Hello 

I am trying the pub/sub on the sensor client /server model

I did set the following for publish of the sensor client config model:

			<example_ble_mesh_set_msg_common(&common, node, config_client.model, ESP_BLE_MESH_MODEL_OP_MODEL_PUB_SET);

			set.model_pub_set.element_addr = node->unicast_addr;//node->unicast;
			set.model_pub_set.publish_addr = PROV_OWN_ADDR;//node->unicast+1;
			set.model_pub_set.publish_app_idx = prov_key.app_idx;
			//set.model_pub_set.cred_flag = true;
			set.model_pub_set.publish_ttl = 7;
			set.model_pub_set.publish_period = 0x004a;
			set.model_pub_set.publish_retransmit = 1;
			set.model_pub_set.model_id = ESP_BLE_MESH_MODEL_ID_SENSOR_SETUP_SRV;
			set.model_pub_set.company_id = ESP_BLE_MESH_CID_NVAL;

			err = esp_ble_mesh_config_client_set_state(&common, &set);
			if (err) {
				ESP_LOGE(TAG, "Failed to send Config Model App Bind");
				return;
			}>

Error at client side =>

E (52637) BLE_MESH: Too short SDU + MIC (len 4)
E (52777) BLE_MESH: Too short SDU + MIC (len 4)
E (62657) BLE_MESH: Too short SDU + MIC (len 4)
E (62787) BLE_MESH: Too short SDU + MIC (len 4)
E (72657) BLE_MESH: Too short SDU + MIC (len 4)
E (72787) BLE_MESH: Too short SDU + MIC (len 4)
E (82657) BLE_MESH: Too short SDU + MIC (len 4)
E (82797) BLE_MESH: Too short SDU + MIC (len 4)

sensor server config command received:

I (4710) EXAMPLE: $$$$$$$$$$ example_ble_mesh_config_server_cb()  $$$$$$$$$$$$$$
I (4710) EXAMPLE: $$$$$$$$ CASE ESP_BLE_MESH_MODEL_OP_MODEL_PUB_SET $$$$$$$$$$$$
I (4720) EXAMPLE: element_addr 0x0005, pub_addr 0x0001, pub_period 0x004a, model_id 0x1101

Alvin1Zhang, 2021-06-30:
@nitinbhosaleemb Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

MrCurio, 2021-08-03:
Hi @nitinbhosaleemb ,
Have you defined the publishing buffer in the model structure of the sensor model?

I think there's an example using a macro for defining that buffer in the examples directory.

closed 2021-11-26 by Alvin1Zhang