espressif/esp-idf #7217: Build failure in ip4_napt.c if CONFIG_LWIP_IPV6 is not set

clementfumey-inventhys, 2021-07-01:
## Environment
- Module or chip used:  ESP32-WROOM-32
- IDF version : v4.3-194-g5b5e46971
- Build System :        CMake or idf.py
- Compiler version :  xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes (Visual Studio with esp-idf extension) but same problem without and with idf.py
- Power Supply:         USB

## Problem Description

Build failure if CONFIG_LWIP_IPV6 is not set and when  

### Expected Behavior

It should be able to build with CONFIG_LWIP_IPV6=n

### Actual Behavior

Compilation error

### Steps to reproduce

1. cd examples/mesh/ip_internal_network and add below line to CMakeLists.txt
set(EXCLUDE_COMPONENTS "coap" "asio")

2. idf.py menuconfig
Component Config -> LWIP -> Enable IPv6 (Uncheck it)

3. build
Then got below build error:
`/home/clement/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c: In function 'ip_napt_enable':
/home/clement/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c:179:81: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     if (netif_is_up(netif) && !ip_addr_isany(&netif->ip_addr) && netif->ip_addr.u_addr.ip4.addr == addr && enable) {
                                                                                 ^~~~~~
                                                                                 addr
/home/clement/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c: In function 'ip_napt_forward':
/home/clement/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c:698:61: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
       ip_napt_modify_addr(iphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
                                                             ^~~~~~
                                                             addr
/home/clement/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c:745:64: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     ip_napt_modify_addr_tcp(tcphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
                                                                ^~~~~~
                                                                addr
/home/clement/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c:746:59: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     ip_napt_modify_addr(iphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
                                                           ^~~~~~
                                                           addr
/home/clement/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c:786:64: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     ip_napt_modify_addr_udp(udphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
                                                                ^~~~~~
                                                                addr
/home/clement/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c:787:59: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
     ip_napt_modify_addr(iphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
                                                           ^~~~~~
                                                           addr
[387/962] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/ipv6/dhcp6.c.obj`

clementfumey-inventhys, 2021-07-01:
Could be related with #6935 ?

david-cermak, 2021-07-01:
Hi @clementfumey-inventhys 

We had a PR trying to fix this some time ago, when `LWIP_IPV6=0` wasn't supported in IDF.
The PR wasn't accepted due to some issues, but you can cherry-pick the commit to temporarily work around the compilation issues, see https://github.com/espressif/esp-lwip/pull/21.

This needs to be fixed now, when the `CONFIG_LWIP_IPV6=0` option is officially supported. Will provide the fix.

ESP-YJM, 2021-07-09:
@clementfumey-inventhys Could you please check the following change whether can help you.
```
diff --git a/src/core/ipv4/ip4_napt.c b/src/core/ipv4/ip4_napt.c
index a227263..482aeea 100644
--- a/src/core/ipv4/ip4_napt.c
+++ b/src/core/ipv4/ip4_napt.c
@@ -176,7 +176,7 @@ ip_napt_enable(u32_t addr, int enable)
   for (netif = netif_list; netif; netif = netif->next) {
     if (netif->napt)
       napt_in_any_netif = 1;
-    if (netif_is_up(netif) && !ip_addr_isany(&netif->ip_addr) && netif->ip_addr.u_addr.ip4.addr == addr && enable) {
+    if (netif_is_up(netif) && !ip_addr_isany(&netif->ip_addr) && (ip_2_ip4(&netif->ip_addr)->addr) == addr && enable) {
       netif->napt = 1;
       ip_napt_init(IP_NAPT_MAX, IP_PORTMAP_MAX);
       break;
@@ -695,7 +695,7 @@ ip_napt_forward(struct pbuf *p, struct ip_hdr *iphdr, struct netif *inp, struct
       /* register src addr and iecho->id and dest info */
       ip_napt_add(IP_PROTO_ICMP, iphdr->src.addr, iecho->id, iphdr->dest.addr, iecho->id);
 
-      ip_napt_modify_addr(iphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
+      ip_napt_modify_addr(iphdr, &iphdr->src, ip_2_ip4(&outp->ip_addr)->addr);
     }
 
     return ERR_OK;
@@ -742,8 +742,8 @@ ip_napt_forward(struct pbuf *p, struct ip_hdr *iphdr, struct netif *inp, struct
 
     if (mport != tcphdr->src)
       ip_napt_modify_port_tcp(tcphdr, 0, mport);
-    ip_napt_modify_addr_tcp(tcphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
-    ip_napt_modify_addr(iphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
+    ip_napt_modify_addr_tcp(tcphdr, &iphdr->src, ip_2_ip4(&outp->ip_addr)->addr);
+    ip_napt_modify_addr(iphdr, &iphdr->src, ip_2_ip4(&outp->ip_addr)->addr);
 
     return ERR_OK;
   }
@@ -783,8 +783,8 @@ ip_napt_forward(struct pbuf *p, struct ip_hdr *iphdr, struct netif *inp, struct
 
     if (mport != udphdr->src)
       ip_napt_modify_port_udp(udphdr, 0, mport);
-    ip_napt_modify_addr_udp(udphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
-    ip_napt_modify_addr(iphdr, &iphdr->src, outp->ip_addr.u_addr.ip4.addr);
+    ip_napt_modify_addr_udp(udphdr, &iphdr->src, ip_2_ip4(&outp->ip_addr)->addr);
+    ip_napt_modify_addr(iphdr, &iphdr->src, ip_2_ip4(&outp->ip_addr)->addr);
     return ERR_OK;
   }
 #endif
```

clementfumey-inventhys, 2021-07-09:
Yes, thank you. It solved my problem. I will take some  hours before I can completely validate everything still work. But there is no more build failure for sure. Thanks

closed 2021-07-09 by clementfumey-inventhys

ESP-YJM, 2021-07-09:
@clementfumey-inventhys  Thanks for your feedback. If possible, could you feedback us result after you validate everything.