espressif/esp-idf #7218: ESP-DSP optimized can't be built

ahmedwahdan, 2021-07-02:
## Environment

- Development Kit:     ESP32-DevKitC
Module or chip used:  ESP32-WROOM-32
- IDF version  4.2
- Build System:         idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     WindowsPlain Command Prompt
- Using an IDE?: No|
- Power Supply:         USB

## Problem Description

After adding ESP-DSP component to esp-idf components, only ANSI C function can be compiled, the optimized functions couldn't be found

[image]

### Expected Behavior
Be able to use optimized DSP functions
### Steps to reproduce

1. Clone DSP component to idf components
2. configure the DSP to use optimized functions

[image]

3. Build any DSP example

igrr, 2021-07-02:
@ahmedwahdan could you please attach your hello_world_main.c, your `main` component CMakeLists.txt, and full build output (text, not screenshot)?

ahmedwahdan, 2021-07-02:
I'm compiling the dsp iir example

igrr, 2021-07-02:
That's odd, the screenshot in your first post shows the filename "main/hello_world_main.c", which isn't present in the IIR example of esp_dsp. So it looks like you have either renamed some files, or copied the contents of one example into the other. Hence the request to share the files you are actually using, plus the build log.

ahmedwahdan, 2021-07-03:
@igrr image replaced, build output attached.