espressif/esp-idf #7222: Noise of the Bluetooth HFP via I2S

m-kiso, 2021-07-02:
## Environment

Hardware: ESP32-WROOM-32E
Software: esp-idf V4.3
Example project: \esp32\examples\bluetooth\bluedroid\classic_bt\hfp_ag

## Problem Description

I has succeeded playing a sine audio in the Example project.
So I try to read audio data from I2S and send to the bluetooth HFP(HCI protocol).

I changed bt_app_hf.c in the Example project.
I take audio data from i2s_read() and put s_m_rb(RingBuffer) instead of bt_app_hf_create_audio_data().
Then, I heard only noise.

What should I do?

### Code to reproduce this issue

I2S config is

```cpp

i2s_config_t i2s_config = {
    .mode 			= I2S_MODE_SLAVE | I2S_MODE_RX,
    .sample_rate		= 8000,
    .bits_per_sample 		= I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format 		= I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format	= I2S_COMM_FORMAT_STAND_MSB,
    .intr_alloc_flags 		= ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count 		= 4,
    .dma_buf_len 		= I2S_BUFFER_SIZE,
    .use_apll             	= false,
    .tx_desc_auto_clear 	= false,
    .fixed_mclk           	= 0
};

```

m-kiso, 2021-07-06:
Hi,

I look forward to hearing from you.

xiongweichao, 2021-07-06:
Hi, @m-kiso 

Different codecs have different sampling rates. The sampling rate using CVSD codec is 8k, and the sampling rate using mSBC codec is 16K. If ESP32 is used as I2S slave, you need to pay attention to the problem of data format (big-endian or little-endian).

Thanks

m-kiso, 2021-07-06:
Hi, @xiongweichao 

Thank you for your reply.

I have the following configuration and I want to skip the audio data, but I hear noise in the headset.

[STM32] ---(CVSD 8kHz , 16bit , mono )---> [ESP32] ---( Bluetooth HFP)---> [Headset] 

Also, I have written the source as attached, but I cannot come up with a source to convert the frequency.
I am sorry, but if you have a sample source to convert the audio data, please provide it.

audio_gateway\src\main.c
```
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs.h"
#include "nvs_flash.h"
#include "esp_system.h"
#include "esp_log.h"
#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_bt_device.h"
#include "esp_gap_bt_api.h"

//add
#include "bt_app_core.h"
#include "uart_ctrl.h"
#include "bt_pair_ctrl.h"
#include "i2s_ctrl.h"

#define TAG "MAIN"

void app_main(void)
{
    esp_log_level_set(TAG, ESP_LOG_INFO);

    /* Initialize NVS — it is used to store PHY calibration data */
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));

    esp_err_t err;
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    if ((err = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
        ESP_LOGE(TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    if ((err = esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT)) != ESP_OK) {
        ESP_LOGE(TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    if ((err = esp_bluedroid_init()) != ESP_OK) {
        ESP_LOGE(TAG, "%s initialize bluedroid failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    if ((err = esp_bluedroid_enable()) != ESP_OK) {
        ESP_LOGE(TAG, "%s enable bluedroid failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    /* create application task */
    printf("start\n");
    ESP_LOGI(TAG, "start");

    i2s_ctrl_init();

    bt_app_task_start_up();

    bt_pair_ctrl_init();

    uart_ctrl_task_start_up();
}

```

Regards,

m-kiso, 2021-07-08:
Hi, @xiongweichao Thank you for your reply.

I don't know what this data format (little-endian or big-endian) is, because the audio data is output from the customer's STM32. We are checking now.

What I want to know in a hurry is what kind of conversion expression source I should write for the following configuration.
Or do I not need the conversion formula?

[STM32] ---(CVSD 8kHz , 16bit , mono )---> [ESP32] ---(Bluetooth HFP)---> [Headset].

Regards,

xiongweichao, 2021-07-09:
Hi, @m-kiso 

Please try to modify the value of I2S_BUFFER_SIZE to be greater than 240. If it still doesn't work, please try the following code if it can solve your problem.

```
i2s_config_t i2s_config = {
    .mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
    .sample_rate = 16000,  // CVSD=8000  mSBC=16000
    .bits_per_sample = 16, /* the DAC module will only take the 8bits from MSB */
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .intr_alloc_flags = 0, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 1024,
    .use_apll = true,
};
```

Thanks

m-kiso, 2021-07-09:
Hi, @xiongweichao 

Thank you for your prompt reply!
I will try and reply as soon as possible.

Thanks,

m-kiso, 2021-07-09:
Hi, @xiongweichao 

I haven't changed the I2S_BUFFER_SIZE yet, but I did printf the data acquired by I2S and graphed it, and found that it was being sent in a digital data-like format.

Do I need to convert this data if I send it via HFP?
[image]

Thanks,

m-kiso, 2021-07-09:
Hi, @xiongweichao 

- I set I2S_BUFFER_SIZE to 240 or more, but there was no improvement.
- If you add the following configuration, ESP32 will reboot on startup

```
i2s_config_t i2s_config = {
    .mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
    .sample_rate = 16000, // CVSD=8000 mSBC=16000
    .bits_per_sample = 16, /* the DAC module will only take the 8bits from MSB */ .
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .intr_alloc_flags = 0, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 1024,
    .use_apll = true,
};
```

If you have a next move, please let me know.

Thanks,

m-kiso, 2021-07-12:
Hi, @xiongweichao 

Is there an update on this?

Thanks,

xiongweichao, 2021-07-12:
Hi, @m-kiso 

Is `bt_ctrl_play` used?

Thanks

m-kiso, 2021-07-12:
Hi, @xiongweichao 

Yes, I do use bt_ctrl_play.
Also, the actual sound you hear is something like this.

[2021_07_12_18_45_24.zip](https://github.com/espressif/esp-idf/files/6800496/2021_07_12_18_45_24.zip)

Thanks,

xiongweichao, 2021-07-12:
Hi, @m-kiso 

But in the source code you provided, I didn't see you using bt_ctrl_play.
You can try using the attachment below, this sound is not noise.

Thanks
[audio_example_file.zip](https://github.com/espressif/esp-idf/files/6800929/audio_example_file.zip)

m-kiso, 2021-07-12:
Hi, @xiongweichao 

Thank you for your prompt reply!

I apologize for the lack of explanation regarding bt_ctrl_play.
bt_ctrl_play is used in the following way when a message is received by UART.

```
void uart_ctrl_request_play_voice(uint8_t* dtmp, size_t size){
    uint8_t command = dtmp[3];

    if(m_resp_code == SUCCESS){
        if(command == '0'){
            ESP_LOGI(UART_CTRL_TAG, "play");
            bt_ctrl_play();
            //bt_a2dp_ctrl_play();
        }else if(command == '1'){
            ESP_LOGI(UART_CTRL_TAG, "stop");
            bt_ctrl_stop();
            //bt_a2dp_ctrl_stop();
        }else{
            ESP_LOGI(UART_CTRL_TAG, "unknown");
            bt_ctrl_stop();
            //bt_a2dp_ctrl_stop();
        }
    }

    uart_ctrl_send_response(m_resp_code, NULL, 0);
}
```

Please tell me, did you check the audio in `audio_example_file.zip` with HFP?

Thanks,

xiongweichao, 2021-07-13:
Hi, @m-kiso 

1. Please check whether the data sent by stm32 is the same as the data received by esp32.
2. If the data of the two devices are inconsistent, please confirm whether esp32 is used as i2s slave or i2s master. If esp32 is used as slave, the mclk of stm32 needs to be provided by ESP32.
3. If these methods still cannot solve your problem, please provide the I2S configuration information of stm32 and esp32.

Thanks

m-kiso, 2021-07-13:
Hi, @xiongweichao 

I have some good news. I think I've solved this problem.
I shifted the `i2s_read` value by a higher bit as shown below, and heard clean audio data.

```
			ret = i2s_read(I2S_NUM, samples, sizeof(samples), &read_bytes, portMAX_DELAY);
			if ((ret == ESP_OK) && read_bytes) {
				len = read_bytes / 2;
					for (i = 0; i < len; i++) {
						buffer[i] = samples[i] >> 8;
					}
					xBytesSent = xStreamBufferSend(stream_buffer, (void*)buffer, len, pdMS_TO_TICKS(100));
			}
```

I'm not sure why the problem was solved by shifting the upper bits, but this problem can be closed.

Thank you for your support.

Alvin1Zhang, 2021-07-19:
@m-kiso Thanks for reporting and sharing updates, and glad the issue resolved, feel free to reopen.

closed 2021-07-19 by Alvin1Zhang