espressif/esp-idf #7223: the undefined reference to `__getreent' issue of blecent demo

hiproz, 2021-07-02:
env: windows7 
chip :esp32c3 
code : nimble/blecent

when i build the demo , there is a failure , maybe the __getreent reason, but i didn't confirm it.
===
the below  is console logs:

g:\esp\esp-idf\examples\bluetooth\nimble\blecent>idf.py build
Executing action: all (aliases: build)
Running ninja in directory g:\esp\esp-idf\examples\bluetooth\nimble\blecent\buil
d
Executing "ninja all"...
[0/1] Re-running CMake...
CMake Warning at G:/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/host/nimble/nimble is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  G:/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  G:/esp/esp-idf/tools/cmake/build.cmake:168 (__build_get_idf_git_revision)
  G:/esp/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  G:/esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file G:/esp/esp-idf/examples/bluetooth/nimble/blecent/sdkco
nfig
Loading defaults file G:/esp/esp-idf/examples/bluetooth/nimble/blecent/sdkconfig
.defaults...
CMake Warning at G:/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (me
ssage):
  Xtensa toolchain
  G:/esp/espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-el
f/bin/riscv32-esp-elf-gcc.exe
  crosstool-ng version 1.24.0.123_64eb9ff doesn't match supported version
  esp-2020r3.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  G:/esp/esp-idf/components/esp_common/project_include.cmake:6 (crosstool_versio
n_check)
  G:/esp/esp-idf/tools/cmake/build.cmake:318 (include)
  G:/esp/esp-idf/tools/cmake/build.cmake:533 (__build_process_project_includes)
  G:/esp/esp-idf/tools/cmake/project.cmake:361 (idf_build_process)
  CMakeLists.txt:6 (project)

-- App "blecent" version: v4.4-dev-1404-gc13afea63-dirty
-- Adding linker script G:/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom
.ld
-- Adding linker script G:/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom
.api.ld
-- Adding linker script G:/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom
.libgcc.ld
-- Adding linker script G:/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom
.newlib.ld
-- Adding linker script G:/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom
.version.ld
-- Adding linker script G:/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom
.eco3.ld
-- Adding linker script G:/esp/esp-idf/examples/bluetooth/nimble/blecent/build/e
sp-idf/esp32c3/esp32c3_out.ld
-- Adding linker script G:/esp/esp-idf/components/esp32c3/ld/esp32c3.project.ld.
in
-- Adding linker script G:/esp/esp-idf/components/esp32c3/ld/esp32c3.peripherals
.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor c
mock coap console cxx driver efuse esp-tls esp32c3 esp_adc_cal esp_common esp_et
h esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota es
p_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_pm es
p_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_clien
t esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_t
[... 108 lines cut ...]
oader/subproject/main G:/esp/esp-idf/components/bootloader/subproject/components
/micro-ecc G:/esp/esp-idf/components/newlib G:/esp/esp-idf/components/partition_
table G:/esp/esp-idf/components/riscv G:/esp/esp-idf/components/soc G:/esp/esp-i
df/components/spi_flash
-- Configuring done
-- Generating done
-- Build files have been written to: G:/esp/esp-idf/examples/bluetooth/nimble/bl
ecent/build/bootloader
[1/1] cmd.exe /C "cd /D G:\esp\esp-idf\examples\bluetooth\nimble\blecent\build\b
ootloader\esp-idf\esptool_py && g:\esp\espressif\python_env\idf4.4_py3.8_env\Scr
ipts\python.exe G:/esp/esp-idf/components/partition_table/check_sizes.py --offse
t 0x8000 bootloader 0x0 G:/esp/esp-idf/examples/bluetooth/nimble/blecent/build/b
ootloader/bootloader.bin"
Bootloader binary size 0x3f10 bytes. 0x40f0 bytes (103%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

===
the CMakeError.log :

The output was:
1
g:/esp/espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: g:/esp/espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/lib/rv32im/ilp32\libc.a(lib_a-errno.o): in function `__errno':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/errno/errno.c:13: undefined reference to `__getreent'
collect2.exe: error: ld returned 1 exit status

igrr, 2021-07-02:
The actual error message is this one:
```

cmd.exe /C "cd . && G:\esp\espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.
4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc -nostartfiles -
march=rv32imc --specs=nosys.specs @CMakeFiles\blecent.elf.rsp -o blecent.elf &
& cd ."
g:/esp/espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/
bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: can
not find -lbtdm_app
g:/esp/espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/
bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: can
not find -lbtbb
collect2.exe: error: ld returned 1 exit status
```

I think you need to run `git submodule update --init` to initialize the submodules. This is also mentioned in the warning message:

```
CMake Warning at G:/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
Git submodule components/bt/host/nimble/nimble is out of date. Run 'git
submodule update --init --recursive' to fix.
```

hiproz, 2021-07-03:
,

> The actual error message is this one:
>
> ```
>
> cmd.exe /C "cd . && G:\esp\espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.
> 4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc [...]

i have do this before but it show nothing of errors. may be there are something lose of git submodule ,i will try again .
thanks for  your reply, it give me a clear direction, exclude the  suspicion of newlib

hiproz, 2021-07-03:
> The actual error message is this one:
>
> ```
>
> cmd.exe /C "cd . && G:\esp\espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.
> 4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc [...]

this is the submodule update --init --recursive log :

g:\esp\esp-idf\examples\bluetooth\nimble\blecent>git submodule update --init --r
ecursive Submodule path '../../../../components/bt/host/nimble/nimble': checked out '5bb7
b402275210ed6f97fe4bd6e452bf0f659452'

and the build error is all the same

VaderMester, 2021-07-03:
I just had the same error. The error is in components/esp_phy/CMakeList.txt, in line 32-34

32    if(CONFIG_IDF_TARGET_ESP32C3 OR CONFIG_IDF_TARGET_ESP32S3)
33        target_link_libraries(${COMPONENT_LIB} PUBLIC btbb)
34    endif()

The btbb library does not exist in IDF. If you comment out these line with a #, it will compile and work properly.

projectgus, 2021-07-05:
Hi @VaderMester,

That doesn't seem right. This library does exist in latest master ESP-IDF for the esp32c3 and esp32s3 targets:
https://github.com/espressif/esp-phy-lib/tree/f2c056340505399429dbc8792e7109b7c69f5d77/esp32c3
https://github.com/espressif/esp-phy-lib/tree/f2c056340505399429dbc8792e7109b7c69f5d77/esp32s3

(f2c0563 is the commit used by current IDF master.)

Maybe you need to run `git submodule update --init --recursive` yourself?

Hi @hiproz,

According to your log you have v4.4-dev-1404-gc13afea63-dirty which is a master branch version from May. Can you try reverting whatever change is making the IDF_PATH directory "dirty" and then do "git pull" to get the latest master branch, please?

hiproz, 2021-07-05:
> Hi @VaderMester,
> 
> That doesn't seem right. This library does exist in latest master ESP-IDF for the esp32c3 and esp32s3 targets:
> https://github.com/espressif/esp-phy-lib/tree/f2c056340505399429dbc8792e7109b7c69f5d77/esp32c3
> https://github.com/espressif/esp-phy-lib/tree/f2c056340505399429dbc8792e7109b7c69f5d77/esp32s3
> 
> (f2c0563 is the commit used by current IDF master.)
> 
> Maybe you need to run `git submodule update --init --recursive` yourself? If it still doesn't work, please open a new issue with as many details as possible.
> 
> Hi @hiproz,
> 
> According to your log you have v4.4-dev-1404-gc13afea63-dirty which is a master branch version from May. Can you try reverting whatever change is making the IDF_PATH directory "dirty" and then do "git pull" and "git submodule update --init" (again) to get the latest master branch, please?

ok，i will try later and give the results ,thanks!

hiproz, 2021-07-06:
i did not  success for a new repo  procedure. so i fix it manually :  
use git pull  under  

components\bt\controller\lib_esp32c3_family 
components\esp_wifi\lib

after success, i can build the demo now  .

thanks for the help of all!

hiproz, 2021-07-06:
ELF file SHA256: 5ee3add4282fa469

Rebooting...
x󠄓P-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806ba
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x10d0
load:0x403ce000,len:0x730
load:0x403d0000,len:0x26b4
entry 0x403ce000
I (77) cpu_start: Pro cpu up.
I (86) cpu_start: Pro cpu start user code
I (86) cpu_start: cpu freq: 160000000
I (86) cpu_start: Application information:
I (88) cpu_start: Project name:     blecent
I (93) cpu_start: App version:      v4.4-dev-1404-gc13afea63-dirty
I (100) cpu_start: Compile time:     Jul  6 2021 23:02:47
I (106) cpu_start: ELF file SHA256:  5ee3add4282fa469...
I (112) cpu_start: ESP-IDF:          v4.4-dev-1404-gc13afea63-dirty
I (119) heap_init: Initializing. RAM available for dynamic allocation:
I (126) heap_init: At 3FC921B0 len 0002DE50 (183 KiB): DRAM
I (132) heap_init: At 3FCC0000 len 0001F260 (124 KiB): STACK/DRAM
I (139) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (146) spi_flash: detected chip: generic
I (150) spi_flash: flash io: dio
I (155) sleep: Configure to isolate all GPIO pins in sleep state
I (161) sleep: Enable automatic switching of GPIO sleep configuration
I (168) cpu_start: Starting scheduler.
W (189) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (189) BTDM_INIT: BT controller compile version [9c99115]
I (199) coexist: coexist rom version 8459080
I (199) phy_init: phy_version 500,985899c,Apr 19 2021,16:05:08
Config struct mismatch: magic=5a5aa5a5, ver=02104270
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x403861b1
file: "../main/main.c" line 538
func: app_main
expression: esp_nimble_hci_and_controller_init()

abort() was called at PC 0x403861b5 on core 0
Core  0 register dump:
MEPC    : 0x42000d38  RA      : 0x403861c4  SP      : 0x3fc95050  GP      : 0x3fc8de00
TP      : 0x3fc8683c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x3fc950b4  A0      : 0x3fc9507c  A1      : 0x3fc950b2
A2      : 0x00000000  A3      : 0x3fc950a9  A4      : 0x00000001  A5      : 0x3fc92000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc95050: 0x3c063b00 0x403861b4 0x3fc950b0 0x4038c010 0x00000000 0x00000000 0x00000000 0x3fc8eec4
3fc95070: 0x3fc950b4 0x3fc8eee0 0x3fc950b0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fc95090: 0x43502074 0x34783020 0x36383330 0x20356231 0x63206e6f 0x2065726f 0x00000030 0x00000000
3fc950b0: 0x01000030 0x38333034 0x35623136 0x00000000 0x00000000 0x00000000 0x00000000 0x403861b8
3fc950d0: 0x00000000 0x00000000 0x00000005 0x42007bdc 0x00000000 0x00000000 0x00000000 0x4205305e
3fc950f0: 0x00000000 0x00000000 0x00000000 0x403891c4 0x00000000 0x00000000 0x00000000 0x00000000
3fc95110: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x27146414 0x01010107 0x00000001 0xffffffff
3fc95130: 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000 0x00000000
3fc95150: 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000
3fc95170: 0x3ff00000 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000
3fc95190: 0x43500000 0x3fc8d600 0x00001598 0x3fc9044c 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fc94f60
3fc951b0: 0x3fc921c4 0x3fc901bc 0x3fc901bc 0x3fc951ac 0x3fc901b4 0x00000018 0x0839258c 0x23fe329f
3fc951d0: 0x3fc951ac 0x00000000 0x00000001 0x3fc941a8 0x6e69616d 0x9d8e1300 0x8fac30cf 0x0034e26f
3fc951f0: 0x00000000 0x3fc951a0 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc929dc
3fc95210: 0x3fc92a44 0x3fc92aac 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
3fc95230: 0x42041fc0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95250: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95270: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95290: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952f0: 0x00000000 0x00000000 0x00000000 0x3fc90000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95310: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95330: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95350: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95370: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95390: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95410: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95430: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

hiproz, 2021-07-06:
> i did not success for a new repo procedure. so i fix it manually :
> use git pull under
> 
> components\bt\controller\lib_esp32c3_family
> components\esp_wifi\lib
> 
> after success, i can build the demo now .
> 
> thanks for the help of all!

but runs error:
===============================
ELF file SHA256: 5ee3add4282fa469

Rebooting...
x󠄓P-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806ba
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x10d0
load:0x403ce000,len:0x730
load:0x403d0000,len:0x26b4
entry 0x403ce000
I (77) cpu_start: Pro cpu up.
I (86) cpu_start: Pro cpu start user code
I (86) cpu_start: cpu freq: 160000000
I (86) cpu_start: Application information:
I (88) cpu_start: Project name:     blecent
I (93) cpu_start: App version:      v4.4-dev-1404-gc13afea63-dirty
I (100) cpu_start: Compile time:     Jul  6 2021 23:02:47
I (106) cpu_start: ELF file SHA256:  5ee3add4282fa469...
I (112) cpu_start: ESP-IDF:          v4.4-dev-1404-gc13afea63-dirty
I (119) heap_init: Initializing. RAM available for dynamic allocation:
I (126) heap_init: At 3FC921B0 len 0002DE50 (183 KiB): DRAM
I (132) heap_init: At 3FCC0000 len 0001F260 (124 KiB): STACK/DRAM
I (139) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (146) spi_flash: detected chip: generic
I (150) spi_flash: flash io: dio
I (155) sleep: Configure to isolate all GPIO pins in sleep state
I (161) sleep: Enable automatic switching of GPIO sleep configuration
I (168) cpu_start: Starting scheduler.
W (189) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (189) BTDM_INIT: BT controller compile version [9c99115]
I (199) coexist: coexist rom version 8459080
I (199) phy_init: phy_version 500,985899c,Apr 19 2021,16:05:08
Config struct mismatch: magic=5a5aa5a5, ver=02104270
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x403861b1
file: "../main/main.c" line 538
func: app_main
expression: esp_nimble_hci_and_controller_init()

abort() was called at PC 0x403861b5 on core 0
Core  0 register dump:
MEPC    : 0x42000d38  RA      : 0x403861c4  SP      : 0x3fc95050  GP      : 0x3fc8de00
TP      : 0x3fc8683c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x3fc950b4  A0      : 0x3fc9507c  A1      : 0x3fc950b2
A2      : 0x00000000  A3      : 0x3fc950a9  A4      : 0x00000001  A5      : 0x3fc92000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc95050: 0x3c063b00 0x403861b4 0x3fc950b0 0x4038c010 0x00000000 0x00000000 0x00000000 0x3fc8eec4
3fc95070: 0x3fc950b4 0x3fc8eee0 0x3fc950b0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fc95090: 0x43502074 0x34783020 0x36383330 0x20356231 0x63206e6f 0x2065726f 0x00000030 0x00000000
[... 4 lines cut ...]
3fc95130: 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000 0x00000000
3fc95150: 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000
3fc95170: 0x3ff00000 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000
3fc95190: 0x43500000 0x3fc8d600 0x00001598 0x3fc9044c 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fc94f60
3fc951b0: 0x3fc921c4 0x3fc901bc 0x3fc901bc 0x3fc951ac 0x3fc901b4 0x00000018 0x0839258c 0x23fe329f
3fc951d0: 0x3fc951ac 0x00000000 0x00000001 0x3fc941a8 0x6e69616d 0x9d8e1300 0x8fac30cf 0x0034e26f
3fc951f0: 0x00000000 0x3fc951a0 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc929dc
3fc95210: 0x3fc92a44 0x3fc92aac 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
3fc95230: 0x42041fc0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95250: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95270: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95290: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952f0: 0x00000000 0x00000000 0x00000000 0x3fc90000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95310: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95330: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95350: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95370: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95390: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95410: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95430: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

hiproz, 2021-07-07:
after running, there is a MEM abort:
====================
ELF file SHA256: 5ee3add4282fa469

Rebooting...
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806ba
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x10d0
load:0x403ce000,len:0x730
load:0x403d0000,len:0x26b4
entry 0x403ce000
I (77) cpu_start: Pro cpu up.
I (86) cpu_start: Pro cpu start user code
I (86) cpu_start: cpu freq: 160000000
I (86) cpu_start: Application information:
I (88) cpu_start: Project name:     blecent
I (93) cpu_start: App version:      v4.4-dev-1404-gc13afea63-dirty
I (100) cpu_start: Compile time:     Jul  6 2021 23:02:47
I (106) cpu_start: ELF file SHA256:  5ee3add4282fa469...
I (112) cpu_start: ESP-IDF:          v4.4-dev-1404-gc13afea63-dirty
I (119) heap_init: Initializing. RAM available for dynamic allocation:
I (126) heap_init: At 3FC921B0 len 0002DE50 (183 KiB): DRAM
I (133) heap_init: At 3FCC0000 len 0001F260 (124 KiB): STACK/DRAM
I (139) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (146) spi_flash: detected chip: generic
I (150) spi_flash: flash io: dio
I (155) sleep: Configure to isolate all GPIO pins in sleep state
I (161) sleep: Enable automatic switching of GPIO sleep configuration
I (168) cpu_start: Starting scheduler.
W (179) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (179) BTDM_INIT: BT controller compile version [9c99115]
I (189) coexist: coexist rom version 8459080
I (189) phy_init: phy_version 500,985899c,Apr 19 2021,16:05:08
Config struct mismatch: magic=5a5aa5a5, ver=02104270
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x403861b1
file: "../main/main.c" line 538
func: app_main
expression: esp_nimble_hci_and_controller_init()

abort() was called at PC 0x403861b5 on core 0
Core  0 register dump:
MEPC    : 0x42000d38  RA      : 0x403861c4  SP      : 0x3fc95050  GP      : 0x3fc8de00
TP      : 0x3fc8683c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x3fc950b4  A0      : 0x3fc9507c  A1      : 0x3fc950b2
A2      : 0x00000000  A3      : 0x3fc950a9  A4      : 0x00000001  A5      : 0x3fc92000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc95050: 0x3c063b00 0x403861b4 0x3fc950b0 0x4038c010 0x00000000 0x00000000 0x00000000 0x3fc8eec4
3fc95070: 0x3fc950b4 0x3fc8eee0 0x3fc950b0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fc95090: 0x43502074 0x34783020 0x36383330 0x20356231 0x63206e6f 0x2065726f 0x00000030 0x00000000
3fc950b0: 0x01000030 0x38333034 0x35623136 0x00000000 0x00000000 0x00000000 0x00000000 0x403861b8
3fc950d0: 0x00000000 0x00000000 0x00000005 0x42007bdc 0x00000000 0x00000000 0x00000000 0x4205305e
3fc950f0: 0x00000000 0x00000000 0x00000000 0x403891c4 0x00000000 0x00000000 0x00000000 0x00000000
3fc95110: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x27146414 0x01010107 0x00000001 0xffffffff
3fc95130: 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000 0x00000000
3fc95150: 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000
3fc95170: 0x3ff00000 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000
3fc95190: 0x43500000 0x3fc8d600 0x00001598 0x3fc9044c 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fc94f60
3fc951b0: 0x3fc921c4 0x3fc901bc 0x3fc901bc 0x3fc951ac 0x3fc901b4 0x00000018 0x0839258c 0x23fe329f
3fc951d0: 0x3fc951ac 0x00000000 0x00000001 0x3fc941a8 0x6e69616d 0x9d8e1300 0x8fac30cf 0x0034e26f
3fc951f0: 0x00000000 0x3fc951a0 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc929dc
3fc95210: 0x3fc92a44 0x3fc92aac 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
3fc95230: 0x42041fc0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95250: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95270: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95290: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc952f0: 0x00000000 0x00000000 0x00000000 0x3fc90000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95310: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95330: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95350: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95370: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95390: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc953f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95410: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc95430: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

hiproz, 2021-07-08:
hi，guys，
for confirm this issue， i create a  new linux vps to avoid the git unknown  problem. 
in the new Linux  env, i can build the demo and running it successfully. 

so, i think this issue may be about windows env problem or the  git forbidden problem, if  anyone have the same problem in the windows, i wish this problem can be confirmed in the future.

Thanks for the help of all!

closed 2021-07-08 by hiproz